SBFspot/SBFspot #418: Errors -1 after successful logon

s56g, 2020-11-18:
Hi
I have rather strange problem on Raspberry Pi 4 (Model B Rev 1.2) running "Raspbian GNU/Linux 10 (buster)" ... 

Logon succeeds but every other command times out. Bluetooth signal level is ok even dough SBFspot says it's 0.0%.
There is no difference in behaviour between onboard bluetooth or usb dongle bluetooth adapter.

```
SBFspot V3.7.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2020, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -d3 -finq -nocsv -nosql
...
Wed Nov 18 12:49:38 2020: INFO: Starting...
sunrise: 07:13
sunset : 16:28
Connecting to 00:XX:XX:XX:XX:XX (1/10)
Initializing...
SUSyID: 125 - SessionID: 9XXXXXXXX (0xXXXXXXXX)
SMA netID=02
SUSyID: 128 - SN: 21103XXXXX
SUSyID: 128 - SN: 21103XXXXX
SUSyID: 128 - SN: 21103XXXXX
SUSyID: 128 - SN: 21103XXXXX
BT Signal=0.0%
Logon OK
getInverterData(sbftest) returned an error: -1
getSoftwareVersion returned an error: -1
getTypeLabel returned an error: -1
getDeviceStatus returned an error: -1
getInverterTemperature returned an error: -1
...
```

Bluetooth libs are 
```
bluetooth (5.50-1.2~deb10u1+rpt2)
libbluetooth-dev (5.50-1.2~deb10u1+rpt2)
```

PjotrekSE, 2020-11-18:
Could be filled up sd card. Try 'df -h'

s56g, 2020-11-18:
@PjotrekSE : good point but ...
```
...
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        57G  2.4G   53G   5% /
...
```

sbf-, 2020-11-18:
Looks like an unreliable BT connection. Try to move the RPi closer to the inverter.
A reset of inverter/RPi might help also.
If not solved, please provide full debug log (-d5 -v5)

sbf-, 2020-11-19:
Ok, I see... inverter reset will not solve the issue.
How many inverters do you have? 16 or even more?
I have to study the log to see what's going wrong. This will take some time...

s56g, 2020-11-20:
Thanks!

I checked with my colleague : there are 40 inverters ... compiled with `const int MAX_INVERTERS = 60;` .

sbf-, 2024-01-27:
No idea if this is still relevant.... closing for now

closed 2024-01-27 by sbf-