SBFspot/SBFspot #419: SMA Energymeter & SMA Inverter: report current MeteringGridMsTotW*

fishpepper, 2020-11-18:
I have a SMA Sunnyboy 2.5 and an SMA energy meter. The Sunnyboy reports the Grid usage on its web ui. I want to expose it via mqtt as well.

I did some digging with the explorer windows app and wireshark. The command is 0x..4634.. as specified here;
https://github.com/SBFspot/SBFspot/blob/606aceabf04261cb8ca87199af4bc547f6f5c8fa/SBFspot/SBFspot.h#L374

I took the following lines from the code sections that is only called on devices with hasBattery. It can be used to trigger the transmission just fine:
https://github.com/SBFspot/SBFspot/blob/606aceabf04261cb8ca87199af4bc547f6f5c8fa/SBFspot/SBFspot.cpp#L380-L399

However, i think there is a bug in the actual conversion/data extraction part:
https://github.com/SBFspot/SBFspot/blob/606aceabf04261cb8ca87199af4bc547f6f5c8fa/SBFspot/SBFspot.cpp#L3394-L3402

The case part uses
MeteringGridMsTotWhOut <--- notice the Wh
and sets
devList[inv]->MeteringGridMsTotWOut = value; <--- notice the W

When I change MeteringGridMsTotWhOut to MeteringGridMsTotWOut (same for In) I can extract the current grid W in and out.

Is this a typo or is it needed this way for the battery devices?

I can work out a fix for this issue and add support for the energy meter.

sbf-, 2020-11-18:
> Is this a typo or is it needed this way for the battery devices?

This must be a typo...

> and add support for the energy meter.

Help with #50? Please go ahead!

fishpepper, 2020-11-19:
ok, please have a look at #420 

This will only show the data your inverter retrieves from the energy meter. This is not a direct readout from the energy meter. So this will not help with #50 but should be enough for reliable grid consumption based home automation.

rucksman, 2021-01-14:
> ok, please have a look at #420

Great to see that there is solution, or at least kind of a workaround.

I checked your commit, downloaded 3.7.0 and modified the files according to your commit. Unfortunately it does not compile. After "make mysql" I get this:
```
test -d mysql/obj/ || mkdir -p mysql/obj/
test -d mysql/bin/ || mkdir -p mysql/bin/
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFspot.o
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/TagDefs.o
TagDefs.cpp: In member function ‘int TagDefs::readall(std::string, std::string)’:
TagDefs.cpp:98:64: warning: lambda expressions only available with -std=c++11 or -std=gnu++11
     boost::split(tag_info, tag, [](char c) { return c == '\\'; }); // boost::is_any_of("=") ==> [](char c) { return c == '='; }
                                                                ^
TagDefs.cpp:98:65: error: no matching function for call to ‘split(std::vector<std::basic_string<char> >&, std::string&, TagDefs::readall(std::string, std::string)::<lambda(char)>)’
     boost::split(tag_info, tag, [](char c) { return c == '\\'; }); // boost::is_any_of("=") ==> [](char c) { return c == '='; }
                                                                 ^
TagDefs.cpp:98:65: note: candidate is:
In file included from /usr/include/boost/algorithm/string.hpp:23:0,
                 from TagDefs.cpp:42:
/usr/include/boost/algorithm/string/split.hpp:140:35: note: template<class SequenceSequenceT, class RangeT, class PredicateT> SequenceSequenceT& boost::algorithm::split(SequenceSequenceT&, RangeT&, PredicateT, boost::algorithm::token_compress_mode_type)
         inline SequenceSequenceT& split(
                                   ^
/usr/include/boost/algorithm/string/split.hpp:140:35: note:   template argument deduction/substitution failed:
TagDefs.cpp: In substitution of ‘template<class SequenceSequenceT, class RangeT, class PredicateT> SequenceSequenceT& boost::algorithm::split(SequenceSequenceT&, RangeT&, PredicateT, boost::algorithm::token_compress_mode_type) [with SequenceSequenceT = std::vector<std::basic_string<char> >; RangeT = std::basic_string<char>; PredicateT = TagDefs::readall(std::string, std::string)::<lambda(char)>]’:
TagDefs.cpp:98:65:   required from here
TagDefs.cpp:98:65: error: template argument for ‘template<class SequenceSequenceT, class RangeT, class PredicateT> SequenceSequenceT& boost::algorithm::split(SequenceSequenceT&, RangeT&, PredicateT, boost::algorithm::token_compress_mode_type)’ uses local type ‘TagDefs::readall(std::string, std::string)::<lambda(char)>’
     boost::split(tag_info, tag, [](char c) { return c == '\\'; }); // boost::is_any_of("=") ==> [](char c) { return c == '='; }
                                                                 ^
TagDefs.cpp:98:65: error:   trying to instantiate ‘template<class SequenceSequenceT, class RangeT, class PredicateT> SequenceSequenceT& boost::algorithm::split(SequenceSequenceT&, RangeT&, PredicateT, boost::algorithm::token_compress_mode_type)’
makefile:116: recipe for target 'mysql/obj/TagDefs.o' failed
make: *** [mysql/obj/TagDefs.o] Error 1
```
Any idea what the problem could be? Running on a RaspberryPi on Raspbian Jessie. Version 3.6 was compiling just fine.

sbf-, 2021-01-14:
I guess your compiler is too old (on Jessie)
You should add -std=c++11 to the compiler options in the makefile (or upgrade your pi to buster)

rucksman, 2021-01-15:
Thanks for your quick response. Upgrade is not possible (at least at the moment) unfortunately. Sorry for this silly question, but where would I have to insert "-std=c++11" in the makefile? My trial and error experiments did not succeed... I do not have any experience in c++.

rucksman, 2021-01-15:
So I made some progress. I added "-std=gnu++11" to CFLAGS in the makefile, and at least it started compiling (I am working on version 3.7.0). But after some time I get:

```
test -d mysql/obj/ || mkdir -p mysql/obj/
test -d mysql/bin/ || mkdir -p mysql/bin/
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -std=gnu++11 -DUSE_MYSQL -o mysql/obj/TagDefs.o
g++ Bluetooth.cpp -c -Wall -O2 -Wno-unused-local-typedefs -std=gnu++11 -DUSE_MYSQL -o mysql/obj/Bluetooth.o
g++ mqtt.cpp -c -Wall -O2 -Wno-unused-local-typedefs -std=gnu++11 -DUSE_MYSQL -o mysql/obj/mqtt.o
g++ db_MySQL.cpp -c -Wall -O2 -Wno-unused-local-typedefs -std=gnu++11 -DUSE_MYSQL -o mysql/obj/db_MySQL.o
g++ db_MySQL_Export.cpp -c -Wall -O2 -Wno-unused-local-typedefs -std=gnu++11 -DUSE_MYSQL -o mysql/obj/db_MySQL_Export.o
g++ mysql/obj/boost_ext.o mysql/obj/misc.o mysql/obj/sunrise_sunset.o mysql/obj/SBFNet.o mysql/obj/CSVexport.o mysql/obj/Ethernet.o mysql/obj/EventData.o mysql/obj/ArchData.o mysql/obj/SBFspot.o mysql/obj/TagDefs.o mysql/obj/Bluetooth.o mysql/obj/mqtt.o mysql/obj/db_MySQL.o mysql/obj/db_MySQL_Export.o -s -o mysql/bin/SBFspot -Wl,-Bdynamic -lpthread -lbluetooth -lboost_date_time -lmysqlclient
mysql/obj/SBFspot.o: In function `GetConfig(Config*)':
SBFspot.cpp:(.text+0x6dec): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x6e30): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x6e54): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x6efc): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x6f1c): undefined reference to `boost::system::system_category()'
mysql/obj/SBFspot.o: In function `_GLOBAL__sub_I_MAX_CommBuf':
SBFspot.cpp:(.text.startup+0x26a0): undefined reference to `boost::system::generic_category()'
SBFspot.cpp:(.text.startup+0x26a8): undefined reference to `boost::system::generic_category()'
SBFspot.cpp:(.text.startup+0x26b0): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text.startup+0x26b8): undefined reference to `boost::system::system_category()'
collect2: error: ld returned 1 exit status
makefile:119: recipe for target 'mysql/bin/SBFspot' failed
make: *** [mysql/bin/SBFspot] Error 1
```
Any idea what I have to change? Thanks for helping, C++ is really not my home turf ...

@fishpepper : I tried incorporating the changes you made to the files I found in your PR 931ae07a4018651aa0152778b381a4c62d8aaa08 into version 3.6.0 of SBFspot because this is the only version I am able to compile. Compilation worked, but there is no output when doing 
`/usr/local/bin/sbfspot.3/SBFspot -finq -nocsv -nosql -v5 -mqtt`
there is nothing in the output:
```
MQTT: Don't know what to do with 'MeteringWTot'
MQTT: Publishing (sbfspot) "MeteringWTot":
```
So does this mean your changes do not work in 3.6.0?

notreallybob, 2021-01-15:
Make sure you have the "boost" libs installed. On ubuntu/debian, it should be "sudo apt-get install libboost-all-dev".

sbf-, 2021-01-16:
> Make sure you have the "boost" libs installed. On ubuntu/debian, it should be "sudo apt-get install libboost-all-dev".

No, it's a bug in 3.7.0 (#403) and fixed in 3.7.1
Alternatively you can add `boost_system` to the makefile:
```
LIBS := pthread bluetooth boost_date_time boost_system
```

rucksman, 2021-01-16:
Thanks @sbf- ! This did the trick, It compiled the energymeter fork from @fishpepper just fine. Also the additonal MQTT data is generated and works.

commit 4f910a9c (2021-12-28): Fix #419 SMA Energymeter & SMA Inverter: report current MeteringGridMsTotW*
    Fix #420/#440 (manually merged) Add inverter power measurements from a connected energy meter
closed 2021-12-28 by sbf-