SBFspot/SBFspot #422: Compile error: start.S:110: undefined reference to `main'

Ceaus, 2021-01-02:
I've downloaded SBFspot-3.7.0. Running make from the `SBFspot` subdir, I'm running into the following error message:
```
dev@dev:~/SBFspot/src/SBFspot-3.7.0/SBFspot> make
test -d  || mkdir -p 
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++  -s -o sqlite/bin/SBFspot -Wl,-Bdynamic -lpthread -lbluetooth -lboost_date_time 
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: /usr/lib64/gcc/x86_64-suse-linux/7/../../../../lib64/crt1.o: in function `_start':
/home/abuild/rpmbuild/BUILD/glibc-2.26/csu/../sysdeps/x86_64/start.S:110: undefined reference to `main'
collect2: error: ld returned 1 exit status
make: *** [makefile:119: sqlite/bin/SBFspot] Fout 1

```
Any suggestion how to fix this?

sbf-, 2021-01-02:
https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite

Ceaus, 2021-01-03:
Thanks for the pointer. I missed that. From the Wiki main page I took "Installation Linux SQLite" as how to install SQLite on Linux. But apparently it's the page for the full build. May I kindly suggest to rephrase that bit on the main page?

After following the instructions I ran into the next compile error:
`undefined reference to boost::system::system_category()`

Googling this up, I came across [this page](https://stackoverflow.com/questions/9723793/undefined-reference-to-boostsystemsystem-category-when-compiling) on stackoverlflow telling me "_Assuming you use gcc, try adding `-lboost_system` to your compiler command line in order to link against that library_." I added that bit to the makefile and it worked. That gave me a successful built. I'm not sure if that's particiular to my environment. Maybe that's something to improve?

Thanks again. Feel free to close this issue.

sbf-, 2021-01-05:
Changed the wiki text to "SBFspot Installation on Linux with SQLite"

For `undefined reference to boost::system::system_category()` see also #394

closed 2021-01-05 by sbf-