SBFspot/SBFspot #423: Archive data: power and yield per MPP

mincequi, 2021-01-11:
When i download my PV data directly from an SMA inverter, i can obtain the power and yield per MPP.
Unfortunately, i do not see those values in my sqlite database, but only the total power and yield per inverter.

Is there a way to get power and yield per MPP?

sbf-, 2021-01-12:
What do you mean with download? Tool used?
SBFspot uses SMAdata2+ protocol (SunnyExplorer compatible)
AFAIK this is not possible with SE, but maybe it is via MODBUS?

mincequi, 2021-01-12:
My STP10.0-3AV-40 has a built-in web server, where I can download archive data per MPP as CSV.
On the other side my STP 20000TL-30 hast no web server and I did not find a way to extract data per MPP or per string.

Also my SE only shows current values per MPP. Unfortunately, this is not true for archive data.... :(

sbf-, 2021-01-12:
So, conclusion: retreiving archive data per MPP is not possible with SBFspot/SunnyExplorer

closed 2021-01-12 by sbf-