SBFspot/SBFspot #424: Connect to specific id of SMA

AnnieBts, 2021-01-14:
Hello,
I am using SBF in a short period of time and so far it is great. I have a raspberry pi 4 model B. The Raspberry Pi comes with implemented bluetooth.
 I would like to set it in a way that, it will connect to a specific bluetooth via the id of the SMA inverter and then take it to a different location that the inverter will be. My problem is that the inverter is not near when I set up the raspberry pi and the SBF settings, so I don't have access to the inverter's bluetooth , that I want the raspberry pi to connect with.
Is there a way to set up the SBF, that I can give a specific SMA id and connect only to that, without scanning for bluetooth near by?

Thanks in advance!

sbf-, 2021-01-16:
So, SBFspot runs fine but you want to move the RPi to another place? Is that correct?
I don't understand exactly what you really want to do.

AnnieBts, 2021-01-16:
The SBFspot runs fine and I want to move raspberry to another place. The inverter there will be different. So, I'm wondering if there is a way to have the SBF prefixed to get data from inverter with specific id.

sbf-, 2021-01-16:
> The SBFspot runs fine and I want to move raspberry to another place. The inverter there will be different. So, I'm wondering if there is a way to have the SBF prefixed to get data from inverter with specific id.

Sure, but you need to know the BT address of the other inverter.
Modify the SBFspot.cfg file:
```
BTAddress=00:00:00:00:00:00
```

closed 2021-02-05 by sbf-