SBFspot/SBFspot #427: How to integrate third party library? (e.g. MQTT client lib).

mincequi, 2021-01-18:
Hi,
i plan to also publish archive data via MQTT. Do you have any preferences on how to add/integrate a MQTT client library? (instead of using a separate executable). This is needed to publish binary data.

One idea is to use git submodules. As alternative one could just commit the third party code directly to SBFSpot.

I had a look at eMQTT5 (preferred) and PAHO C++ library.

sbf-, 2021-01-20:
I have no preferences as long as it works on Windows and Linux and it doesn't conflict with the current MQTT solution. In the future the mosquito method can be fased out, but for now we have to keep it.
Using submodules, I suppose you will always have the latest version? Committing the code to SBFspot someone has to do the updates?

mincequi, 2021-01-22:
I will provide a PR. My solution leverages CMake to optionally support libmosquittopp, while keeping the old solution. My CMake script now uses system-wide installations of those libs (if present).

To support external third-party libs, i will investigate vcpkg (from Microsoft) and conan.io. This should do the trick.

Please notice, that i only support CMake for now. Since Visual Studio also supports Cmake, i think, it could be a good idea to drop Makefile and VCProj files. CMake could do it all :)

sbf-, 2021-01-24:
Well, to be honest, not so happy with this approach...
You only support CMake, but I have to rework all of the docs?

This is what I get for now (after installing cmake - yet another build system)
```
pi@raspberrypi:~/mincequi/SBFspot/SBFspot $ cmake .
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   date_time
-- Checking for module 'bluez'
--   Found bluez, version 5.50
-- Checking for module 'msgpack'
--   No package 'msgpack' found
-- Checking for module 'libmosquittopp'
--   No package 'libmosquittopp' found
-- Configuring done
CMake Error at CMakeLists.txt:79 (add_executable):
  Target "SBFspot" links to target "PkgConfig::MessagePack" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:79 (add_executable):
  Target "SBFspot" links to target "PkgConfig::Mosquitto" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
-- Build files have been written to: /home/pi/mincequi/SBFspot/SBFspot
```

mincequi, 2021-01-24:
Thanks for your feedback. I did not test against missing libs. There is an update on my PR which will fix this.

Well, due to the cross-platform nature of your project, i think, CMake can help. Once set up and working properly, it can save you a lot of time and maintenance. I fully agree on your scepticism against another build system. However, CMake became the de facto standard of build systems on Linux (arguable, there are more available). Also within the Microsoft (Visual Studio) ecosystem CMake rised to the recommended way to support cross-platform builds: https://docs.microsoft.com/en-us/cpp/build/projects-and-build-systems-cpp.
SBFSpot will only benefit from it. :)
Currently, I consider Cmake only as a side-path for your project, but i am willing to help you out with updating docs and fixing bugs, once this is running :)

My PR is updated and shall not fail again, if the optional libs are not present. To enable support for my change, you can install the libs:
> sudo apt install libmosquittopp-dev libmsgpack-dev

Anyhow, thanks for testing and i am looking forward to provide some more PRs and get some feedback from you.

closed 2021-04-12 by mincequi