SBFspot/SBFspot #430: No upload PVoutput after power outage

eterpstra, 2021-01-23:
Hello,

Since a poweroutage, my SMA Tripple is not uploading to PVoutput anymore. Everything seems to be working fine. I have installed a new VM and did a fresh installation to see whats wrong.

```
/var/logs/sbfspot/ 

 GNU nano 4.8                           SBFspotUpload20210123.log                                      
[10:07:46] INFO: SBFspotUploadDaemon Version 2.0.0
[10:07:46] INFO: Starting Daemon...
[10:07:46] WARNING: SMA@Work is not yet member of SBFspot Team. Consider joining at http://pvoutput.org>
[10:14:46] INFO: Stopping Daemon...
[10:14:46] INFO: SBFspotUploadDaemon Version 2.0.0
[10:14:46] INFO: Starting Daemon...
```

MyPlant_20210123.log:

```
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 879586704 (0x346D6D90)
Inverter IP address: 192.168.1.30 from SBFspot.cfg
Logon OK
SUSyID: 181 - SN: xxxxxxxxxxxxx
Device Name:      STP 5000TL-20 871
Device Class:     Solar Inverters
Device Type:      STP 5000TL-20
Software Version: 02.52.02.R
Serial number:    xxxxxxxxxxxxxxxxxxxxx
SUSyID: 181 - SN: xxxxxxxxxxxx
Device Status:      Ok
SUSyID: 181 - SN: xxxxxxxxxxxxx
Device Temperature: 37.4      C
SUSyID: 181 - SN: xxxxxxxxxxxxxxx
GridRelay Status:      Closed
SUSyID: 181 - SN: xxxxxxxxxxxxx
Pac max phase 1: 5000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 181 - SN: xxxxxxxxxxxxx
Energy Production:
        EToday: 1.488kWh
        ETotal: 26793.778kWh
        Operation Time: 17913.13h
        Feed-In Time  : 17403.65h
SUSyID: 181 - SN: xxxxxxxxxxxxx
```

What can be the issue?

sbf-, 2021-01-23:
See [wiki](https://github.com/SBFspot/SBFspot/wiki/No-uploads-to-PVOutput.org) for some troubleshooting tips

eterpstra, 2021-01-23:
I checked.

This is working fine:

sqlite> select * from vwinverters;
`xxxxxx|STP 5000TL-20 871|STP 5000TL-20|02.52.02.R|2021-01-23 14:07:01|596|4459|26796749|17916.8|17407.3|OK|Closed|38.84`

But then Query for daydata: (you should see most recent data) doenst give any data back:

```
sqlite> select * from vwdaydata order by timestamp desc limit 5;
sqlite> 
```

also this doenst give any data:

```
SELECT strftime('%Y%m%d,%H:%M',TimeStamp) DTTM,V1,V2,V3,V4,V5,V6,V7,V8,V9,V10,V11,V12
FROM [vwPvoData]
WHERE TimeStamp>DATE(DATE(),'-14 day')
    AND PVoutput IS NULL
--  AND Serial=304xxxxxx
ORDER BY TimeStamp
LIMIT 5;
```

sbf-, 2021-01-23:
Are you sure SBFspot is running every 5 mins (cron job?)
Maybe a corrupted db? Try `pragma integrity_check;` This should return `OK`

eterpstra, 2021-01-23:
crontab -e : 

```
## SBFspot
*/1 * * * * /usr/local/bin/sbfspot.3/daydata
55 05 * * * /usr/local/bin/sbfspot.3/monthdata
```

```
sqlite> pragma integrity_check;
ok
```

sbf-, 2021-01-24:
DB read-only?

eterpstra, 2021-01-24:
Not sure how to check that..  I have created a database with:

`sqlite3 SBFspot.db < /home/root/SBFspot/SBFspot/CreateSQLiteDB.sql`  which worked fine and in the past.

sbf-, 2021-01-25:
Run SBFspot manually as user pi (not root/sudo) with verbose logging (-v5) and look for errors.
Then, running this query should return several records:
```
select * from daydata order by timestamp desc limit 10;
```

sbf-, 2021-01-29:
Is this fixed?

closed 2021-02-05 by sbf-