SBFspot/SBFspot #433: CRITICAL: Logon failed. Check 'USER' Password

VenimK, 2021-02-02:
Can 't seem to connect on this

root@domoticz:~/SBFspot/SBFspot# /usr/local/bin/sbfspot.3/SBFspot -d5 -v5
SBFspot V3.8.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=192.168.0.182
Password=<undisclosed>
MIS_Enabled=0
Plantname=MyPlant
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=50.8
Longitude=4.33
Timezone=Europe/Brussels
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Tue Feb  2 14:32:15 2021: INFO: Starting...
sunrise: 07:17
sunset : 16:35
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 899442147 (0x359C65E3)
Inverter IP address: 192.168.0.182 from SBFspot.cfg
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 26 00 10 60 65 09 A0 
00000020: FF FF FF FF FF FF 00 00 7D 00 
00000030: E3 65 9C 35 00 00 00 00 00 00 
00000040: 01 80 00 02 00 00 00 00 00 00 
00000050: 00 00 00 00 00 00 00 00 
58 Bytes sent to IP [192.168.0.182]
ethGetPacket()
select() returned 1
MAX_CommBuf is now 98 bytes
Received 98 bytes from IP [192.168.0.182]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 4E 00 10 60 65 13 A0 
00000020: 7D 00 E3 65 9C 35 00 C0 99 01 
00000030: B2 7B 4F B3 00 00 00 00 00 00 
00000040: 01 80 01 02 00 00 00 00 00 00 
00000050: 00 00 00 00 00 03 00 00 00 FF 
00000060: 00 00 00 00 00 00 01 00 99 01 
00000070: B2 7B 4F B3 00 00 0A 00 0C 00 
00000080: 00 00 00 00 00 00 00 00 00 00 
00000090: 01 01 00 00 00 00 00 00 
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 00 00 10 60 65 13 A0 7D 00 E3 
00000010: 65 9C 35 00 C0 99 01 B2 7B 4F 
00000020: B3 00 00 00 00 00 00 01 80 01 
00000030: 02 00 00 00 00 00 00 00 00 00 
00000040: 00 00 03 00 00 00 FF 00 00 00 
00000050: 00 00 00 01 00 99 01 B2 7B 4F 
00000060: B3 00 00 0A 00 0C 00 00 00 00 
00000070: 00 00 00 00 00 00 00 01 01 00 
00000080: 00 00 00 00 
<<<=================================>>>
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 22 00 10 60 65 08 A0 
00000020: FF FF FF FF FF FF 00 03 7D 00 
00000030: E3 65 9C 35 00 03 00 00 00 00 
00000040: 02 80 0E 01 FD FF FF FF FF FF 
00000050: 00 00 00 00 
54 Bytes sent to IP [192.168.0.182]
logonSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 3A 00 10 60 65 0E A0 
00000020: 99 01 B2 7B 4F B3 00 01 7D 00 
00000030: E3 65 9C 35 00 01 00 00 00 00 
00000040: 03 80 0C 04 FD FF 07 00 00 00 
00000050: 84 03 00 00 6F 62 19 60 00 00 
00000060: 00 00 D8 F7 F1 FD C1 B9 BA B8 
00000070: B9 BA E7 88 00 00 00 00 
78 Bytes sent to IP [192.168.0.182]
ethGetPacket()
select() returned 1
Received 66 bytes from IP [192.168.0.182]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 2E 00 10 60 65 0B E0 
00000020: 7D 00 E3 65 9C 35 00 01 99 01 
00000030: B2 7B 4F B3 00 01 00 01 00 00 
00000040: 03 80 0D 04 FD FF 07 00 00 00 
00000050: 84 03 00 00 6F 62 19 60 00 00 
00000060: 00 00 00 00 00 00 
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 00 00 10 60 65 0B E0 7D 00 E3 
00000010: 65 9C 35 00 01 99 01 B2 7B 4F 
00000020: B3 00 01 00 01 00 00 03 80 0D 
00000030: 04 FD FF 07 00 00 00 84 03 00 
00000040: 00 6F 62 19 60 00 00 00 00 00 
00000050: 00 00 
<<<=================================>>>
Tue Feb  2 14:32:15 2021: CRITICAL: Logon failed. Check 'USER' Password
Tue Feb  2 14:32:15 2021: INFO: Done.

Connected via LAN

sbf-, 2021-02-02:
Well, check and double check the password in SBFspot.cfg
It is case sensitive.
Are you able to login using SunnyExplorer?

VenimK, 2021-02-02:
Don 't have a windows pc @ the moment
i'm using the password on sunnyportal en that works
my inverter is a Sunny Boy 3.0 1AV-41

sbf-, 2021-02-02:
Is this the same password? Can you try with the default password `0000`?

closed 2021-02-05 by sbf-