SBFspot/SBFspot #436: SBFspot stopped uploading to pvoutput.org

roelvdh, 2021-02-11:
Hi,

I have 2 SMA-systems and have happily used SBFspot for quite some time uploading both systems to pvoutput.org.
My System-1 output is correctly reported on a daily basis. Only now I noticed that System-2 has not been uploaded since Dec 2nd, 2020. I am not aware of any changes to the configs, nor SBFspot, nor pvoutput.org.
The *.csv-files seem to be OK at /home/pi/smadata/2021.

/usr/local/bin/sbfspot.3/SBFspotUpload.cfg :
PVoutput_SID=2002nnnnnnn:mmmmm,2002rrrrrrrrrr:sssss
PVoutput_Key=ttttttttttttttttttttttttttttttttttttttttttttt
SQL_Database=/home/pi/smadata/SBFspot.db

I have 2 questions:
 - what could be the reason for not updating the second system, while the first system is OK?
 - how can I retroactively upload all the *.csv-files since Dec 2nd, 2020 for the second system?
 
Roel

sbf-, 2021-02-11:
There are some troubleshooting tips in the wiki
https://github.com/SBFspot/SBFspot/wiki/No-uploads-to-PVOutput.org

roelvdh, 2021-02-11:
The SBF wiki is absolutely excellent. You may want to know my upload problem solved itself by rebooting the pi.
This is strange as the System1-data have been uploaded, while System-2 was not, without any changes to the configs. 
Would it be recommendable rebooting the pi on a regular basis?
Thank you very much for providing really excellent software.

sbf-, 2021-02-13:
> Would it be recommendable rebooting the pi on a regular basis?

Personally I see no reason to do so. If this issue happens again it might be a valid option.

closed 2021-02-28 by sbf-