SBFspot/SBFspot #437: CRITICAL: Logon failed. Check 'USER' Password

mincequi, 2021-02-16:
Hi,

from time to time i get this issue. SBFspot runs for days and all of a sudden, i cannot login at the inverter anymore.
Since i have two inverters, i can change the Bluetooth address using the second inverter and all works again.
Then, this issue arises and i switch back to the original bluetooth address of the first inverter and it will keep running again, until the next login failure. I am using a custom password having a "!" in it. I read somewhere, this might cause issues.

Here are my logs:

MQTT_Data=Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvTemperature,InvGridRelay,EToday,ETotal,PACTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2
MQTT_ItemFormat=MSGPACK
### End of Config ###
Tue Feb 16 16:07:07 2021: INFO: Starting...
sunrise: 06:20
sunset : 16:35
sunrise: 06:20
sunset : 16:35
Next poll: Tue Feb 16 16:08:00 2021

Connecting to 00:80:25:25:EE:B2 (1/10)
Initializing...
SUSyID: 125 - SessionID: 993753928 (0x3B3B7B48)
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 17 00 69 00 00 00 00 00 00 
00000010: 01 00 00 00 00 00 01 02 76 65 
00000020: 72 0D 0A 
23 Bytes sent
bthGetPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 B2 EE 25 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 05 00 00 00 00 01 00 00 
00000030: 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 B2 EE 25 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 05 00 00 00 00 01 00 00 
00000030: 00 
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=05
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00 
00000010: B2 EE 25 25 80 00 02 00 00 04 
00000020: 70 00 05 00 00 00 00 01 00 00 
00000030: 00 
31 Bytes sent
bthGetPacket(10)
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 B2 EE 25 25 80 00 
00000010: 00 00 00 00 00 00 0A 00 B2 EE 
00000020: 25 25 80 00 01 11 71 DA 7D 1A 
00000030: 00 
cmd=10
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 B2 EE 25 25 80 00 
00000010: 00 00 00 00 00 00 0A 00 B2 EE 
00000020: 25 25 80 00 01 11 71 DA 7D 1A 
00000030: 00 
<<<=================================>>>
Root device address: 00:80:25:25:EE:B2
Local BT address: 00:1A:7D:DA:71:11
bthGetPacket(5)
[... 201 lines cut ...]
cmd=1
PacketLength=86
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 0E D0 7D 00 48 
00000010: 7B 3B 3B 00 01 80 00 7E 13 C9 
00000020: 7D 00 01 01 00 00 00 05 80 0D 
00000030: 04 FD FF 07 00 00 00 84 03 00 
00000040: 00 E0 ED 2B 60 00 00 00 00 D5 
00000050: F1 F6 FB F3 F1 BA BD B8 BC A9 
00000060: 88 C0 4C 7E 
<<<=================================>>>
Tue Feb 16 16:08:00 2021: CRITICAL: Logon failed. Check 'USER' Password
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3B 00 45 11 71 DA 7D 1A 00 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 08 A0 FF FF FF FF FF 
00000030: FF 00 03 7D 5D 00 48 7B 3B 3B 
00000040: 00 03 00 00 00 00 06 80 0E 01 
00000050: FD FF FF FF FF FF E9 E1 7E 
send() returned an error: 88
sunrise: 06:20
sunset : 16:35
Next poll: Tue Feb 16 16:09:00 2021

sbf-, 2021-02-21:
Is this with your modified code running as daemon, keeping the connection open?

mincequi, 2021-02-21:
Yes, this is a modified version running in daemon mode. However, i do not keep the connection open.
I logOn/Off each single poll. Might there be an issue due to previously written variables?

sbf-, 2021-02-25:
The issue with the '!' in the password is about a bug in [sbfspot-config](https://github.com/SBFspot/sbfspot-config/issues/31)
But while testing this I found that when you add an '!' at the end of your PW, the '!' is not taken into account. So 'password' and 'password!' both work.

sbf-, 2021-02-25:
> Tue Feb 16 16:08:00 2021: CRITICAL: Logon failed. Check 'USER' Password
> Next poll: Tue Feb 16 16:09:00 2021

Maybe the frequency (60 per minute) is too high?

closed 2021-04-18 by sbf-

tfvtfv, 2022-01-29:
The problem seems to be teh following
- new versions of sunny explorer force the user to set a password, default passwords no longer allowed
- sbfspot seems to have problems handling non default password
- Setting the password in the cfg file seems to not work properly

What has worked for me:
- set identical user and installer passwords in sunny explorer.
- only use letters and numbers
- add the password in the command line using syntax as below:

$ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv -installer -password:<xxxx>

sbf-, 2022-01-30:
User and Installer passwords don't need to be the same.
User login needs password in cfg file
Installer login needs password in command line (`-installer -password:****`)
A password has must have at least 8 characters: (at least) one uppercase and one lowercase letter and one numeric. Special characters '?', '_', '!' and '-' are allowed. All the rest like '$', '*' 'é' are refused.
[image]

When using [sbfspot-config](https://github.com/SBFspot/sbfspot-config) there might be an issue with special characters. I need to check this.

tfvtfv, 2022-01-30:
@sbf
I used a user password as specified above and manually inserted this into the config file, since the sbfspot-config could not handle this. However, it refused connection, and i was not able to find out why.

So meanwhile I ended up downgrading sunny explorer to a version allowing standard passwords and using the 0000 password.

nihonjin98, 2022-02-02:
I have the same issue.  Despite rolling back sunny explorer and setting the password to letters only it still gives me the same error with user password.  Not sure if its related to the fact that sunny portal also requires a complex password set for the PV system and from what i understand when you set the PV system password in sunny portal this applies the same password to any devices registered within sunny portal.  Even though i have rolled back sunny explorer to allow for simple password you cannot change the sunny portal to a simple password.  Anyway sbf I look forward to an update where SBFSpot can handle complex passwords. Thanks.

sbf-, 2022-02-02:
> Anyway sbf I look forward to an update where SBFSpot can handle complex passwords.

I don't believe SBFspot has a problem with complex passwords. I use it in my production environment and have no issues at all.
The password used for Sunny Portal is not the same as for Sunny Explorer or SBFspot.
If you provide me a debuglog (-d5 -v5) I can have a closer look. Send it to sbfspot at skynet dot be if you don't want to post here for security reasons.

To be clear: sbfspot-config might have issues with complex passwords, so you will have to edit the config file when using sbfspot-config

nihonjin98, 2022-02-02:
Here is my debug log if this can help you find why my user pass error is happening:

`SBFspot V3.8.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=192.168.1.133
Password=<undisclosed>
MIS_Enabled=0
Plantname=DanSolar
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=-27.47
Longitude=153.03
Timezone=Australia/Brisbane
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Thu Feb  3 08:14:08 2022: INFO: Starting...
sunrise: 05:24
sunset : 18:38
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 929860710
Inverter IP address: 192.168.1.133 from SBFspot.cfg
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 26 00 10 60 65 09 A0
00000020: FF FF FF FF FF FF 00 00 7D 00
00000030: 66 8C 6C 37 00 00 00 00 00 00
00000040: 01 80 00 02 00 00 00 00 00 00
00000050: 00 00 00 00 00 00 00 00
58 Bytes sent to [192.168.1.133]
select() returned 1
MAX_CommBuf is now 98 bytes
98 bytes received from [192.168.1.133]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 4E 00 10 60 65 13 90
00000020: 7D 00 66 8C 6C 37 00 00 8A 00
00000030: 37 42 FA 7E 00 00 00 00 00 00
00000040: 01 80 01 02 00 00 00 00 00 00
00000050: 00 00 00 00 00 03 00 00 00 FF
00000060: 00 00 3C 74 00 20 01 00 8A 00
00000070: 37 42 FA 7E 00 00 0A 00 0C 00
00000080: 00 00 00 00 00 00 03 00 00 00
00000090: 01 01 00 00 00 00 00 00
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
[... 105 lines cut ...]
78 Bytes sent to [192.168.1.134]
select() returned 1
78 bytes received from [192.168.1.134]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 3A 00 10 60 65 0E D0
00000020: 7D 00 66 8C 6C 37 00 01 8A 00
00000030: B5 52 F9 7E 00 01 01 00 00 00
00000040: 05 80 0D 04 FD FF 07 00 00 00
00000050: 84 03 00 00 31 02 FB 61 00 00
00000060: 00 00 F6 E9 EF E9 F0 E9 F5 E9
00000070: 88 88 88 88 00 00 00 00
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 00 00 10 60 65 0E D0 7D 00 66
00000010: 8C 6C 37 00 01 8A 00 B5 52 F9
00000020: 7E 00 01 01 00 00 00 05 80 0D
00000030: 04 FD FF 07 00 00 00 84 03 00
00000040: 00 31 02 FB 61 00 00 00 00 F6
00000050: E9 EF E9 F0 E9 F5 E9 88 88 88
00000060: 88 00 00 00
<<<=================================>>>
Thu Feb  3 08:14:09 2022: CRITICAL: Logon failed. Check 'USER' Password
Thu Feb  3 08:14:09 2022: INFO: Done.
`

nihonjin98, 2022-03-04:
I have updated my SBFspot to latest version and here is my debug log for latest version now.  

`SBFspot V3.9.3
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
IP_Address=192.168.1.133,192.168.1.134
Password=<undisclosed>
Plantname=DanSolar
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=-27.47
Longitude=153.03
Timezone=Australia/Brisbane
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/pi/smadata/SBFspot.db
End of Config

Fri Mar  4 13:50:06 2022: INFO: Starting...
sunrise: 05:44
sunset : 18:14
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 879421886
Inverter IP address: 192.168.1.133 from SBFspot.cfg
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 26 00 10 60 65 09 A0
00000020: FF FF FF FF FF FF 00 00 7D 00
00000030: BE E9 6A 34 00 00 00 00 00 00
00000040: 01 80 00 02 00 00 00 00 00 00
00000050: 00 00 00 00 00 00 00 00
58 Bytes sent to [192.168.1.133]
select() returned 1
MAX_CommBuf is now 98 bytes
98 bytes received from [192.168.1.133]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 4E 00 10 60 65 13 90
00000020: 7D 00 BE E9 6A 34 00 00 8A 00
00000030: 37 42 FA 7E 00 00 00 00 00 00
00000040: 01 80 01 02 00 00 00 00 00 00
00000050: 00 00 00 00 00 03 00 00 00 FF
00000060: 00 00 3C 74 00 20 01 00 8A 00
00000070: 37 42 FA 7E 00 00 0A 00 0C 00
00000080: 00 00 00 00 00 00 03 00 00 00
00000090: 01 01 00 00 00 00 00 00
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
[... 105 lines cut ...]
78 Bytes sent to [192.168.1.134]
select() returned 1
78 bytes received from [192.168.1.134]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 3A 00 10 60 65 0E D0
00000020: 7D 00 BE E9 6A 34 00 01 8A 00
00000030: B5 52 F9 7E 00 01 01 00 00 00
00000040: 05 80 0D 04 FD FF 07 00 00 00
00000050: 84 03 00 00 6E 8C 21 62 00 00
00000060: 00 00 D9 FD E9 FA FC 02 B9 B9
00000070: 88 88 88 88 00 00 00 00
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 00 00 10 60 65 0E D0 7D 00 BE
00000010: E9 6A 34 00 01 8A 00 B5 52 F9
00000020: 7E 00 01 01 00 00 00 05 80 0D
00000030: 04 FD FF 07 00 00 00 84 03 00
00000040: 00 6E 8C 21 62 00 00 00 00 D9
00000050: FD E9 FA FC 02 B9 B9 88 88 88
00000060: 88 00 00 00
<<<=================================>>>
Fri Mar  4 13:50:06 2022: CRITICAL: Logon failed. Reason unknown (1)
Fri Mar  4 13:50:06 2022: INFO: Done.
`

sbf-, 2022-03-04:
Can you try with one inverter (only one IP address in config)?