SBFspot/SBFspot #438: No MQTT publish after Sunset

MAPEZ1, 2021-02-20:
I’m using SBFspot MQTT towards Homebridge / Homekit / EveApp. 
Works good with SB4.0 to read values to trigger sensors / switches since a while.
Now I added a SB2.5 Storage to get Battery Level and Temperature of Battery.
That works also good during Daytime, but unfortunately the MQTT publishing stops certain time after Sunset. Means I can not trigger a sensor or switch when Battery Level falls below a value
as the last published value (% and C°) is kept.
Calling SBFspot -v -finq -nocsv -nosql still provides the data, but not via MQTT.
Is that intended to be so ? … can it be changed ?

sbf-, 2021-02-21:
No, that is not intended.
-finq -mqtt should work, but I have to check it.

MAPEZ1, 2021-02-21:
Thanks a lot. Pls let me know if I should check anything or provide more info.

wmcs2001, 2021-02-22:
Same issue here as I reinstalled SBFSpot on a Raspberry PI4. Before that it worked perfect in a Docker Container on my Synology NAS. '/usr/local/bin/sbfspot.3/SBFspot -v -finq -mqtt' forces a MQTT update of the data in the broker (and openHAB in my case). But no updates 'during the night'. I didn't have that issue before (on my Synology). Thanks for looking in to it!

MAPEZ1, 2021-02-26:
Sorry newbie question ... but how can I check if sbfspot is started with -finq -mqtt

mwuerr, 2021-02-26:
Hi,
I had the same issue. On my system the solution consisted of two steps:
1. add -finq so that the script /usr/local/bin/sbfspot.3/daydata : 
sudo nano /usr/local/bin/sbfspot.3/daydata
change the line that calls sbfspot to 
/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad1 -am0 -ae1 -mqtt -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg $1 $2 $3 $4 $5 &>>$log
this ensures that the script does something even after sunset.
2. update the crontab entry for this script so that it is also called between 22:00 and 6:00
crontab -e
replace the 6-22 with a *
so that the content is:

## SBFspot
*/5 * * * * /usr/local/bin/sbfspot.3/daydata
55 05 * * * /usr/local/bin/sbfspot.3/monthdata

sbf-, 2021-02-28:
Sure, the daydata crontab has to run 24/7
Another option is to leave daydata script "as is" and add the -finq to the crontab:
```
*/5 * * * * /usr/local/bin/sbfspot.3/daydata -finq
```

MAPEZ1, 2021-03-03:
Thank's a lot both for your help :-)
daydata -fing did it ... happy

closed 2021-03-03 by MAPEZ1