SBFspot/SBFspot #439: Occasional infinite loop in ethRead()

mincequi, 2021-02-22:
I experience an occasional infinite loop situation within ethRead(). This has to do with my EnergyMeter on local network.

`int ethRead(unsigned char *buf, unsigned int bufsize)
{
// ...
	do
	{
// ...
		if (FD_ISSET(sock, &readfds))
			bytes_read = recvfrom(sock, (char*)buf, bufsize, 0, (struct sockaddr *)&addr_in, &addr_in_len);
		else
		{
			if (DEBUG_HIGHEST) puts("Timeout reading socket");
			return -1;
		}
// ...
	} while (bytes_read == 600 || bytes_read == 608); // keep on reading if data received from Energy Meter (600 bytes) or Sunny Home Manager (608 bytes)
}`

The only exit condition of that loop is, if the socket times out OR bytes_read != 600 OR bytes_read != 608.
Image the case, we do not get an answer from the Inverter, but the EnergyMeter continuously broadcasts. Then we would be stuck in that loop.

I can provide a fix for this or discuss a solution.

I see multiple approaches here:
1.  Add another exit condition (some counter or timer).
2. Do not use a "wild" socket, which accepts any host
3. Use a higher level (networking library) which would also help dealing with multiple platforms.

Generally, i would go for 3, because i believe it will solve some other issues as well. However, changes would be rather big and probably something for SBFspot v4.
I can implement and test this, however, i am a little dependent on your will, whether you want to integrate that changes. As you know, i am working on a fork of SBFspot, where i fix most of the issues, i experienced using SBFspot. This already developed pretty far away from your master and i wonder, if you are still willing to merge those changes in your v4 branch.

Boost.Asio would be a great candidate to deal with UDP communication. And to get this all on another level, we could also use Qt. This would solve all the hassle while dealing with SQL, Bluetooth, Networking, MQTT, JSON Parsing, INI file parsing, command line parsing in a cross platform manner.

So, if you agree to merge my fork back, we could discuss about some library integration. Boost.Asio would be a nice abstraction for **UDP networking** and Qt could basically abstract **all** that low-level stuff we are dealing with.

No idea, if you ever had contact with Qt, but i can promise: it is awesome.

mincequi, 2021-02-23:
Related to #50 .

sbf-, 2021-02-25:
> I can implement and test this, however, i am a little dependent on your will, whether you want to integrate that changes. As you know, i am working on a fork of SBFspot, where i fix most of the [...]

ATM I don't have much time, that's why I didn't merge any PR's
If it keeps on working, I agree with your changes. Unfortunately, I can't test all of them.

I heard of QT, but I never used it

closed 2021-12-16 by sbf-