SBFspot/SBFspot #441: logon failed

wuytens, 2021-02-27:
logon failed with sma 3000tl-0 inverter both for user and installer

 ./SBFspot -d5 -v5
SBFspot V3.7.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2020, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:16:14:EB
Password=<undisclosed>
MIS_Enabled=0
Plantname=MyPlant
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=50.8
Longitude=4.33
Timezone=Europe/Brussels
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=15
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Sat Feb 27 11:23:32 2021: INFO: Starting...
sunrise: 07:31
sunset : 18:19
Connecting to 00:80:25:16:14:EB (1/15)
Initializing...
SUSyID: 125 - SessionID: 912411593 (0x36624BC9)
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 EB 14 16 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 EB 14 16 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=01
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00
00000010: EB 14 16 25 80 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
[... 293 lines cut ...]
Received 18 bytes
Received 66 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 54 00 2A EB 14 16 25 80 00
00000010: 65 54 60 EB 27 B8 01 00 7E FF
00000020: 03 60 65 0E 50 7D 5D 00 34 65
00000030: BE 3A 00 01 71 00 44 43 2F 7D
00000040: 5D 00 01 00 01 00 00 04 80 0D
00000050: 04 FD FF 0A 00 00 00 84 03 00
00000060: 00 E0 1D 3A 60 00 00 00 00 EC
00000070: EC EC EC BB BB BB BB BB BB BB
00000080: BB 32 0F 7E
cmd=1
PacketLength=84
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 0E 50 7D 00 34
00000010: 65 BE 3A 00 01 71 00 44 43 2F
00000020: 7D 00 01 00 01 00 00 04 80 0D
00000030: 04 FD FF 0A 00 00 00 84 03 00
00000040: 00 E0 1D 3A 60 00 00 00 00 EC
00000050: EC EC EC BB BB BB BB BB BB BB
00000060: BB 32 0F 7E
<<<=================================>>>
Sat Feb 27 11:24:32 2021: CRITICAL: Logon failed. Check 'INSTALLER' Password

wm20320, 2021-02-28:
Same here but solved for me. 
I don't no if this information is useful, i'm not an expert. Since I updated my SMA Tripower with a new FW Sunny Explorer told me that 0000 and 1111 are no longer valid passwords and I have to change it. After PW change all works fine again.

sbf-, 2021-02-28:
@wuytens Try with SunnyExplorer if possible. This will provide more info in case 0000/1111 aren't valid like @wm20320 said.
It is not possible to change the PW with SBFspot.

sbf-, 2021-03-06:
@wuytens Is this fixed?

wuytens, 2021-03-06:
Yes all working now it was stupid me. With sunny explorer you could login with default password 0000 even if I changed it to something else. So with the correct password it works. Thx a lot

sbf-, 2021-03-06:
Thanks for the feedback. Closing the issue now.

closed 2021-03-06 by sbf-