SBFspot/SBFspot #442: Failed to get current plant time (-1)

pvaberg, 2021-03-02:
Recently, the settime function stopped working. To eliminate the fact that it is a hardware malfunction, I tried it with a Raspberry Pi Box and a notebook. The notebook also runs Windows, where I still can set the the time with Sunny Explorer. I attached the debuglogs for Pi and Notebook. Any Ideas what might be the problem?
debuglog notebook
[https://pastebin.com/cgzb0nmc](url)

debuglog pi
[https://pastebin.com/eFe9v0gF](url)

sbf-, 2021-03-04:
Need to check this, but it sounds promising...
Thanks.

npkamen, 2021-04-26:
I'm getting similar errors.  Any joy in debugging the issue?

david-a, 2021-04-27:
I'm having another issue in this regard (setting time). 
I have a SMA Sunny Boy 4000TL-20 inverter. Recently, I've noticed that in 99% of the cases, the inverter is not accepting the time synchronization. Automatic or manual, from its WebBox, Windows SUNNY EXPLORER, and even using SBFspot software on a RaspberryPi. The sync says its successfully changed the time in the inverter, but when checked again - it's the same as before. The inverter loses it's time sync (about 1 hour gap) suddenly, sometime in the middle of the day.
Why is that? and does the inverter has a limit of how many times its time is syned?
It looks like it's an issue of the inverter, not the SBFspot software, but still, hoping to learn something helpful from you guys

hardisco, 2021-05-04:
Same issue here using SBFspot V3.8.1 (installed over V3.0.3)
Raspbian GNU/Linux 10 (buster) on Raspberry Pi 3 Model B+ Rev 1.3

Logon OK
SetPlantTime()
Timeout reading socket
No data!
Failed to get current plant time (-1)
logoffSMAInverter()

[mydebug.log](https://github.com/SBFspot/SBFspot/files/6418950/mydebug.log)

sbf-, 2021-05-04:
> I'm getting similar errors. Any joy in debugging the issue?

@npkamen 
Difficult to debug as on my system I have no issues...
But as @pvaberg discovered V2.1.0 does it right I made a few changes (See settime branch)
It would be nice if someone can test this.

sbf-, 2021-05-06:
@david-a Have a look at the inverter’s events, maybe there is some error message.

unexxx, 2021-07-26:
I had the same issue --> my inverter was one hour behind on my system time --> corrected it with sunny explorer and all good.

hardisco, 2021-09-13:
> settime branch

I compiled and ran the changes in settime branch.  Sadly nothing changed - same error message.

faleandroid, 2021-12-27:
Has this problem been solved? I'm running the last version, i cannot upload on PVOutput and the only error i get is exactly this one. Does someone know how to solve it?

sbf-, 2021-12-27:
> Has this problem been solved? I'm running the last version, i cannot upload on PVOutput and the only error i get is exactly this one. Does someone know how to solve it?

I was unable to fix this issue until now. The only solution is to use sunnyexplorer (windows only)

macamic2020, 2021-12-27:
I am pvaberg, had to get a new account
Another workaround under linux running V2.1.0 and the v3.x parallel via cron 
V2.1.0 it sets the proper time and than it works as intended
22	6-9	*	*	*	/home/pi/bin/smaspot
and 
V3.x
after 9 
*/5 9-22 * * * /usr/local/bin/sbfspot.3/daydata
not pretty but it works 
The problem seems to be the 3600 s time difference limit introduced in V3.x. If the time of the inverter in the morning is off by 3601 s,  sbfspot.3 won't work.

sbf-, 2021-12-29:
> The problem seems to be the 3600 s time difference limit introduced in V3.x. If the time of the inverter is off by 3601 s, sbfspot.3 won't work.

To my knowledge, 3600 is the timezone difference in seconds (UTC <-> Local Time)
The 3601 is the same, but the 1 is indicating DLS (daylight saving) is ON

sbf-, 2021-12-29:
If someone is willing to test the latest changes (settime branch):
I'm using the old code of V2.1.0 without getting the current inverter time. Hope this works.

```
git clone -b settime https://github.com/SBFspot/SBFspot.git
cd SBFspot/SBFspot
make sqlite
```

sstidl, 2022-11-27:
> If someone is willing to test the latest changes (settime branch): I'm using the old code of V2.1.0 without getting the current inverter time. Hope this works.
> 
> ```
> git clone -b settime https://github.com/SBFspot/SBFspot.git
> cd SBFspot/SBFspot
> make sqlite
> ```

I'm willing to help but would need a compiled binary for raspberry pi w rev1.
Armhf architecture

sstidl, 2022-11-27:
> I'm willing to help but would need a compiled binary for raspberry pi w rev1. Armhf architecture

Never mind, I managed to compile the code and it worked. Time jump of 5 years and now my inverter is in 2022. :-D

Thank you so much for maintaining this excellent example of the power of open source and it's community!

sbf-, 2022-11-27:
> Never mind, I managed to compile the code and it worked. Time jump of 5 years and now my inverter is in 2022. :-D

Thanks, now I can implement this in the latest version.

lampra1, 2023-02-03:
@sbf- Is there a release with this fix yet? If not, could you please provide a windows sqlite files with this fix for testing?
I have two inverters that switch off during the night and the next day, sometimes the time gets set and sometimes not. When time is not correct, nothing is uploaded to pvoutput

bundy67, 2024-01-15:
Hello, hoping someone can help . We had a power outage and when that happens my SMA 3000TL loses the system time. I normally use Sunny Explorer to reset this and I tried the same and whilst Sunny Explorer says the time is updated, when I connect via SBFSpot it reports a system time from 15 years ago!   Result is since this power outage I can no longer update to PV Output.  

As per instructions here I run 

**/usr/local/bin/sbfspot.3/SBFspot -v -nosql -nocsv -finq | grep -i time**

With the resultant output 

**Failed to get current plant time (-1)
SetPlantTime returned an error: -1
	Operation Time: 43177.21h
	Feed-In Time  : 42487.26h
Current Inverter Time: 03/01/2008 03:55:17
Inverter Wake-Up Time: 02/01/2008 17:42:34
Inverter Sleep Time  : 03/01/2008 03:55:21
startTime: 15/01/2024 00:00:00
startTime: 01/01/2024 12:00:00**

pi@raspberrypi:~ $ 

Please can anyone provide some suggestions?

sbf-, 2024-01-15:
Can yo try [this](https://github.com/SBFspot/SBFspot/issues/442#issuecomment-1328192012)?

lampra1, 2024-01-15:
Sometimes, the `settime` argument fails due to [large ](https://github.com/SBFspot/SBFspot/blob/b789d7f30aa2b1afb6ee5076c96dd0506c5ea4b1/SBFspot/SBFspot.cfg#L97)time difference between the inverter and the pc.
There is also a possibility, for older inverters, that there is a hardware issue and the inverter losses the time, hence the large time difference mentioned above. In this case, you need to come up with some kind of workaround. 
As I had the same issue, i.e inverter date returning to the year 2001, you might want to test the [workaround](https://github.com/SBFspot/SBFspot/discussions/624) that works for me.

bundy67, 2024-01-16:
> Can you try [this](https://github.com/SBFspot/SBFspot/issues/442#issuecomment-1328192012)?

Thanks for the suggesitons. Gave above a go but recieved this error 

pi@raspberrypi:~**/SBFspot/SBFspot $ make sqlite
test -d sqlite/obj/ || mkdir -p sqlite/obj/
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/boost_ext.o 
In file included from boost_ext.cpp:35:0:
boost_ext.h:47:29: fatal error: boost/version.hpp: No such file or directory
 #include <boost/version.hpp>
                             ^
compilation terminated.
makefile:116: recipe for target 'sqlite/obj/boost_ext.o' failed
make: *** [sqlite/obj/boost_ext.o] Error 1**

Thanks in advance for any advice or pointers for what next

sbf-, 2024-01-16:
> boost_ext.h:47:29: fatal error: boost/version.hpp: No such file or directory

Sorry, it's a bit more complicated ;-)
You should follow the [manual installation wiki](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)
Don't download/untar source code, but clone the 'settime' branch instead:
`git clone -b settime https://github.com/SBFspot/SBFspot.git`

bundy67, 2024-01-20:
Hi all, well I seem to have found a solution  (purely by chance) 

Fiddling through Sunny Explorer I went and made an edit under Plant Settings and time. In the Drop down for "time" and "date" I noticed for Time format I had multiple choices HH:mm (1) OR HH:mm (0) 

Whlst they looked the same, I didnt understand why one had a (1) in brackets and the other had (0) in brackets. 

So I changed it to the option that said (1) and then (as I had tried before and failed) I agian tried to update the Plant Time to match my PC time. 

I then ran  -  /usr/local/bin/sbfspot.3/SBFspot -settime

and instead of getting the dreaded "Failed to get current plant time (-1)"  I now see my plant time has been updated to 2024 

This time it worked !!!

As soon as that Plant time updated I immediately started seeing updates on PVOutput again 

thanks all for your help. Maybe this little "fidge" will help someone else one day.

sbf-, 2024-01-20:
> Fiddling through Sunny Explorer I went and made an edit under Plant Settings and time. In the Drop down for "time" and "date" I noticed for Time format I had multiple choices HH:mm (1) OR HH:mm (0)

I verified it and the settings are different: (uppercase) HH:mm for 24h time and (lowercase) hh:mm for 12h time
In my case, the (1) is showing the current selected value. Probably something went wrong due to the power failure at your side.

Thanks for reporting this.

lampra1, 2024-01-20:
Sadly, in my case with multiple inverters, with (1) showing as the current selected value, two of the inverters do loose the time.

rverho, 2024-01-28:
Hi, 

have same issue as others and managed to build [this](https://github.com/SBFspot/SBFspot/issues/442#issuecomment-1328192012?) with warning : 
`g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/ArchData.o
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/SBFspot.o
SBFspot.cpp: In function ‘E_SBFSPOT SetPlantTime(time_t, time_t, time_t)’:
SBFspot.cpp:1017:34: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 2 has type ‘int32_t’ {aka ‘int’} [-Wformat=]
 1017 |         printf("TZ offset (s): %ld\n", tzOffset);
      |                                ~~^     ~~~~~~~~
      |                                  |     |
      |                                  |     int32_t {aka int}
      |                                  long int
      |                                %d
SBFspot.cpp: In function ‘E_SBFSPOT ethInitConnection(InverterData**, const char*)’:
SBFspot.cpp:396:54: warning: ‘%s’ directive writing between 8 and 2147483645 bytes into a region of size 20 [-Wformat-overflow=]
  396 |             sprintf(inverters[devcount]->IPAddress, "%s", IP_Address);
      |                                                      ^~
SBFspot.cpp:396:20: note: ‘sprintf’ output between 9 and 2147483646 bytes into a destination of size 20
  396 |             sprintf(inverters[devcount]->IPAddress, "%s", IP_Address);
      |             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/TagDefs.o
g++ Bluetooth.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/Bluetooth.o
g++ mqtt.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/mqtt.o
g++ db_SQLite.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/db_SQLite.o
`
before running updated code 
Current Inverter Time: 09/02/2001 12:15:02
Inverter Wake-Up Time: 03/02/2001 07:06:45
Inverter Sleep Time  : 03/02/2001 07:06:48

and after (original SBFspot is in normal directory /usr/local/bin/sbfspot.3/SBFspot and this version in ~/test_sma_settime/SBFspot/SBFspot/sqlite/bin )

pi@raspberrypi:~/test_sma_settime/SBFspot/SBFspot/sqlite/bin $ /usr/local/bin/sbfspot.3/SBFspot -v -loadlive -finq | grep -i time
getOperationTime returned an error: 21
Current Inverter Time: 15:30
Inverter Wake-Up Time: 07:06
Inverter Sleep Time  : 07:06
startTime: 28/01/2024 00:00:00
pi@raspberrypi:~/test_sma_settime/SBFspot/SBFspot/sqlite/bin $ ./SBFspot -v -loadlive -finq | grep -i time
Reading config '/home/pi/test_sma_settime/SBFspot/SBFspot/sqlite/bin/SBFspot.cfg'
startTime = 65B57CE0 -> 28/01/2024 00:00:00
        Operation Time: 0.00h
        Feed-In Time  : 0.00h
Current Inverter Time: 15:30
Inverter Wake-Up Time: 07:06
Inverter Sleep Time  : 07:06
startTime = 65B57CE0 -> 28/01/2024 00:00:00

investigation continues ...

rverho, 2024-01-28:
settime branch works : 
Current Inverter Time: 28/01/2024 15:45:43
Inverter Wake-Up Time: 03/02/2001 07:06:45
Inverter Sleep Time  : 03/02/2001 07:06:48

commit 242f98d0 (2024-02-25): Fix #442 Failed to get current plant time
closed 2024-02-25 by sbf-