SBFspot/SBFspot #443: How to get MQTT working?

supermarcie, 2021-03-05:
Hi
I don't find how to get MQTT working.
I configured everything for MQTT, but nothing happens
it looks like it is an extension, but were to find the code?

can someone help me

supermarcie, 2021-03-05:
I'm bit further.
I installed the moquitto client.
I already have a moqsuitto broker running 

I now get following error:
Error: Invalid port given: 0

I defined the port, also put the port in comment but no difference still get the error

can someone help me?

supermarcie, 2021-03-05:
it's working but some difficulties to retrieve the MQTT in Home assistant.
Does anyone has this working?

sbf-, 2021-03-06:
This is more a question for HA forum

supermarcie, 2021-03-08:
Found out everything, can be closed

closed 2021-03-08 by supermarcie

Twoink, 2021-10-04:
> Found out everything, can be closed

Would be nice if you share the solution. I'm facing the same issues at the moment.
Is it possible to check a mqtt log to figure out what is happening?