SBFspot/SBFspot #444: EToday of current day not written to database

mahadi, 2021-03-06:
When i use `/usr/local/bin/sbfspot.3/SBFspot -v3 -ad1 -am0 -ae0 -finq -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg` the total yield of the current day is not inserted into the database. 

I would like to see the current state of the day in my personal visualization which queries the SBFSpot database.

I guess it is related to the implementation? Does SBFspot handle updating of existing data or does it only insert into the database?
Or what is the reason that it is not inserted in the database?

sbf-, 2021-03-06:
See #290 
I am working on a fix

sbf-, 2021-03-21:
Fixed in V3.8.0

closed 2021-03-21 by sbf-

mahadi, 2021-07-03:
@sbf-: I think this issue is not yet solved, just updated to V3.8.2. Existing values in the database are not updated, so still the current value (EToday) is not present... Is it true that updating is not supported?

sbf-, 2021-07-03:
What do you mean with "existing values"?
In which table you're looking?

mahadi, 2021-07-05:
I am querying the MonthData table and as set in the cron job, it gets updated only once a day. So this means the energey produced at the current can not be read from this table.

[image]

sbf-, 2021-07-05:
Monthdata doesn't have EToday until the day is complete. So, July 5th is inserted the next day.
What you need is the Inverters table. There you have the updated EToday for each inverter.