SBFspot/SBFspot #446: Cannot connect to inverter via bluetooth

rucksman, 2021-03-14:
Im am running SBFspot currently  via Speedwire (IP_Address in cfg-file). This works fine. No I have to change to bluetooth, but this does not work at all. Running "hcitool scan" does find only 2 SMA bluetooth sockets but not the inverter. Is there any setting in the inverter or elsewhere which prevents hcitool to detect the inverter?

Same problem (not finding the inverter via bluetooth) does appear with SMA tool "Sunny Explorer". Again only the bluetooth sockets are found. In the SMA web portal "www.sunnyportal.com" I can see the inverter and the sockets.

My devices are:
- SB-3600SE
- 2 SMA bluetooth sockets
- Home Manager (first generation)
- Energy Meter (first generation)

Is the still active speedwire SBFspot interfering? Or is there another problem I did not take into account? Thanks for helping!

sbf-, 2021-03-15:
Pretty sure your Smart Energy inverter doesn't have BT interface.
From the brochure:

> Interface: Speedwire / Webconnect

I don't see Bluetooth communication interface

rucksman, 2021-03-23:
You are right. There is no Bluetooth interface. I was very sure that there is one because of the bluetooth sockets, but it seems as if these are only communicating with the Homemanager and not the inverter.

closed 2021-03-23 by rucksman