SBFspot/SBFspot #447: Unknown Inverter Type

Trinitus, 2021-03-19:
Report this issue at https://github.com/SBFspot/SBFspot/issues with following info:
0x00002482 and Inverter Type=<SMA Sunny Tripower 6.0 - STP6.0-3AV-40> (e.g. SB1300TL-10)
SUSyID: 378 - SN: 3005912810
Device Name:      STP6.0-3AV-40 810
Device Class:     Solar Inverters
Device Type:      UNKNOWN TYPE
Software Version: 03.00.08.R
Serial number:    3005912810
SUSyID: 378 - SN: 3005912810
Device Status:      Ok
SUSyID: 378 - SN: 3005912810
Device Temperature: 0.0°C
SUSyID: 378 - SN: 3005912810
GridRelay Status:      Closed
SUSyID: 378 - SN: 3005912810
Pac max phase 1: 6000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 378 - SN: 3005912810

sbf-, 2021-03-19:
What version of SBFspot are you using? An old one I presume.

Trinitus, 2021-03-19:
Yeah, just noticed that I have downloaded V3.5.0...

closed 2021-03-19 by Trinitus