SBFspot/SBFspot #448: Error:sqlite3_exec() returned: "UNIQUE constraint failed ........."

ufankhau, 2021-03-20:
Having installed the latest version 3.8.0 on a Raspberry Pi 4 model B V1.1 the following error message appears when "testing" SBFspot with command /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv ... looks like the primary_key in the database table is not correctly set ??

Error: sqlite3_exec() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1616229654,2110618169,1236,562,1.922,1.142,643.38,492.28,579,573,580,2.544,2.559,2.57,227.76,224.06,226.14,2341,69520324,50.01,22617.3,21991.2,0,'OK','Closed',52.66)
[2021-03-20 09:40:54.452] Error: [spot_data]exec_query() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1616229654,2110618169,1236,562,1.922,1.142,643.38,492.28,579,573,580,2.544,2.559,2.57,227.76,224.06,226.14,2341,69520324,50.01,22617.3,21991.2,0,'OK','Closed',52.66)

sbf-, 2021-03-21:
The combination of time (1616229654 - Sat, 20 Mar 2021 08:40:54 GMT) and serial number (2110618169) is already in the database

aoverweel, 2021-03-23:
I have exact the same issue on a PI4 with v3.8.0. 
Inverter Sleep Time  : 23/03/2021 10:24:00
[2021-03-23 10:24:00.453] Error: sqlite3_exec() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1616491440,1992031704,476,0,0.98,0,482.1,0,0,0,0,1.77,0,0,230.82,0,0,795,6506037,50.02,8020.51,7965.19,0,'OK','Closed',                                                          0)
[2021-03-23 10:24:00.453] Error: [spot_data]exec_query() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1616491440,1992031704,476,0,0.98,0,482.1,0,0,0,0,1.77,0,0,230.82,0,0,795,6506037,50.02,8020.51,7965.19,0,'OK','Closed',                                                          0)

sbf-, 2021-03-23:
Strange... never seen this before.
What is the query interval? Every 5 min or 1 sec?

sbf-, 2021-03-24:
@mincequi This bug is introduced by PR #425 

Spotdata is exported twice in `Inverter.cpp`

```
    // Open DB
#if defined(USE_SQLITE) || defined(USE_MYSQL)
    if (!m_config.nosql)
    {
#if defined(USE_MYSQL)
        m_db.open(m_config.sqlHostname, m_config.sqlUsername, m_config.sqlUserPassword, m_config.sqlDatabase, m_config.sqlPort);
#elif defined(USE_SQLITE)
        m_db.open(m_config.sqlDatabase);
#endif
        if (m_db.isopen())
        {
            time_t spottime = time(NULL);
            m_db.type_label(m_inverters);
            m_db.device_status(m_inverters, spottime);
            m_db.spot_data(m_inverters, spottime);
            if (hasBatteryDevice)
                m_db.battery_data(m_inverters, spottime);
        }
    }
#endif

    exportSpotData();
```

mincequi, 2021-03-24:
Ok, true. This might be introduced due to merging....

Can you checkout and try df58ff29828ff9ac7b107ae2630f41fd89168841 from my fork? This should fix the issue. Then let's see if we can integrate that one.

brevilo, 2021-04-09:
Wouldn't it be easier to fix the bug first? Removing `exportSpotData()` from the snippet above should do the trick, shouldn't it?

Cheers

sbf-, 2021-04-09:
@brevilo it is fixed, but because it's not a blocking issue I didn't commit it yet.

`exportSpotData()` is OK

The fix is
``` c++
    // Open DB
#if defined(USE_SQLITE) || defined(USE_MYSQL)
    if (!m_config.nosql)
    {
#if defined(USE_MYSQL)
        m_db.open(m_config.sqlHostname, m_config.sqlUsername, m_config.sqlUserPassword, m_config.sqlDatabase, m_config.sqlPort);
#elif defined(USE_SQLITE)
        m_db.open(m_config.sqlDatabase);
#endif
/* Fix #448
        if (m_db.isopen())
        {
            time_t spottime = time(NULL);
            m_db.type_label(m_inverters);
            m_db.device_status(m_inverters, spottime);
            m_db.spot_data(m_inverters, spottime);
            if (hasBatteryDevice)
                m_db.battery_data(m_inverters, spottime);
        }
*/
    }
#endif

    exportSpotData();

```

brevilo, 2021-04-10:
Yep, just figured that after looking more closely at the code  👍 

Just wanted to argue that rather large refactorings should probably be dealt with separately from bug fixes 😇

commit dcb41acf (2021-04-13): Fix #448 Error:sqlite3_exec() returned: "UNIQUE constraint failed"
closed 2021-04-13 by sbf-