SBFspot/SBFspot #449: SBFspot for MacOS (OSX)

sbf-, 2021-03-30:
I wonder if someone already tried to compile and run SBFspot on MacOS

mincequi, 2021-03-30:
Sure ;)

Works on my fork.

sbf-, 2021-03-31:
Nice... but without bluetooth nor upload daemon?
Having zero experience, what are the steps to take?

mincequi, 2021-03-31:
True, without Bluetooth nor upload.
From my perspective, the least effort is to port this app to Qt and create a single executable for polling the inverter **and** doing the upload.

I already integrated the reading of an energy meter using Qt and the next steps are:
- port the ethernet stuff to Qt
- then port bluetooth
- port SQL access
- using QtService to realize cross plattform service/daemon.

This should be enough for a proper cross-platform solution. Any help is highly appreciated :)

closed 2021-05-20 by sbf-