SBFspot/SBFspot #451: Failed to upload - Couldn't connect to server

CHH25, 2021-04-14:
I have used SBFspot for some months, and it has worked fine. However a few days ago I recognized that data from one of my inverters has never been uploaded to PVOutput. I tried to solve this problem by some reconfigurations and updates, but now I cannot upload any data to PVoutput. Reason: ERROR: addBatchStatus() returned Couldn't connect to server.

I have two inverters, sharing NetID=2 (MIS_Enabled=1). I have installed SBFspot on Windows 10. It seems as if SBFspot works ok, as I can see all expected data in SFBspot.db. But SBFspotUpload fails, except for a few times, just after I have restarted the services.

When data was uploaded to PVoutput successfully SBFspot version 3.7.0 was installed. 

In order to upload data from both inverters I tried several solutions:
* I changed PVoutput_SID=SerialNmbrInverter_1:PVoutput_System_ID_1,SerialNmbrInverter_2:PVoutput_System_ID_1 (note that I have used PVoutput_System_ID_1 for both inverters)
* I installed SBFspot 3.7.1
* I created a new SBFspot.db
* I reinstalled the the same setup (Version 3.7.0), as was able to upload data before.
* I have been through all troubleshooting tips in the Wiki document: No uploads to PVOutput.org
* I have restarted SBFspot programs several times and also rebooted/restarted my computer several times
* I can ping pvoutput.org

Below I have attached an example of the log file. Please notice that sometimes a few datapoints are uploaded. Please also notice that datapoints from both inverters have been uploaded, but not in the same sequence.

[SBFspotUpload20210413.log](https://github.com/SBFspot/SBFspot/files/6310393/SBFspotUpload20210413.log)

sbf-, 2021-04-14:
Looks like a network error...

Try this in your browser (replace APIKEY and SYSTEMID with yours):
`https://pvoutput.org/service/r2/getsystem.jsp?key=APIKEY&sid=SYSTEMID`

You should get some PVO system info

Install [curl for windows](https://curl.se/windows/) and enter following command in cmd.exe box
`curl -v -H "X-Rate-Limit:1" -H "X-Pvoutput-Apikey:APIKEY" -H "X-Pvoutput-SystemId:SYSTEMID" https://pvoutput.org/service/r2/getsystem.jsp?teams=1`

CHH25, 2021-04-14:
I received the correct PVO system info.

Here folllows the curl command:

C:\Users\Public\SMAdata>curl -v -H "X-Rate-Limit:1" -H "X-Pvoutput-Apikey:APIKEY" -H "X-Pvoutput-SystemId:SYSTEMID" https://pvoutput.org/service/r2/getsystem.jsp?teams=1
*   Trying 45.56.66.169...
* TCP_NODELAY set
* Connected to pvoutput.org (45.56.66.169) port 443 (#0)
* schannel: SSL/TLS connection with pvoutput.org port 443 (step 1/3)
* schannel: checking server certificate revocation
* schannel: sending initial handshake data: sending 183 bytes...
* schannel: sent initial handshake data: sent 183 bytes
* schannel: SSL/TLS connection with pvoutput.org port 443 (step 2/3)
* schannel: failed to receive handshake, need more data
* schannel: SSL/TLS connection with pvoutput.org port 443 (step 2/3)
* schannel: encrypted data got 2898
* schannel: encrypted data buffer: offset 2898 length 4096
* schannel: sending next handshake data: sending 126 bytes...
* schannel: SSL/TLS connection with pvoutput.org port 443 (step 2/3)
* schannel: encrypted data got 242
* schannel: encrypted data buffer: offset 242 length 4096
* schannel: SSL/TLS handshake complete
* schannel: SSL/TLS connection with pvoutput.org port 443 (step 3/3)
* schannel: stored credential handle in session cache
> GET /service/r2/getsystem.jsp?teams=1 HTTP/1.1
> Host: pvoutput.org
> User-Agent: curl/7.55.1
> Accept: */*
> X-Rate-Limit:1
> X-Pvoutput-Apikey:APIKEY
> X-Pvoutput-SystemId:SYSTEMID
>
* schannel: client wants to read 102400 bytes
* schannel: encdata_buffer resized 103424
* schannel: encrypted data buffer: offset 0 length 103424
* schannel: encrypted data got 581
* schannel: encrypted data buffer: offset 581 length 103424
* schannel: decrypted data length: 521
* schannel: decrypted data added: 521
* schannel: decrypted data cached: offset 521 length 102400
* schannel: encrypted data length: 31
* schannel: encrypted data cached: offset 31 length 103424
* schannel: server closed the connection
* schannel: schannel_recv cleanup
* schannel: decrypted data returned 521
* schannel: decrypted data buffer: offset 0 length 102400
< HTTP/1.1 200 OK
< Date: Wed, 14 Apr 2021 16:17:26 GMT
< Set-Cookie: JSESSIONID=1u4ng51kl30p0gpxysqhonvx9;Path=/; Secure
< Expires: Thu, 01 Jan 1970 00:00:00 GMT
< X-Rate-Limit-Remaining: 58
< X-Rate-Limit-Limit: 60
< X-Rate-Limit-Reset: 1618419600
< Content-Type: text/plain;charset=UTF-8
< Vary: Accept-Encoding, User-Agent
< Connection: close
< Server: Jetty(7.6.17.v20150415)
< Strict-Transport-Security: max-age=15768000
<
Skin59,7105,2730,29,245,Aleo,2,6000,SMA Sunny Boy 3000,SE,20.0,Medium,20121024,55.747258,12.411647,5;;;0* schannel: client wants to read 102400 bytes
* schannel: server indicated shutdown in a prior call
* schannel: schannel_recv cleanup
* Closing connection 0
* schannel: shutting down SSL/TLS connection with pvoutput.org port 443
* schannel: clear security context handle

sbf-, 2021-04-14:
OK, and does it also work with `http://` instead op `https://` ?

CHH25, 2021-04-14:
Yes. Here is the result:

```
C:\Users\Public\SMAdata>curl -v -H "X-Rate-Limit:1" -H "X-Pvoutput-Apikey:APIKEY" -H "X-Pvoutput-SystemId:SYSTEMID" http://pvoutput.org/service/r2/getsystem.jsp?teams=1
*   Trying 45.56.66.169...
* TCP_NODELAY set
* Connected to pvoutput.org (45.56.66.169) port 80 (#0)
> GET /service/r2/getsystem.jsp?teams=1 HTTP/1.1
> Host: pvoutput.org
> User-Agent: curl/7.55.1
> Accept: */*
> X-Rate-Limit:1
> X-Pvoutput-Apikey:APIKEY
> X-Pvoutput-SystemId:SYSTEMID
>
< HTTP/1.1 200 OK
< Date: Wed, 14 Apr 2021 16:57:59 GMT
< Set-Cookie: JSESSIONID=o80g6r3u4qic1qi0nd300zu37;Path=/
< Expires: Thu, 01 Jan 1970 00:00:00 GMT
< X-Rate-Limit-Remaining: 54
< X-Rate-Limit-Limit: 60
< X-Rate-Limit-Reset: 1618419600
< Content-Type: text/plain;charset=UTF-8
< Vary: Accept-Encoding, User-Agent
< Connection: close
< Server: Jetty(7.6.17.v20150415)
<
Skin59,7105,2730,29,245,Aleo,2,6000,SMA Sunny Boy 3000,SE,20.0,Medium,20121024,55.747258,12.411647,5;;;0* Closing connection 0
```

CHH25, 2021-04-15:
I am not familiar with curl, but I think that the two tests indicate that the connection to PVOutput.org is ok.

However, SBFspotUpload still cannot upload any data to PVOutput.org (Errors: "Couldn't resolve host name" or "Couldn't connect to server").

I can manually upload data using PVOutput's webservice 'Load Live Outputs', but of course I want to let SBFspotUpload do it automatically.

I would appreciate any ideas, how to get SBFspotUpload working again.

sbf-, 2021-04-16:
Yes, it looks like the connection is OK
The only difference is while testing you're running as "user" and the SBFspotUpload service runs as "Local Service"
Maybe some firewall rules come in between here?

sbf-, 2021-04-16:
I tested V3.8.0 again and I have no issues at all - don't know what's going wrong on your system
```
[21:38:03] INFO: Starting ServiceWorkerThread()
[21:38:03] INFO: Uploading 30 datapoints, starting with 20210416,07:10,42275670,0 => OK (200)
[21:39:31] INFO: Uploading 30 datapoints, starting with 20210416,09:40,42276051,240 => OK (200)
[21:40:31] INFO: Uploading 30 datapoints, starting with 20210416,12:10,42278587,2592 => OK (200)
[21:41:31] INFO: Uploading 30 datapoints, starting with 20210416,14:40,42284225,1584 => OK (200)
```

CHH25, 2021-04-16:
Yes, you are right. The problem is my firewall. If I turn off the firewall, then SBFspotUpload starts sending datapoints to PVOutput. If I turn it on, the same error occurs.

I have looked though the firewall log and have found the reason. The only warning regarding this issue was that when I downloaded SBFspot version 3.7.1, SBFspotUploadService.exe was quarantined and blocked. I have now allowed this file, and data is again uploaded to PVOutput.

However, my initial issue was that SBFspot did only upload data from one of my two inverters. This is still the case. Do you have any suggestion to solve that problem?

sbf-, 2021-04-17:
Please post sbfspotupload.cfg without the api key

CHH25, 2021-04-17:
Here is the sbfspotupload.cfg without the api key.

```
################################################################################
#  SBFspotUpload.cfg - Configuration file for SBFspotUploadService/Daemon
#  (c)2012-2018, SBF (https://github.com/SBFspot/SBFspot)
#
#  DISCLAIMER:
#  A user of SBFspotUploadService/Daemon software acknowledges that he or she is
#  receiving this software on an "as is" basis and the user is not relying on
#  the accuracy or functionality of the software for any purpose. The user
#  further acknowledges that any use of this software will be at his own risk 
#  and the copyright owner accepts no responsibility whatsoever arising from
#  the use or application of the software. 
#
################################################################################

################################
### Log Settings             ###
################################
# Windows: C:\Users\Public\SMAdata\Logs
# Linux  : /home/pi/smadata/logs
LogDir=C:\Users\Public\SMAdata\Logs
#LogDir=C:\Users\Public\SMAdata\Logs

#LogLevel=debug|info|warning|error (default info)
LogLevel=info

################################
### PVoutput Upload Settings ###
################################
#PVoutput_SID
#Map inverters to PVoutput System ID's
#PVoutput_SID=SerialNmbrInverter_1:PVoutput_System_ID_1,SerialNmbrInverter_2:PVoutput_System_ID_2
#e.g. PVoutput_SID=200212345:4321
PVoutput_SID=2130088778:14988

#PVoutput_Key
#Sets PVoutput API Key
PVoutput_Key=APIKEY

################################
### SQL DB Settings          ###
################################
# SQL_Database (Fullpath to SQLite DB)
# Windows: C:\Users\Public\SMAdata\SBFspot.db
# Linux  : /home/pi/smadata/SBFspot.db
SQL_Database=C:\Users\Public\SMAdata\SBFspot.db
#SQL_Database=/home/pi/smadata/SBFspot.db

# Reserved for MySQL
#SQL_Database=SBFspot
#SQL_Hostname=<Network Name> or <IP-address>
#SQL_Username=SBFspotUser
#SQL_Password=SBFspotPassword
```

sbf-, 2021-04-17:
And where is your second system defined?
`PVoutput_SID=2130088778:14988`

Multiple system syntax is:
`PVoutput_SID=SerialNmbrInverter_1:PVoutput_System_ID_1,SerialNmbrInverter_2:PVoutput_System_ID_2`

CHH25, 2021-04-17:
I have also tried with this configuration (actually running now), but still only data from one inverter:
`PVoutput_SID=2130088778:14988,2140005747:14988`

I only have one SYSID at PVOutput (14988). Do I need two SYSID's?

sbf-, 2021-04-17:
Yes, the last one will overwrite the data of the first when using only one SID

CHH25, 2021-04-17:
Just for clarification.
Setting MIS_Enabled=1 is only used by SBFspot, when data is retrieved from the system for updating SBFspot.db.
When SBFspotUploader uploads data to PVOutput, each inverter is regarded as a system, even though I have only one plant/system consisting of two inverters.
It is not possible in SBFspot to add the output from each of the merged inverters into a total output for the plant?

sbf-, 2021-04-17:
Yes, that is possible with a modification of the pvo view in the db. But I don´t recommend it as it is more difficult to detect when a system is performing badly

CHH25, 2021-04-17:
The main reason why I want to merge data from my two inverters is that I formerly (from 2012) have used PV BeanCounter, which has done that before upload to PVOutput. 
However, I will give your recommendations a try.

sbf-, 2021-04-17:
The way to go is [PVO Data Aggregation](https://pvoutput.org/help/data_aggregation.html) - however this needs an active donation account

closed 2021-04-18 by sbf-

CHH25, 2021-04-19:
Thanks for helping me.
Now I use PVO Data Aggregation, and I have reloaded correct data for the last 90 days.