SBFspot/SBFspot #452: Error regarding sqlite3 Raspi Zero W (No PVoutput and domoticz)

Slappyjoez, 2021-04-16:
I'm having difficulty with SBFspot on my Raspberry Pi Zero W. I have Domoticz running with a connection to my P1 DSMR port, this works great. Only I am encountering errors when running SBFspot.

(Useful) info:
- Raspberry Pi Zero W
- Inverter: SMA  SB4.0-1AV-41 303
- I have full access to user and installer account on inverter
- WLAN access does not seem to be the problem
- Inverter IP 192.168.68.105 (accessible through browser)
- PVOutput API and SID are correct (double checked)

I have installed SBFspot using the command:
`curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash`

Went through the entire setup, it is generating daily and monthly CSV files. SBFspot.db contains the correct database structure. The SpotData table contains records that seem to correspond with the inverter information/readings.

However I can't get it to upload data to PVOutput and it won't show current information in Domoticz. The only errors I have encountered when running certain commands are these (I've attached the full logfile to this issue): 

`[2021-04-15 20:28:10.662] Error: sqlite3_exec() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1618511290,3006734303,0,0,0,0,427.16,0,0,0,0,0,0,0,236.28,0,0,13812,364800,50,264.798,251.719,0,'OK','Open',0)
[2021-04-15 20:28:10.671] Error: [spot_data]exec_query() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1618511290,3006734303,0,0,0,0,427.16,0,0,0,0,0,0,0,236.28,0,0,13812,364800,50,264.798,251.719,0,'OK','Open',0)`

I would like to know what steps I can take to figure this out. One other thing I have noticed is that I can't edit the SBFspot.config file (permission denied). Is that standard or is this a user access problem?

[SBFSpot logfile.txt](https://github.com/SBFspot/SBFspot/files/6323747/SBFSpot.logfile.txt)

Slappyjoez, 2021-04-16:
I fixed the PVoutput. I checked the /var/log/sbfspot.3 files and encountered issue #313. This has now been resolved, I had 2 entries in the DB with power readings exceeding the PV plant total power. It has started uploading historical readings but is currently blocking further progress due to the fact it has exceeded the 60 requests per hour. It looks like it will keep pushing the old data until it is done, so it will probably be resolved tomorrow. 

Only issue still remaining is the fact that the SBFspot Domoticz device is now showing data in the log for the current month, but not the current power or 24h graph. What could be causing this? Or will this be fixed once the job for uploading to PVoutput has been finished?

sbf-, 2021-04-16:
I think this is an issue for Domoticz support.

Slappyjoez, 2021-04-17:
I've created a post on the Domoticz forum for further investigation. [Link](https://www.domoticz.com/forum/viewtopic.php?f=14&t=36061)

closed 2021-04-17 by Slappyjoez