SBFspot/SBFspot #457: Serial number too big for defined field

diedvdyk, 2021-05-14:
My serial number is 3xxxxxxxxxxxx . 
I had to alter in all the tables the definition for serial from "Serial int(4) NOT NULL" to  "Serial bigint(4) NOT NULL" because I got errors that my serial number is too big.
I am using Mariadb and I am on 3.7.1

stripwax, 2021-05-14:
Hm thinking about the actual usecase here, it would probably be better to
have just a small uniqueID int in the tables, and then just one other table
that maps from uniqueID to actual serial number.
The views could the do that via join, but the underlying tables wouldn't
all need to be changed (unless you really owned more than 4 billion
inverters - unlikely!)
Maybe that's how this should be done.  And then the type of the (real)
Serial could be whatever you want (alphanumeric, if necessary) and only one
table would need to be changed.

sbf-, 2021-05-14:
See #330

stripwax, 2021-05-14:
Ah indeed, and in particular https://github.com/SBFspot/SBFspot/issues/387
(which is marked Closed due to #330)

commit 4abc6f6d (2021-05-22): Fix #330 : From Sqlite3 to MySQL
Fix #457 : Serial number too big for defined field
closed 2021-05-22 by sbf-