SBFspot/SBFspot #459: Incorrect Calculated EToday

mrombouts, 2021-05-19:
Hello, 

SBFSPOT 3.8.1 on Raspberry PI 4

OS: 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Connecting to 2 SMA inverters:
Inverter 1: SB 3300TL HC (Bluetooth connected)
Inverter 2: SB2.5-1VL-40 (Speedwire connected)

Since a few weeks, I have every morning an incorrect value for EToday when its a calculated value on my BT connected inverter. Once the inverter starts to produce power, the value is taken from the inverter and is correct. 

From the daily values, I grepped for EToday and ETotal. 
It looks as long the daily production is 0, for some reason, the Total production is calculated. When the Daily poroduction is not 0 any more, the correct value is taken from the inverter. 

I only see this on my Bleutooth inverter, the Speedwire connected inverter does not have the issue. 

`openhabian@openhabian:/mnt/usb/sbfspot.3 $ grep  "Starting\|EToday:\|ETotal:" dak_20210519.log
Wed May 19 06:00:01 2021: INFO: Starting...
Wed May 19 06:02:01 2021: INFO: Starting...
Wed May 19 06:04:01 2021: INFO: Starting...
Wed May 19 06:06:01 2021: INFO: Starting...
Wed May 19 06:08:01 2021: INFO: Starting...
Wed May 19 06:10:01 2021: INFO: Starting...
        EToday: 0.000kWh
        ETotal: 28676.079kWh
Wed May 19 06:12:01 2021: INFO: Starting...
        EToday: 0.000kWh
        ETotal: 28676.079kWh
Wed May 19 06:14:01 2021: INFO: Starting...
        EToday: 0.000kWh
        ETotal: 28676.079kWh
Wed May 19 06:16:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:18:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:20:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:22:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:24:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:26:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:28:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:30:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:32:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:34:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:36:01 2021: INFO: Starting...
Calculated EToday: 28676.079kWh
        EToday: 28676.079kWh
        ETotal: 28676.079kWh
Wed May 19 06:38:01 2021: INFO: Starting...
        EToday: 0.002kWh
        ETotal: 28676.081kWh
`

Any suggestions to get this corrected? 
Thanks

sbf-, 2021-05-19:
SBFspot should not calculate EToday if inverter is capable to provide it (see #290)

mrombouts, 2021-05-20:
So in this case, the inverter provides EToday, and we can assume SBFspot should not calculate it, correct? 
Any idea why is it calculating? 
While waiting for a fix, can I downgrade to a previous version (e.g. 3.7.0) - I didn't have this before, so I expect it started after I upgraded to 3.8.1. Not sure however what my previous version was

Thanks, 
Michiel

mrombouts, 2021-05-21:
Downgraded to 3.7.1 and seems to be working.
Will check tomorrow morning if the incorrect values have disappeared. 

If you want me to test the fix once available, just let met know

commit 0464d987 (2021-05-22): Fix #459 Incorrect Calculated EToday
closed 2021-05-22 by sbf-