SBFspot/SBFspot #465: Stopped uploading to PV Output - Error: sqlite3_exec() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing

petje72, 2021-05-27:
Pi stopped logging and sending it to PVoutput on May 24.

I restarted the pi and executed this command

```
/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv -startdate:20210523
SBFspot V3.8.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq -nocsv -startdate:20210523
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Thu May 27 13:41:52 2021: INFO: Starting...
sunrise: 05:33
sunset : 21:34
Connecting to 00:80:25:26:F7:43 (1/10)
Initializing...
SUSyID: 125 - SessionID: 916772929 (0x36A4D841)
SMA netID=03
Serial Nr: 7757087E (2002192510)
BT Signal=78.4%
Logon OK
Failed to get current plant time (-1)
SetPlantTime returned an error: -1
SUSyID: 99 - SN: 2002192510
Device Name:      Ameri Solar 5x260Wp
Device Class:     Solar Inverters
Device Type:      SB 1200
Software Version: 12.12.206.R
Serial number:    2002192510
SUSyID: 99 - SN: 2002192510
Device Status:      Ok
SUSyID: 99 - SN: 2002192510
Device Temperature: 0.0°C
SUSyID: 99 - SN: 2002192510
GridRelay Status:      ?
SUSyID: 99 - SN: 2002192510
Pac max phase 1: 1200W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 99 - SN: 2002192510
Energy Production:
	EToday: 0.929kWh
	ETotal: 9107.480kWh
	Operation Time: 36548.67h
	Feed-In Time  : 33801.61h
SUSyID: 99 - SN: 2002192510
DC Spot Data:
	String 1 Pdc:   0.000kW - Udc: 148.00V - Idc:  4.331A
	String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	Calculated Total Pdc:   0.000kW
SUSyID: 99 - SN: 2002192510
AC Spot Data:
	Phase 1 Pac :   0.000kW - Uac: 237.90V - Iac:  2.471A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.587kW - Calculated Pac:   0.000kW
	Efficiency  :    0.00%
SUSyID: 99 - SN: 2002192510
Grid Freq. : 49.99Hz
SUSyID: 99 - SN: 2002192510
Current Inverter Time: 27/05/2021 13:42:06
Inverter Wake-Up Time: 27/05/2021 13:42:05
Inverter Sleep Time  : 27/05/2021 13:42:06
**[2021-05-27 13:42:04.319] Error: sqlite3_exec() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1622115724,2002192510,0,0,4.331,0,148,0,0,0,0,2.471,0,0,237.9,0,0,929,9107480,49.99,36548.7,33801.6,78.4314,'OK','?',0)
[2021-05-27 13:42:04.322] Error: [spot_data]exec_query() returned: 'UNIQUE constraint failed: SpotData.TimeStamp, SpotData.Serial' while executing
INSERT INTO SpotData VALUES(1622115724,2002192510,0,0,4.331,0,148,0,0,0,0,2.471,0,0,237.9,0,0,929,9107480,49.99,36548.7,33801.6,78.4314,'OK','?',0)**
********************
* ArchiveDayData() *
********************
startTime = 60A97EE0 -> 23/05/2021 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime = 608D34B0 -> 01/05/2021 13:00:00
Reading events: 2021-May-01
Thu May 27 13:42:05 2021: INFO: Done.
```

-------
nothing is uploaded to PV output.
Pi is connected with SMA
Disk isn't full
```
df -h
Bestandssysteem Grootte Gebruikt Besch Geb% Aangekoppeld op
/dev/root           59G     6,2G   50G  12% /
devtmpfs           430M        0  430M   0% /dev
tmpfs              463M        0  463M   0% /dev/shm
tmpfs              463M     6,3M  456M   2% /run
tmpfs              5,0M     4,0K  5,0M   1% /run/lock
tmpfs              463M        0  463M   0% /sys/fs/cgroup
/dev/mmcblk0p1     253M      49M  204M  20% /boot
tmpfs               93M     8,0K   93M   1% /run/user/1000
```
System Log shows no system or disk errors

Anyone have any idea what is going on and how to fix?
thank you

petje72, 2021-05-27:
I guess it something with plant time, but how to fix it? I tried this:

```
/usr/local/bin/sbfspot.3/SBFspot -v -nosql -nocsv -finq | grep -i time
SetPlantTime returned an error: -1
Failed to get current plant time (-1)
	Operation Time: 36549.07h
	Feed-In Time  : 33802.01h
Current Inverter Time: 27/05/2021 14:05:54
Inverter Wake-Up Time: 27/05/2021 14:05:54
Inverter Sleep Time  : 27/05/2021 14:05:55
startTime = 60AEC4E0 -> 27/05/2021 00:00:00
startTime = 608D34B0 -> 01/05/2021 13:00:00

pi@raspberrypi:~ $ /usr/local/bin/sbfspot.3/SBFspot -settime
SBFspot V3.8.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -settime
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Thu May 27 14:06:28 2021: INFO: Starting...
sunrise: 05:33
sunset : 21:34
Connecting to 00:80:25:26:F7:43 (1/10)
Initializing...
SUSyID: 125 - SessionID: 987930749 (0x3AE2A07D)
SMA netID=03
Serial Nr: 7757087E (2002192510)
BT Signal=78.4%
Logon OK
Failed to get current plant time (-1)
Thu May 27 14:06:39 2021: INFO: Done.
```

sbf-, 2021-05-27:
Unique constraint issue is fixed in V3.8.1
settime issue is waiting for testing ([see settime branch](https://github.com/SBFspot/SBFspot/tree/settime))

> nothing is uploaded to PV output.

See [wiki](https://github.com/SBFspot/SBFspot/wiki/No-uploads-to-PVOutput.org)

petje72, 2021-05-28:
updated. error message is gone but still no upload to pvoutput.
looks like SBFspot is not reading out SMA correct, value is 0 kW. SMA inverter is operating OK, 100% operational

`$ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv
SBFspot V3.8.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri May 28 10:18:19 2021: INFO: Starting...
sunrise: 05:32
sunset : 21:35
Connecting to 00:80:25:26:F7:43 (1/10)
Initializing...
SUSyID: 125 - SessionID: 915583013 (0x3692B025)
SMA netID=03
Serial Nr: 7757087E (2002192510)
BT Signal=78.8%
Logon OK
SUSyID: 99 - SN: 2002192510
Device Name:      Ameri Solar 5x260Wp
Device Class:     Solar Inverters
Device Type:      SB 1200
Software Version: 12.12.206.R
Serial number:    2002192510
SUSyID: 99 - SN: 2002192510
Device Status:      Ok
SUSyID: 99 - SN: 2002192510
Device Temperature: 0.0°C
SUSyID: 99 - SN: 2002192510
GridRelay Status:      ?
SUSyID: 99 - SN: 2002192510
Pac max phase 1: 1200W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 99 - SN: 2002192510
Energy Production:
	EToday: 0.589kWh
	ETotal: 9109.330kWh
	Operation Time: 36561.15h
	Feed-In Time  : 33812.59h
SUSyID: 99 - SN: 2002192510
DC Spot Data:
	String 1 Pdc:   0.000kW - Udc: 166.00V - Idc:  2.351A
	String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	Calculated Total Pdc:   0.000kW
SUSyID: 99 - SN: 2002192510
AC Spot Data:
	Phase 1 Pac :   0.000kW - Uac: 236.30V - Iac:  1.503A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.355kW - Calculated Pac:   0.000kW
	Efficiency  :    0.00%
SUSyID: 99 - SN: 2002192510
Grid Freq. : 50.00Hz
SUSyID: 99 - SN: 2002192510
Current Inverter Time: 28/05/2021 10:18:23
Inverter Wake-Up Time: 28/05/2021 10:18:23
Inverter Sleep Time  : 28/05/2021 10:18:23
********************
* ArchiveDayData() *
********************
startTime = 60B01660 -> 28/05/2021 00:00:00
**********************`

sbf-, 2021-05-28:
Can be an issue with inverter. Try to reset it (power down)
If not fixed, have a look with sunny explorer.

petje72, 2021-05-28:
it was powered down yesterday. I don't have a Windows computer.. will fix that (VM) and take a look with sunny explorer

sbf-, 2021-06-05:
> it was powered down yesterday. I don't have a Windows computer.. will fix that (VM) and take a look with sunny explorer

Any progress?

petje72, 2021-06-05:
SMA is fine, bluetooth piggyback OK too. SDcard is corrupt so I will do new config on a new sdcard. you can close this issue

closed 2021-06-05 by sbf-