SBFspot/SBFspot #466: SBFspotUploadDaemon causing excessive disk writes

ot3, 2021-05-27:
I'm seeing a very high SSD GB written/day on /var/tmp. I have issued the following command:

`sudo fatrace -c -t -f W -s 120`

which results in a very large number (185752 in 25s )of lines like:

`14:47:55.527344 SBFspotUploadDa(21902): W /var/tmp/etilqs_c719517b3fb204a8 (deleted)`

Analysis shows the following activity from SBFspotUploadDaemon:

- observed period: 25s
- number of files deleted: 185752
- number of file deletions/s: 445805

This may have a totally innocent explanation but this activity is destroying my SSD. Any idea what's going on? Restarting the daemon makes no difference.

SBFspot 3.5

sbf-, 2021-05-27:
I don't see this kind of files written to disk...
```
pi@sbfspot:~ $ sudo ./fatrace -c -t -f W -s 120
18:22:30.208472 SBFspotUploadDa(533): WO /home/pi/smadata/SBFspot.db-shm
18:22:30.392742 SBFspotUploadDa(533): CW /home/pi/smadata/SBFspot.db-shm (deleted)
18:22:30.393039 SBFspotUploadDa(533): CW /home/pi/smadata/SBFspot.db-wal
18:22:30.393922 SBFspotUploadDa(533): CW /home/pi/smadata/SBFspot.db
18:22:54.738703 rsyslogd(300): W /var/log/syslog
18:22:54.738976 rsyslogd(300): W /var/log/kern.log
18:23:00.978639 rsyslogd(300): W /var/log/syslog
18:23:00.978980 rsyslogd(300): W /var/log/kern.log
18:23:00.978980 rsyslogd(300): W /var/log/messages
18:23:30.408498 SBFspotUploadDa(533): WO /home/pi/smadata/SBFspot.db-shm
18:23:30.606983 SBFspotUploadDa(533): CW /home/pi/smadata/SBFspot.db-shm (deleted)
18:23:30.606983 SBFspotUploadDa(533): CW /home/pi/smadata/SBFspot.db-wal (deleted)
18:23:30.607740 SBFspotUploadDa(533): CW /home/pi/smadata/SBFspot.db
```

Using V3.8.1 with SQLite 3.27.2

ot3, 2021-05-27:
Doesn't surprise me; I'll update to the latest version and report back. Thanks for trying the same command. 

In hindsight I think this problem has existed in my installation for more than 2 years when I installed this version. It has probably already destroyed a 2.5" HDD and is now working on its replacement :-)

sbf-, 2021-05-27:
etilqs_* files are SQLite temp files - etilqs is sqlite reversed ;-)
So I suppose the SQLite version is important here.

ot3, 2021-05-27:
Yes, I found that out as well. I'm now running 3.8.2 and the latest SQLite version 3.22.0 that comes with Ubuntu. The upgrade did not help. As an experiment I made a backup of SBFspot.db and created a fresh database. The backup database is 288550912 bytes. Are you aware of any issues with databases above a certain limit?

Right now, no etilqs files are being generated; we'll see tomorrow what happens during production.

ot3, 2021-05-28:
The system is running fine now. The original database was fine, I checked with:
`sqlite3 /home/pi/smadata/SBFspot.db "pragma integrity_check;";`

Recreating the database caused the SSD write rate to drop from 110GB/day to 0.9GB/day.

Conclusions:
- Sqlite3 caused excessive writes: 445805 file deletions/minute
- Most likely this caused an earlier 2.5" 1TB HDD to fail after 2 years; it's successor  - a 500GB SSD - was well on its way to early retirement.
- Starting with a fresh database reduces the SSD wear >100x and solves the problem
- The original database was over 200MB; the root cause for the excessive writes is unknown as the database was structurally sound and far below Sqlite3's size limit of 256TB.

closed 2021-05-28 by ot3

sbf-, 2021-05-28:
I just wonder how your db got so big?
Mine has only 82Mb since beginning 2014 (about 360k records in spotdata and daydata tables)
Inverter polling frequency every 5 minutes

ot3, 2021-05-28:
Polling every minute probably explains a lot and since 2019 2 inverters. I'll try to get into the habit of creating a new database at the beginning of the year.