SBFspot/SBFspot #474: Upload not working

unexxx, 2021-07-26:
Hi,

I installed SBFspot on my RP4b and it worked fine for a few minutes I think as I got some output on PVOutput but now it is no longer working. I tripple checked all the config files and reinstalled even and still I get the same issue. I can query the SQLDB and it looks like there is content, MyPlant log file in /var/log/ is also populated without errors. However the uploaddaemon log is not good I think:

[20:31:17] INFO: SBFspotUploadDaemon Version 3.0.1
[20:31:17] INFO: Starting Daemon...
[20:32:11] INFO: SBFspotUploadDaemon Version 3.0.1
[20:32:11] INFO: Starting Daemon...
[21:02:17] INFO: SBFspotUploadDaemon Version 3.0.1
[21:02:17] INFO: Starting Daemon...
[21:02:18] WARNING: MySolar is not yet member of SBFspot Team. Consider joining at http://pvoutput.org/listteam.jsp?tid=613
[21:04:43] INFO: SBFspotUploadDaemon Version 3.0.1
[21:04:43] INFO: Starting Daemon...

Is all that is in there when I start the daemon... and the daemon looks like it is hanging when I manually launch it at CLI. 

I am lost on what I can still look at... any ideas are VERY much appreciated!

sbf-, 2021-07-26:
There's a [wiki page](https://github.com/SBFspot/SBFspot/wiki/No-uploads-to-PVOutput.org) with usefull tips & tricks

unexxx, 2021-07-28:
Thanks for the link! It all worked fine the next day so I assume PVO needed some time to adopt the changes / reflect them. Will keep the link in bookmarks for potential future need. Thanks again!

closed 2021-07-28 by unexxx