SBFspot/SBFspot #475: CSV Extended Header parsing

sandlerr, 2021-07-29:
Hi,
The CSV parser in Ruby is choking on CSV files with headers like this

```
sep=;
Version CSV1|Tool SE|Linebreaks CR/LF|Delimiter semicolon|Decimalpoint dot|Precision 3

;SN: xxx;SN: xxx
;SB 5000TL-21;SB 5000TL-21
;xxx;xxx
;Total yield;Power
;Counter;Analog
dd/MM/yyyy hh:mm tt;kWh;kW
01/07/2021 12:00 AM;22939.257;0.312
```

Is there any documentation or code sample for consuming extended CSV headers? Should users just throw out the first eight lines before passing the rest to a standard CSV parser?

The error is: ` CSV::MalformedCSVError Exception: Unquoted fields do not allow new line <"\n"> in line 2.`

sbf-, 2021-07-30:
Set CSV_ExtendedHeader=0 in config to remove the header

sandlerr, 2021-07-30:
Hi sbf, thank you for the quick response. Yes, I'm aware that the extended header may be disabled, however I have a bunch of CSV files that were already generated, and it seems like there is some useful information encoded in the extended header. I am just wondering whether / how others are parsing it.

closed 2021-08-01 by sbf-