SBFspot/SBFspot #480: SBFUploadDaemon stopped uploading data to PVoutput

ced2git, 2021-08-16:
Hello, 

Since the 7th of august, SBFuploaddaemon stopped working. 

I've updated to the latest version `[13:42:11] INFO: SBFspotUploadDaemon Version 2.0.0
[13:42:11] INFO: Starting Daemon...
[18:00:32.270] INFO: SBFspotUploadDaemon Version 3.0.1
[18:00:32.385] INFO: Starting Daemon...
[18:00:33.112] DEBUG: Retrieving new datapoints from DB...
[18:01:47.723] DEBUG: Retrieving new datapoints from DB...
[18:02:34.021] DEBUG: Retrieving new datapoints from DB...
[18:04:30.474] DEBUG: Retrieving new datapoints from DB...
[18:05:17.194] DEBUG: Retrieving new datapoints from DB...
[18:05:57.509] DEBUG: Retrieving new datapoints from DB...
[18:07:30.807] DEBUG: Retrieving new datapoints from DB...
[18:08:17.040] DEBUG: Retrieving new datapoints from DB...
[18:08:57.184] DEBUG: Retrieving new datapoints from DB...
`

No datapoints are sent to PVOutput. Can someone help me ? 

I'm using the sqlite version.

sbf-, 2021-08-16:
You can start [here](https://github.com/SBFspot/SBFspot/wiki/No-uploads-to-PVOutput.org)

ced2git, 2021-08-16:
ok. perfect. It was a question of clock on the Inverter. 

I'm using a rpi zero to retrieve the info from the inverter in my garage. After few days, the rpi stop connecting to the inverter and i assume that inverter desync their clocks. . 

I cannot fix it with SBFSpot, i need to use sunny explorer for that. This is a bit cumbersome.

sbf-, 2021-08-16:
Feel free to test the [settime branch](https://github.com/SBFspot/SBFspot/tree/settime)

closed 2021-08-28 by sbf-