SBFspot/SBFspot #482: some data missing on 2. Inverter

AingTii, 2021-09-05:
Hello,

I have two inverters SB3000 and SB3800 both are found by SBFSpot, but some data are missing.
I seems from the Master (SB3000) that ist configured with the BT Adresse I get all Data.
the second one SB3800 there are some important data missing.
Like Pdc1, Pac1, PdvTot, Efficentcy, BT Signal.
But other are there ldc1, Udc1, lac1 uac1.... so I thing there may some issue?

`
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Tacker/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Tacker/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

DeviceType | Serial | Pdc1 | Pdc2 | Idc1 | Idc2 | Udc1 | Udc2 | Pac1 | Pac2 | Pac3 | Iac1 | Iac2 | Iac3 | Uac1 | Uac2 | Uac3 | PdcTot | PacTot | Efficiency | EToday | ETotal | Frequency | OperatingTime | FeedInTime | BT_Signal | Condition | GridRelay | Temperature
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
SB 3000 |   | 95 | 0 | 0,268 | 0 | 356 | 0 | 74 | 0 | 0 | 0,326 | 0 | 0 | 228 | 0 | 0 | 95 | 74 | 77,895 | 0,036 | 30689,167 | 49,98 | 49967,694 | 45079,316 | 69,412 | Ok | ? | 0
SB 3800 |   | 0 | 0 | 0,523 | 0 | 271 | 0 | 0 | 0 | 0 | 0,51 | 0 | 0 | 229,3 | 0 | 0 | 0 | 117 | 0 | 0,066 | 49082,913 | 49,98 | 50968,588 | 48884,081 | 0 | Ok | ? | 0

</body>

</html>
`
thank you
Jens

sbf-, 2021-09-05:
Probably a communication issue. Can you provide a debug log (-d5 -v5)

AingTii, 2021-09-05:
OK, I will try, 
I'm right the command is....

` /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv -d5 -v5^C
`
[debug.txt](https://github.com/SBFspot/SBFspot/files/7111726/debug.txt)

I hope you mean taht, find attached file pls

thank you

sbf-, 2021-09-05:
In config, set CalculateMissingSpotValues=1
Not all inverters report their internal temperature, that´s why it´s 0

AingTii, 2021-09-05:
CalculateMissingSpotValues=1 is already set
maybe I should try 0 ?

AingTii, 2021-09-06:
I tried both, 
CalculateMissingSpotValues=1
and
CalculateMissingSpotValues=0

0 no values
1 same like before, the 2. inverter no values.

`SBFspot V3.8.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Mon Sep  6 08:50:27 2021: INFO: Starting...
sunrise: 04:37
sunset : 18:11
Connecting to 00:80:25:15:61:AB (1/10)
Initializing...
SUSyID: 125 - SessionID: 941129997 (0x3818810D)
SMA netID=02
SUSyID: 99 - SN: xxxxxxxx56
SUSyID: 99 - SN: xxxxxxxx32
BT Signal=68.2%
Logon OK
Failed to get current plant time (-1)
SetPlantTime returned an error: -1
SUSyID: 99 - SN: xxxxxxxx56
Device Name:      SN: xxxxxxxx56
Device Class:     Solar-Wechselrichter
Device Type:      SB 3000
Software Version: 12.09.122.R
Serial number:    xxxxxxxx56
SUSyID: 99 - SN: xxxxxxxx32
Device Name:      SN: xxxxxxxx32
Device Class:     Solar-Wechselrichter
Device Type:      SB 3800
Software Version: 12.09.122.R
Serial number:    xxxxxxxx32
SUSyID: 99 - SN: xxxxxxxx56
Device Status:      Ok
SUSyID: 99 - SN: xxxxxxxx32
Device Status:      Ok
SUSyID: 99 - SN: xxxxxxxx56
Device Temperature: 0.0°C
SUSyID: 99 - SN: xxxxxxxx32
Device Temperature: 0.0°C
SUSyID: 99 - SN: xxxxxxxx56
GridRelay Status:      ?
SUSyID: 99 - SN: xxxxxxxx32
GridRelay Status:      ?
SUSyID: 99 - SN: xxxxxxxx56
Pac max phase 1: 3000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 99 - SN: xxxxxxxx32
Pac max phase 1: 3800W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 99 - SN: xxxxxxxx56
Energy Production:
        EToday: 1.328kWh
        ETotal: 30706.024kWh
        Operation Time: 49984.04h
        Feed-In Time  : 45094.92h
SUSyID: 99 - SN: xxxxxxxx32
Energy Production:
        EToday: 2.096kWh
        ETotal: 49109.396kWh
        Operation Time: 50985.09h
        Feed-In Time  : 48900.24h
SUSyID: 99 - SN: xxxxxxxx56
DC Spot Data:
        String 1 Pdc:   0.746kW - Udc: 356.00V - Idc:  2.096A
        String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
        Calculated Total Pdc:   0.746kW
SUSyID: 99 - SN: xxxxxxxx32
DC Spot Data:
        String 1 Pdc:   0.000kW - Udc: 273.00V - Idc:  4.383A
        String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
        Calculated Total Pdc:   0.000kW
SUSyID: 99 - SN: xxxxxxxx56
AC Spot Data:
        Phase 1 Pac :   0.708kW - Uac: 230.00V - Iac:  3.080A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.708kW - Calculated Pac:   0.708kW
        Efficiency  :   94.91%
SUSyID: 99 - SN: xxxxxxxx32
AC Spot Data:
        Phase 1 Pac :   0.000kW - Uac: 231.20V - Iac:  4.916A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   1.136kW - Calculated Pac:   0.000kW
        Efficiency  :    0.00%
SUSyID: 99 - SN: xxxxxxxx56
Grid Freq. : 50.02Hz
SUSyID: 99 - SN: xxxxxxxx32
Grid Freq. : 50.01Hz
SUSyID: 99 - SN: xxxxxxxx56
Current Inverter Time: 06/09/2021 08:50:53
Inverter Wake-Up Time: 06/09/2021 08:50:52
Inverter Sleep Time  : 06/09/2021 08:50:54
SUSyID: 99 - SN: xxxxxxxx32
Current Inverter Time: 06/09/2021 08:50:53
Inverter Wake-Up Time: 06/09/2021 08:50:52
Inverter Sleep Time  : 06/09/2021 08:50:54
********************
* ArchiveDayData() *
********************
startTime = 61355A00 -> 06/09/2021 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime = 612F6B40 -> 01/09/2021 12:00:00
Reading events: 2021-Sep-01
Mon Sep  6 08:50:43 2021: INFO: Done.`

sbf-, 2021-09-06:
Hmm, it should make a difference...
Can you run it with -v5 so I can see the actual config being read?

AingTii, 2021-09-06:
oh maybe you missunderstand.

default was already CalculateMissingSpotValues=1

I changed to =0 and I both has missing data.

I will make in both configurations.

AingTii, 2021-09-06:
ok, here the debug -v5

debug_1.txt means with  config CalculateMissingSpotValues=1
debug_0.txt means with  config CalculateMissingSpotValues=0

[debug_1.txt](https://github.com/SBFspot/SBFspot/files/7116781/debug_1.txt)
[debug_0.txt](https://github.com/SBFspot/SBFspot/files/7116782/debug_0.txt)

dfference is with 1 the first inverter is calulated, the second not
with 0 both not calculated

Thanks

sbf-, 2021-09-08:
I confirm it's a bug. SBFspot is only calculating missing values for first inverter.

commit ef1122ae (2021-09-12): Fix #482 Some data missing on 2nd Inverter
closed 2021-09-12 by sbf-