SBFspot/SBFspot #497: Problem - Compile SBFspot with SQLite support : make sqlite

JanCLinux, 2021-11-14:
Hello,
I'am a real novice in using Linux and I need some help to solve below error:

According the SBFspot installation instructions all went well until the command:

sudo make sqlite

I got the following response:

test -d sqlite/obj/ || mkdir -p sqlite/obj/
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/boost_ext.o 
g++ main.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_SQLITE -o sqlite/obj/main.o 
In file included from SBFNet.h:39:0,
                 from SBFspot.h:38,
                 from SQLselect.h:38,
                 from Inverter.h:37,
                 from main.cpp:35:
bluetooth.h:67:10: fatal error: bluetooth/bluetooth.h: No such file or directory
 #include <bluetooth/bluetooth.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
makefile:116: recipe for target 'sqlite/obj/main.o' failed
make: *** [sqlite/obj/main.o] Error 1

Looking forward to some help on this,
Jan

notreallybob, 2021-11-14:
It's missing some libraries. Did you forget to run these 2 commands first?
sudo apt-get --no-install-recommends -y install bluetooth libbluetooth-dev
sudo apt-get -y install ntpdate sqlite3 libsqlite3-dev libboost-all-dev libcurl3-dev build-essential

JanCLinux, 2021-11-14:
Many thanks. I indeed needed to install the bluetooth library.

closed 2021-11-14 by JanCLinux