SBFspot/SBFspot #502: Data Gap (20 days ago)

BionicThomas, 2021-12-20:
Hi experts!

I only noticed today (20th of December) that the Pi stopped working on the 17th of November, so more than 30 days ago.
on PVOutput I have donated and therefore should be able to upload up to 60 days into the past.
via
`/usr/local/bin/sbfspot.3/SBFspot -v -finq -ad35`  I was able to get data from my Inverter (SMA Tripower) between the 8th of December and today and SPFspot uploaded it to PVOutput.
But the gap between the 17th of December and the 8th of December remains.
Can I close the gap with a command to upload this old data, too?

sbf-, 2021-12-21:
Is the data available in the database? Maybe there was an issue with your inverter.

BionicThomas, 2021-12-21:
You are correct looks like I should just have been patient. Today i find the gap is closed. Took maybe some time to upload the 20 missing days or the donation status was not set yet.
Thanks for the quick response and the good work to all the contributors!

closed 2021-12-21 by BionicThomas