SBFspot/SBFspot #507: SB 3000TL-21 still supported? incomplete output with latest versions

pooh22, 2021-12-31:
Recently I've started to play with home assistant and I would like to integrate my solar data into it. I found that sbfspot-config and recent versions of SBFspot will send mqtt data, but I have issues.
I've tried installing the latest version (3.9.0 and I think 3.8.4 as well), which segfaulted, but 3.6.0-3.8.3 produce incomplete output

The expected output is:
```
root@sunnyboy:/home/pi/src/sbfspot-config# /home/pi/bin/sbfspot -cfg/home/pi/smadata/SBFspot.cfg -mqtt -settime -finq                                
SBFspot V3.3.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2016, SBF (https://sbfspot.codeplex.com)
Compiled for Linux (LE) 32 bit

Commandline Args: -v -finq
Reading config './SBFspot.cfg'
Warning: Ignoring keyword 'MQTT_Publisher'
Warning: Ignoring keyword 'MQTT_Host'
Warning: Ignoring keyword 'MQTT_Topic'
Warning: Ignoring keyword 'MQTT_ItemFormat'
Warning: Ignoring keyword 'MQTT_ItemDelimiter'
Warning: Ignoring keyword 'MQTT_PublisherArgs'
Warning: Ignoring keyword 'MQTT_Data'
Fri Dec 31 13:46:15 2021: INFO: Starting...
sunrise: 08:45
sunset : 16:33
Connecting to 00:80:25:2c:86:f8 (1/10)
Connecting to 00:80:25:2c:86:f8 (2/10)
Connecting to 00:80:25:2c:86:f8 (3/10)
Connecting to 00:80:25:2c:86:f8 (4/10)
Connecting to 00:80:25:2c:86:f8 (5/10)
Connecting to 00:80:25:2c:86:f8 (6/10)
Connecting to 00:80:25:2c:86:f8 (7/10)
Connecting to 00:80:25:2c:86:f8 (8/10)
Connecting to 00:80:25:2c:86:f8 (9/10)
Initializing...
SUSyID: 125 - SessionID: 906454326 (0x36076536)
SMA netID=01
Serial Nr: 7EF8F650 (2130245200)
BT Signal=72.5%
Logon OK
Local Host Time: 2021-12-31 13:46:23
Plant Time     : 2021-12-31 13:46:23 (0 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 2021-12-31 13:45:56
SUSyID: 138 - SN: 2130245200
Device Name:      SN: 2130245200 (JBW)
Device Class:     Solar Inverters
Device Type:      SB 3000TL-21
Software Version: 02.55.03.R
Serial number:    2130245200
SUSyID: 138 - SN: 2130245200
Device Status:      Ok
SUSyID: 138 - SN: 2130245200
Device Temperature: 28.5°C
SUSyID: 138 - SN: 2130245200
GridRelay Status:      Closed
SUSyID: 138 - SN: 2130245200
Pac max phase 1: 3000W
Pac max phase 2: 3000W
Pac max phase 3: 3000W
SUSyID: 138 - SN: 2130245200
Energy Production:
        EToday: 1.075kWh
        ETotal: 25428.327kWh
        Operation Time: 33783.57h
        Feed-In Time  : 32256.07h
SUSyID: 138 - SN: 2130245200
DC Spot Data:
        String 1 Pdc:   0.158kW - Udc: 150.73V - Idc:  1.055A
        String 2 Pdc:   0.311kW - Udc: 274.16V - Idc:  1.139A
SUSyID: 138 - SN: 2130245200
AC Spot Data:
        Phase 1 Pac :   0.469kW - Uac: 233.26V - Iac:  2.014A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.469kW
SUSyID: 138 - SN: 2130245200
Grid Freq. : 49.98Hz
SUSyID: 138 - SN: 2130245200
Current Inverter Time: 2021-12-31 13:46:19
Inverter Wake-Up Time: 2021-12-31 08:32:37
Inverter Sleep Time  : 2021-12-31 13:46:23
ExportSpotDataToCSV()
********************
* ArchiveDayData() *
********************
startTime = 61CE39F0 -> 31/12/2021 00:00:00
ExportDayDataToCSV()
**********************
* ArchiveMonthData() *
**********************
startTime = 61A755B0 -> 01/12/2021 12:00:00
ExportMonthDataToCSV()
Reading events: 2021-Dec-01
ExportEventsToCSV()
Fri Dec 31 13:46:30 2021: INFO: Done.
```

The output I get from recent versions is:
```
root@sunnyboy:/home/pi/src/sbfspot-config# /usr/local/bin/sbfspot.3/SBFspot -cfg/home/pi/smadata/SBFspot.cfg -mqtt -settime -finq
SBFspot V3.6.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2019, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -cfg/home/pi/smadata/SBFspot.cfg -mqtt -settime -finq
Reading config '/home/pi/smadata/SBFspot.cfg'
Fri Dec 31 13:54:26 2021: INFO: Starting...
sunrise: 08:45
sunset : 16:33
Connecting to 00:80:25:2c:86:f8 (1/10)
Initializing...
SUSyID: 125 - SessionID: 965486163 (0x398C2653)
SMA netID=01
Serial Nr: 7EF8F650 (2130245200)
BT Signal=73.3%
Logon OK
Local Host Time: 2021-12-31 13:54:28
Plant Time     : 2021-12-31 13:54:28 (0 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 2021-12-31 13:45:56
```
or
```
root@sunnyboy:/home/pi/src/sbfspot-config# /usr/local/bin/sbfspot.3/SBFspot -cfg/home/pi/smadata/SBFspot.cfg -mqtt -settime -finq
SBFspot V3.8.3
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -cfg/home/pi/smadata/SBFspot.cfg -mqtt -settime -finq
Reading config '/home/pi/smadata/SBFspot.cfg'
Fri Dec 31 13:56:41 2021: INFO: Starting...
sunrise: 08:45
sunset : 16:33
Connecting to 00:80:25:2c:86:f8 (1/10)
Initializing...
SUSyID: 125 - SessionID: 813876602 (0x3082C57A)
SMA netID=01
Serial Nr: 7EF8F650 (2130245200)
BT Signal=73.3%
Logon OK
Local Host Time: 2021-12-31 13:56:44
Plant Time     : 2021-12-31 13:56:43 (-1 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 2021-12-31 13:45:56
Adjusting plant time...
New plant time is now 2021-12-31 13:56:44
Fri Dec 31 13:56:44 2021: INFO: Done.
```
If necessary I can provide my config as well, perhaps I'm doing something wrong?

At the moment I'm using the old version via xinetd to produce output when a tcp connection is made to a certain port. My icinga monitoring does this every minute to produce energy monitoring graphs, etc.

sbf-, 2022-01-02:
Remove the -settime argument

closed 2022-01-02 by sbf-

pooh22, 2022-01-02:
thanks, and adding -v2 gets me the old output I'm parsing with some other script :-)
On to the next hurdle integrating the solar panels in home assistant...