SBFspot/SBFspot #511: Installation aborting - revert back to 3.8.4

Gunhawk007, 2022-01-04:
Hi I have tried the previous 3.9.0 release as well as this one. After the curl command I got the prompt for the configuration screen, but when using my previous config, I got a red prompt screen "Installation aborting".
When I manually launch the /usr/local/bin/sbfspot.3/SBFspot -v5 to get the maximum verbose messages back I can connect, giving out proper BT-Signal, Time information and after "Last Time Set: 18/11/2021 16:15:02" I got a Segmentation Fault.
(Speicherzugriffsfehler).
I even tried to create a new profile but still getting this error.
Not sure what is wrong there. 

I tried to revert back to version 3.8.4, but having logon problems "Critical Logon failed. Check USER Password" for some reasons though I haven't changed anything in the config.

sbf-, 2022-01-07:
With latest version, can you test with SynchTime=0 in config file?

Gunhawk007, 2022-01-07:
Hi, I just changed the SynchTime="7" to SynchTime="0" as requested and re-run the script to get the 3.9.1. version installed. However, the install failure remains same. When executing "SBFspot -v2" manually, I still got the segmentation fault.
pi@solarpi:~ $ /usr/local/bin/sbfspot.3/SBFspot -v2
SBFspot V3.9.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v2
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Jan  7 12:32:43 2022: INFO: Starting...
sunrise: 08:16
sunset : 16:09
Connecting to 00:80:25:A5:6E:45 (1/10)
Initializing...
SUSyID: 125 - SessionID: 940018399 (0x38078ADF)
SMA netID=01
Serial Nr: 7EF70FC2 (2130120642)
BT Signal=81.6%
Logon OK
Local Host Time: 07/01/2022 12:32:44
Plant Time     : 07/01/2022 12:33:16 (+32 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 18/11/2021 16:15:02
Speicherzugriffsfehler

sbf-, 2022-01-07:
Can you provide a full debuglog (-d5 -v5)?

kriznik, 2022-01-09:
Hi I have same issue (reverted back to 3.8.4)

here is log from debug:

```
dietpi@pidns:~$ sudo /usr/local/bin/sbfspot.3/SBFspot -finq -d5 -v5
SBFspot V3.9.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -finq -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:2D:1E:91
Password=<undisclosed>
MIS_Enabled=0
Plantname=MyPlant
OutputPath=/home/dietpi/smadata/%Y
OutputPathEvents=/home/dietpi/smadata/%Y/Events
Latitude=50.08
Longitude=14.44
Timezone=Europe/Prague
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=0
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
### End of Config ###
Sun Jan  9 17:56:21 2022: INFO: Starting...
sunrise: 08:00
sunset : 16:18
Connecting to 00:80:25:2D:1E:91 (1/10)
Initializing...
SUSyID: 125 - SessionID: 959941727 (0x39378C5F)
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 91 1E 2D 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 91 1E 2D 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=01
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00
00000010: 91 1E 2D 25 80 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
[... 239 lines cut ...]
00000040: 00 01 1E 82 10 8F FA DA 61 53
00000050: 4E 3A 20 33 30 34 39 34 34 30
00000060: 34 38 00 00 00 FE FF FF 00 00
00000070: 00 00 00 00 00 00 00 00 00 00
00000080: 00 01 1F 82 08 8F FA DA 61 41
00000090: 1F 00 01 42 1F 00 00 FE FF FF
00000100: 00 00 00 00 00 00 00 00 00 00
00000110: 00 00 00 00 00 00 00 00 00 00
00000120: 00 01 20 82 08 8F FA DA 61 8A
00000130: 23 00 01 8B 23 00 00 8C 23 00
00000140: 00 8E 23 00 00 8F 23 00 00 E5
00000150: 23 00 00 E6 23 00 00 E7 23 00
00000160: 00 01 20 82 08 8F FA DA 61 E9
00000170: 23 00 00 E8 23 00 00 FE FF FF
00000180: 00 00 00 00 00 00 00 00 00 00
00000190: 00 00 00 00 00 00 00 00 00 00
00000200: 00 7B 8D 7E
<<<=================================>>>
MAX_pcktBuf is now 204 bytes
INV_NAME    : 'SN: 304944048' Sun Jan  9 16:09:03 2022
INV_CLASS   : 'Solar Inverters' Sun Jan  9 16:09:03 2022
INV_TYPE    : 'STP 5000TL-20' Sun Jan  9 16:09:03 2022
Segmentation fault
dietpi@pidns:~$
```

Gunhawk007, 2022-01-10:
Here we go:

pi@solarpi:~ $ /usr/local/bin/sbfspot.3/SBFspot -d -v5
SBFspot V3.9.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -d -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:A5:6E:45
Password=<undisclosed>
MIS_Enabled=0
Plantname=MyPlant
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=52.52
Longitude=13.4
Timezone=Europe/Berlin
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Mon Jan 10 09:20:17 2022: INFO: Starting...
sunrise: 08:14
sunset : 16:13
Connecting to 00:80:25:A5:6E:45 (1/10)
Initializing...
SUSyID: 125 - SessionID: 904992901 (0x35F11885)
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
cmd=2
SMA netID=01
31 Bytes sent
getPacket(5)
Received 18 bytes
Received 13 bytes
cmd=10
Received 18 bytes
Received 16 bytes
cmd=5
Local BT address: B8:27:EB:F6:3D:5B
63 Bytes sent
getPacket(1)
Received 18 bytes
MAX_CommBuf is now 88 bytes
Received 88 bytes
cmd=1
PacketLength=106
Serial Nr: 7EF70FC2 (2130120642)
logoffSMAInverter()
[... 346 lines cut ...]
Received 18 bytes
Received 66 bytes
cmd=1
PacketLength=84
Logon OK
SetPlantTime()
91 Bytes sent
getPacket(1)
Received 18 bytes
Received 74 bytes
cmd=1
PacketLength=92
Local Host Time: 10/01/2022 09:18:22
Plant Time     : 10/01/2022 09:18:57 (+35 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 18/11/2021 16:15:02
64 Bytes sent
getPacket(1)
Received 18 bytes
Received 88 bytes
cmd=1
PacketLength=106
INV_SWVER   : '02.54.02.R' Sun Jan  9 17:50:08 2022
64 Bytes sent
pi@solarpi:~ $

sbf-, 2022-01-10:
Someone able to debug this? I cannot reproduce this issue.

kriznik, 2022-01-10:
log i've provided is not helping? ... i can reproduce that on two rpi's ... so whatever you need me to do I can.

sbf-, 2022-01-11:
> log i've provided is not helping? ... i can reproduce that on two rpi's ... so whatever you need me to do I can.

Well, it gives me an idea where it goes wrong, but not the exact spot as I can't reproduce it on my own systems (it might be inverter related)
Since 3.8.4 several changes were made to SBFspot.cpp. You can help by commenting out some of the cases in this file (around line 2476)
```
switch (lri)
{
    case GridMsTotW: //SPOT_PACTOT
```

starting with `case NameplateLocation: //INV_NAME`
Put `//` in front of each line in the case statement like this:
``` c++
//    case NameplateLocation: //INV_NAME
//        //This function gives us the time when the inverter was switched on
//        devList[inv]->WakeupTime = datetime;
//        devList[inv]->DeviceName = std::string((char *)recptr + 8, strnlen((char *)recptr + 8, recordsize - 8)); // Fix #506
//        devList[inv]->flags |= type;
//        debug_text("INV_NAME", devList[inv]->DeviceName.c_str(), datetime);
//        break;
```
Then, build SBFspot (make sqlite) and test it. If the segmentation fault is still there, exclude more cases (one at a time and keep the previous changes)
Possible culprits are `NameplateLocation, NameplatePkgRev, NameplateModel, NameplateMainModel`

Thanks for your help

kriznik, 2022-01-11:
seems like that part of the code is not an issue, I've commented whole section from lines 2657 till 2663, build that and result is segmentation fault as well

(different raspberry, dietpiOS)

btw strange is at the end there is Tue Jan 11 16:24:28 2022
which is completely wrong as i'm looking into sunny explorer and inverter has got correct time 21:59 Tue Jan 11 (as i ran it couple of mins ago)

```
dietpi@DietPi:~/SBFspot/SBFspot$ sudo /usr/local/bin/sbfspot.3/SBFspot -finq -d5 -v5
SBFspot V3.9.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -finq -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:2D:1E:91
Password=<undisclosed>
MIS_Enabled=0
Plantname=MyPlant
OutputPath=/home/dietpi/smadata/%Y
OutputPathEvents=/home/dietpi/smadata/%Y/Events
Latitude=50.08
Longitude=14.44
Timezone=Europe/Prague
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/dietpi/smadata/SBFspot.db
### End of Config ###
Tue Jan 11 21:59:12 2022: INFO: Starting...
sunrise: 07:58
sunset : 16:21
Connecting to 00:80:25:2D:1E:91 (1/10)
Initializing...
SUSyID: 125 - SessionID: 845349180 (0x3263013C)
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 91 1E 2D 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 91 1E 2D 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
[... 288 lines cut ...]
00000040: 00 01 1E 82 10 2C A1 DD 61 53
00000050: 4E 3A 20 33 30 34 39 34 34 30
00000060: 34 38 00 00 00 FE FF FF 00 00
00000070: 00 00 00 00 00 00 00 00 00 00
00000080: 00 01 1F 82 08 2C A1 DD 61 41
00000090: 1F 00 01 42 1F 00 00 FE FF FF
00000100: 00 00 00 00 00 00 00 00 00 00
00000110: 00 00 00 00 00 00 00 00 00 00
00000120: 00 01 20 82 08 2C A1 DD 61 8A
00000130: 23 00 01 8B 23 00 00 8C 23 00
00000140: 00 8E 23 00 00 8F 23 00 00 E5
00000150: 23 00 00 E6 23 00 00 E7 23 00
00000160: 00 01 20 82 08 2C A1 DD 61 E9
00000170: 23 00 00 E8 23 00 00 FE FF FF
00000180: 00 00 00 00 00 00 00 00 00 00
00000190: 00 00 00 00 00 00 00 00 00 00
00000200: 00 21 D0 7E
<<<=================================>>>
MAX_pcktBuf is now 204 bytes
10821E01 40 Tue Jan 11 16:24:28 2022 Device name: 'SN: 304944048'
INV_CLASS   : 'Solar Inverters' Tue Jan 11 16:24:28 2022
INV_TYPE    : 'STP 5000TL-20' Tue Jan 11 16:24:28 2022
Segmentation fault
dietpi@DietPi:~/SBFspot/SBFspot$
```

kriznik, 2022-01-11:
seems like segmentation fault is gone when **`NameplateModel`** is commented out

remknul, 2022-01-12:
Let me just let you know I experience the same bug. But do know I just pulled the source code, then ported it to FreeBSD and compile everything using the CLANG compiler. Currently debugging the issue with lldb. Trying to understand what the code does. 
In my case it also crashes at NameplateModel.
More info later.

Let me know if I can help.

remknul, 2022-01-12:
Oke, this is what I see (slightly more detail):

I run "SBFspot -v -finq -nocsv"
The I set a breakpoint at SBFspot.cpp:2672

First break:

type = TypeLabel
ii = 121
(BYTE *) recptr = 0x00000000002b7209 "\x01 \x82\b\x87\x8e?a\x8a#"
lri = NameplateModel

I continue, this goes fine, but then it reaches line 2672 again and breaks again. 
All parameters the same, except:

ii = 161
(BYTE *) recptr = 0x00000000002b7231 "\x01 \x82\b\x87\x8e?a???

The hitting continue again to actually execute line 2672 again results in SIGSEGV:

* thread #1, name = 'SBFspot', stop reason = signal SIGSEGV: invalid address (fault address: 0x0)
    frame #0: 0x00000000002818e3 SBFspot`getInverterData(devList=0x00000008010e4000, type=TypeLabel) at SBFspot.cpp:2672:101
   2669                                     break;
   2670
   2671                                 case NameplateModel: //INV_TYPE
-> 2672                                     devList[inv]->DeviceType = tagdefs.getDesc(getattribute(recptr).front());
   2673                                     if (devList[inv]->DeviceType.empty())
   2674                                     {
   2675                                         devList[inv]->DeviceType = "UNKNOWN TYPE";

sbf-, 2022-01-12:
OK, I know what's going on... trying to fix the getattribute() function
Thanks @remknul and @kriznik

remknul, 2022-01-12:
Oke, no need to debug any further? Can we wait for a fix? 
Let me know if any help needed.

Anybody interested in the patches to allow for compilation on FreeBSD?

Gunhawk007, 2022-01-12:
OK Great, please let me know when its fixed - happy to test it on my site

sbf-, 2022-01-12:
> Anybody interested in the patches to allow for compilation on FreeBSD?

Should be possible to include changes in the source...
```
#if defined(freebsd)
    // some freeBSD stuff
#endif
```

remknul, 2022-01-12:
> > Anybody interested in the patches to allow for compilation on FreeBSD?
> 
> Should be possible to include changes in the source...
> 
> ```
> #if defined(freebsd)
>     // some freeBSD stuff
> #endif
> ```

Yes, that's how I did it. Mostly changes in compiler settings, and system include files, and locations.
So the SBFspot binary now compiles correctly under FreeBSD. I cannot test bluetooth however.

closed 2022-01-12 by sbf-

remknul, 2022-01-12:
Running without a core dump for me too! Thanks for this very quick fix. Appreciated ! :)

sbf-, 2022-01-12:
Thanks to you all for debugging and testing

kriznik, 2022-01-12:
perfect, works! thanks

commit 5e5984bd (2022-01-12): Fix #511 Segmentation fault on V3.9.1