SBFspot/SBFspot #514: getOperationTime returned an error: 21

codierpaste, 2022-01-10:
Hi everybody,

I am from Germany and new in github.

I have two 5000TL-20 for more than 10 years and decided to connect them to a rasperry.

The installation worked very well without any show stopper (thnx to the team!!!).

I connect them via BT with a Net ID, because I was able to connect to one inverter and grapp the data.
After this, I am still able to get some data for both inverters, but not all and I have now an error message.

The command /usr/local/bin/sbfspot.3/SBFspot -v -finq

gives me the following:

SBFspot V3.9.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Mon Jan 10 21:12:12 2022: INFO: Starting...
sunrise: 08:14
sunset : 16:13
Connecting to 00:80:25:09:B5:86 (1/10)
Initializing...
SUSyID: 125 - SessionID: 890090174 (0x350DB2BE)
SMA netID=04
SUSyID: 78 - SN: 2100064253
SUSyID: 78 - SN: 2100060752
BT Signal=72.5%
Logon OK
Local Host Time: 10/01/2022 21:12:14
Plant Time     : 10/01/2022 21:12:14 (0 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 10/01/2022 21:11:38
SUSyID: 78 - SN: 2100064253
Device Name:      SN: 2100064253
Device Class:     Solar-Wechselrichter
Device Type:      SB 5000TL-20
Software Version: 03.20.01.R
Serial number:    2100064253
SUSyID: 78 - SN: 2100060752
Device Name:      SN: 2100060752
Device Class:     Solar-Wechselrichter
Device Type:      SB 5000TL-20
Software Version: 03.20.01.R
Serial number:    2100060752
Packet status: 21
SUSyID: 78 - SN: 2100064253
Device Status:      Ok
SUSyID: 78 - SN: 2100060752
Device Status:      Ok
SUSyID: 78 - SN: 2100064253
Device Temperature: 0.0°C
SUSyID: 78 - SN: 2100060752
Device Temperature: 0.0°C
SUSyID: 78 - SN: 2100064253
GridRelay Status:      Information liegt nicht vor
SUSyID: 78 - SN: 2100060752
GridRelay Status:      Information liegt nicht vor
Packet status: 21
**getOperationTime returned an error: 21**
SUSyID: 78 - SN: 2100064253

I check the configuration regarding MIS support and swaped the BT adress without any success.

Thank you and best regards,

Henning

sbf-, 2022-01-11:
21 is not really an error, but an "Information not available" status.
This has been changed recently and I agree it's not a very clear message.
Do you have the same output during the day?

codierpaste, 2022-01-11:
Hello,

yes I checked, and it is a little bit... weired.

All information are now available, even the Total of each inverter, that I missed yesterday.

Could it be, that after dawn the BT net doesn't work correctly?
I keep an eye on this and give feedback soon.

Thank you very much so far!

pi@SunPI:~ $ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv
SBFspot V3.9.1
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Tue Jan 11 14:29:33 2022: INFO: Starting...
sunrise: 08:13
sunset : 16:14
Connecting to 00:80:25:09:B5:86 (1/10)
Initializing...
SUSyID: 125 - SessionID: 856091264 (0x3306EA80)
SMA netID=04
SUSyID: 78 - SN: 2100064253
SUSyID: 78 - SN: 2100060752
BT Signal=71.8%
Logon OK
Local Host Time: 11/01/2022 14:29:36
Plant Time     : 11/01/2022 14:29:35 (-1 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 08/01/2022 05:05:03
SUSyID: 78 - SN: 2100064253
Device Name:      SN: 2100064253
Device Class:     Solar-Wechselrichter
Device Type:      SB 5000TL-20
Software Version: 03.20.01.R
Serial number:    2100064253
SUSyID: 78 - SN: 2100060752
Device Name:      SN: 2100060752
Device Class:     Solar-Wechselrichter
Device Type:      SB 5000TL-20
Software Version: 03.20.01.R
Serial number:    2100060752
Packet status: 21
SUSyID: 78 - SN: 2100064253
Device Status:      Ok
SUSyID: 78 - SN: 2100060752
Device Status:      Ok
SUSyID: 78 - SN: 2100064253
Device Temperature: 42.2°C
SUSyID: 78 - SN: 2100060752
Device Temperature: 42.0°C
SUSyID: 78 - SN: 2100064253
GridRelay Status:      Geschlossen
SUSyID: 78 - SN: 2100060752
GridRelay Status:      Geschlossen
SUSyID: 78 - SN: 2100064253
Energy Production:
        EToday: 4.036kWh
        ETotal: 64544.686kWh
        Operation Time: 53126.06h
        Feed-In Time  : 52178.71h
SUSyID: 78 - SN: 2100060752
Energy Production:
        EToday: 4.010kWh
        ETotal: 64456.473kWh
        Operation Time: 53189.96h
        Feed-In Time  : 52244.67h
SUSyID: 78 - SN: 2100064253
DC Spot Data:
        MPPT 1 Pdc:   0.155kW - Udc: 361.66V - Idc:  0.430A
        MPPT 2 Pdc:   0.155kW - Udc: 360.77V - Idc:  0.431A
        Calculated Total Pdc:   0.310kW
AC Spot Data:
        Phase 1 Pac :   0.307kW - Uac: 240.37V - Iac:  1.281A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.307kW - Calculated Pac:   0.307kW
        Efficiency  :   99.03%
SUSyID: 78 - SN: 2100060752
DC Spot Data:
        MPPT 1 Pdc:   0.158kW - Udc: 361.03V - Idc:  0.440A
        MPPT 2 Pdc:   0.155kW - Udc: 360.69V - Idc:  0.430A
        Calculated Total Pdc:   0.313kW
AC Spot Data:
        Phase 1 Pac :   0.307kW - Uac: 240.73V - Iac:  1.277A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.307kW - Calculated Pac:   0.307kW
        Efficiency  :   98.08%
SUSyID: 78 - SN: 2100064253
Grid Freq. : 49.99Hz
SUSyID: 78 - SN: 2100060752
Grid Freq. : 49.99Hz
SUSyID: 78 - SN: 2100064253
Current Inverter Time: 11/01/2022 14:29:35
Inverter Wake-Up Time: 11/01/2022 08:48:17
Inverter Sleep Time  : 11/01/2022 14:29:36
SUSyID: 78 - SN: 2100060752
Current Inverter Time: 11/01/2022 14:29:34
Inverter Wake-Up Time: 11/01/2022 08:48:12
Inverter Sleep Time  : 11/01/2022 14:29:39
********************
* ArchiveDayData() *
********************
startTime: 11/01/2022 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime: 01/01/2022 12:00:00
Reading events: 2022-Jan-01
Tue Jan 11 14:29:40 2022: INFO: Done.

sbf-, 2022-02-17:
No more info received