SBFspot/SBFspot #519: SMA2500TLST values wrong in CSV

aoverweel, 2022-02-03:
Today I noticed that my older SMA 2500 TSLT single tracker read-outs are not put in correct cells on the export CSV. It seems that the values for Pdc2, Idc2 are shifted causing a conflict with the upload to PV output.

sbf-, 2022-02-03:
What version of SBFspot are you using?
Upload to PVoutput does not use CSV files, but SQLite/MySQL

aoverweel, 2022-02-04:
I'm currently using v3.9.2 with MySQL support. I use a combi with Domoticz and PVOutput. I have 3 inverters, 2x 3.6 with 2 MPP and 1x 2.5 with 1MPP. I noticed a lower output in Domoticz and after investigation, the SMA 2.5 is missing the Pdc2, Idc2 and Udc2 values. Then all values are shifted. If there set to a zero then it should be fine.

d/MM/yyyy HH:mm:ss	DeviceName	DeviceType	Serial	Pdc1	Pdc2	Idc1	Idc2	Udc1	Udc2	Pac1
3/2/2022 12:49	SB3.6-1AV-40 704	Sunny Boy 3.6	1992031704	0,000	194,000	0,000	0,950	0,000	204,200	184,000
3/2/2022 12:49	SB 2500TLST-21 576	SB 2500TLST-21	2140020576	143,000		0,608		236,700		140,000

commit e0e38ff7 (2022-02-25): Fix #519 SMA2500TLST values wrong in CSV
closed 2022-02-25 by sbf-