SBFspot/SBFspot #521: Regression: BT Signal always 0

brevilo, 2022-02-12:
Hi, after upgrading from `V3.8.2` to current `master` I noticed that "BT Signal" is always 0, for both inverters. Bisecting the changes since `V3.8.2` revealed that commit 116fa95f is responsible for the regression. I'll have a closer look at it but @sbf- is surely quicker than me.

Thanks

brevilo, 2022-02-12:
Looking at the commit above the only possibly relevant [changes](https://github.com/SBFspot/SBFspot/commit/116fa95f4004f745ecf297bbd1218745299df4b3#diff-d0e7af58c70123136ea8c609848273dd26bdc5872188209abfbbd1695aa44fe3) are those to `getInverterData()`, I reckon.

sbf-, 2022-02-13:
`getBT_SignalStrength()` is called during initialization phase and is not using `getInverterData()`, so it's almost impossible these changes are responsible for the issue.
Can you try by connecting to your second inverter (change BT address in config file) It seems another device responds to the request. Also, a full debuglog (-d5 -v5) might be usefull.
I can't reproduce it, but I only have 1 inverter.

brevilo, 2022-02-13:
> so it's almost impossible these changes are responsible for the issue

Agreed, hence my "I don't see how" above. Yet I can reproduce the issue reliably (100% so far) when switching between the two commits.

> Can you try by connecting to your second inverter 

Apprently no, SBFSpot can't connect to the second inverter (yet `hcitool scan` finds both reliably). Is this even meant to work? I remember vaguely that the two inverters are somehow daisy chained when it comes to BT queries. Doesn't only one act as root/parent, being the only one to query directly?

Anyway here are the augmented logs, using the same MAC as before (the root device):

Debug log using https://github.com/SBFspot/SBFspot/commit/116fa95f4004f745ecf297bbd1218745299df4b3 (bad):
```
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3B 00 45 0A 71 DA 7D 1A 00
00000010: FF FF FF FF FF FF 01 00 7E FF
00000020: 03 60 65 08 A0 FF FF FF FF FF
00000030: FF 00 03 7D 5D 00 C8 E7 3C 32
00000040: 00 03 00 00 00 00 03 80 0E 01
00000050: FD FF FF FF FF FF 66 B9 7E
59 Bytes sent
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 14 00 6A 0A 71 DA 7D 1A 00
00000010: 43 68 61 53 74 74 03 00 05 00
20 Bytes sent
getPacket(4)
Received 18 bytes
Received 9 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 79 62 18 25 80 00
00000010: 0A 71 DA 7D 1A 00 07 00 08 00
00000020: 03 43 68 61 53 74 74
Wrong sender: <ROOT/DEVICE0-MAC>
Timeout reading socket
No data!
BT Signal=0.0%
```

Debug log using https://github.com/SBFspot/SBFspot/commit/4f910a9c0e087354a4c69a53f87cab0291f20fc2 (good, parent of https://github.com/SBFspot/SBFspot/commit/116fa95f4004f745ecf297bbd1218745299df4b3):
```
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3B 00 45 0A 71 DA 7D 1A 00
00000010: FF FF FF FF FF FF 01 00 7E FF
00000020: 03 60 65 08 A0 FF FF FF FF FF
00000030: FF 00 03 7D 5D 00 D8 5D 1F 38
00000040: 00 03 00 00 00 00 03 80 0E 01
00000050: FD FF FF FF FF FF 65 88 7E
59 Bytes sent
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 14 00 6A 0A 71 DA 7D 1A 00
00000010: 79 62 18 25 80 00 03 00 05 00
20 Bytes sent
getPacket(4)
Received 18 bytes
Received 6 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 18 00 66 79 62 18 25 80 00
00000010: 0A 71 DA 7D 1A 00 04 00 05 00
00000020: 00 00 A3 00
cmd=4
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 18 00 66 79 62 18 25 80 00
00000010: 0A 71 DA 7D 1A 00 04 00 05 00
00000020: 00 00 A3 00
<<<=================================>>>
BT Signal=63.9%
```

I'm wondering about the `Wrong sender` in case of failure. What exactly does that mean? The MAC address printed there is the one of the root device (device 0), so it doesn't seem wrong to me...

brevilo, 2022-02-13:
> I'm wondering about the `Wrong sender` in case of failure. What exactly does that mean? The MAC address printed there is the one of the root device 

Hm, `isValidSender()` checks the *first* argument passed to it for validity. Why are you printing the second argument with the `Wrong sender` message(s)?

brevilo, 2022-02-13:
> `getBT_SignalStrength()` is called during initialization phase and is not using `getInverterData()`, so it's almost impossible these changes are responsible for the issue.

Update: I partially reverted https://github.com/SBFspot/SBFspot/commit/116fa95f4004f745ecf297bbd1218745299df4b3 and kept the hex format string changes as is. It works. This leaves the device name changes. Maybe some kind of memory corruption...? Stay tuned...

brevilo, 2022-02-13:
> `getBT_SignalStrength()` is called during initialization phase and is not using `getInverterData()`, so it's almost impossible these changes are responsible for the issue.

Update: I partially reverted https://github.com/SBFspot/SBFspot/commit/116fa95f4004f745ecf297bbd1218745299df4b3 and kept the hex format string changes as is. It works. This leaves the device name changes. Maybe some kind of memory corruption...?

Apart from the CSV export there are only two write statements: https://github.com/SBFspot/SBFspot/commit/116fa95f4004f745ecf297bbd1218745299df4b3#diff-d0e7af58c70123136ea8c609848273dd26bdc5872188209abfbbd1695aa44fe3L2690 and https://github.com/SBFspot/SBFspot/commit/116fa95f4004f745ecf297bbd1218745299df4b3#diff-cee52afde59db3f806f2c6990e43799796800d5505fc23ac28cd215c610d3191L105. The second one is guarded correctly as far as I can see, so it's probably the first.

sbf-, 2022-02-13:
In case of wrong sender, SBFspot sends to `74:74:53:61:68:43` (AFAIK MAC addresses for SMA should start with [00:80:25](https://coffer.com/mac_find/?string=stollman) or [00:15:BB](https://coffer.com/mac_find/?string=sma+solar))
When good, the destination is `00:80:25:18:62:79` which is a valid MAC
So, the path of memory corruption can be something to look at.

sbf-, 2022-02-14:
Hmm, wrong sender `74:74:53:61:68:43` translates to `ChaStt` which is `State of Charge` in the taglist files...

brevilo, 2022-02-14:
Is this by any chance part of the `InverterData` struct? I've come to suspect that adding the variable length `std::string DeviceName` for the fixed length `char[]` might mess with the remaining content (considering packed and/or aligned memory as well). I pretty much ruled out the other DeviceName-related changes and memory layout changes like this can lead to this kind of issue if there are any assumptions of it still being made elsewhere in the code.

brevilo, 2022-02-14:
Ok, I'm sorry but I think we can forget about all of the above reasoning as the memory corruption I saw probably was due to screw-up of mine - I didn't run a `make clean` before I upgraded and during all my tests, assuming that `make` would take care of rebuilding everything relevant.

Now that I cleansed the build tree I see a different cause for the problem: the root device (device 0) isn't the one I connect to (but the second inverter) which is probably why I always get a BT signal of 0 (I've always only gotten the signal for the root device so far). Unfortunately, the old version I used is also affected by now, so I don't have a "good" version for bisecting the codebase. No idea yet how to track this down.

I'll open a separate issue to not confuse people more than it already happened...

closed 2022-02-14 by brevilo

brevilo, 2022-02-15:
tl;dr: current [master](https://github.com/SBFspot/SBFspot/commit/5455dd4b69ecdfdbd425de8acc488c1edee012b3) works just fine! This issue was a red herring...

Background: this issue was really down to the dirty build. The subsequent root device confusion (see previous post) was probably just a separate artifact that manifested itself while I was debugging the original issue. I got locked out of the inverter's user account at some point (messed with the data packets) and had to use SE to regain access. I presume the plant network's inverter chain got confused at that point. Long story short, the next day (after plant boot-up) everything returned to normal.

brevilo, 2022-02-15:
Yep, that's known and the other signal is usually at 100% anyway :)

Minor reminder: have a look at this [comment](https://github.com/SBFspot/SBFspot/issues/521#issuecomment-1038177412). Maybe that's something (unrelated) for improvement...

sbf-, 2022-02-15:
> Minor reminder: have a look at this [comment](https://github.com/SBFspot/SBFspot/issues/521#issuecomment-1038177412). Maybe that's something (unrelated) for improvement...

I'll have a look at it... that's very old code ;-)

sbf-, 2022-02-24:
> Hm, `isValidSender()` checks the _first_ argument passed to it for validity. Why are you printing the second argument with the `Wrong sender` message(s) instead of the first? This is unrelated to issue at hand, but you probably want to verify that you print the correct MAC.

That looks OK to me. First (MAC inverter in config) and second (MAC responding device) are compared. If different, print second (MAC responding device)