SBFspot/SBFspot #523: Bad Data uploaded to PVOutput

SjlCairns, 2022-02-22:
Since 16-2-2022, PVOutput has been receiving wrong solar data just after midnight.  See PVOutput site "Fred Home Power".
Since 16-2-2022, table vwDayData has data in the Power column from Midnight to 5.40am.  This data is a replica of data in vwDayData from 1.50pm the previous day.  Table vwSpotData seems OK.
Log file "MyPlant_20220221.log" (as an example) has a "Calculated EToday : 8.724kwhrs" entry from 6am to approx 6.30am (every 5 mins until the Grid Relay is closed).
The system is running on a Pi with Bullseye, and SBFspot version 3.9.1, sqlite3 = V3.34.1 & Boost = v1.74.0
The location timezone is UTC + 10 Hrs (Queensland- Australia).  The log file attached indicates the time is correct on the Pi and Inverter.  Inverter is SMA SB-5000TL
Current Database is attached if it might help.

A similar problem surfaced with this same inverter and an earlier version of sbfspot about 1 year ago.  At that time, I modified the view vwPvoData to exclude any times in the TimeStamp column that were at night.  About a month ago, a new Pi and latest version of SBFspot were setup, and it has been running well until 16-2-2022.  I can go back in and change vwPvoData again, but thought you may have an easier solution.
Any Help is much appreciated.
Thanks for your great program.
Cheers,
Sid.

[MyPlant_20220221.log](https://github.com/SBFspot/SBFspot/files/8113831/MyPlant_20220221.log)
[SBFspot.zip](https://github.com/SBFspot/SBFspot/files/8113845/SBFspot.zip)
.

sbf-, 2022-02-22:
I am aware of a similar problem [here](https://gathering.tweakers.net/forum/list_message/62758534#62758534) - Sorry, in Dutch
This seems to be an inverter issue. To be sure, you should check with sunnyexplorer.

SjlCairns, 2022-02-23:
Thanks for your comments.  The site is at a friends holiday home about 250km away, so I cannot use sunnyexplorer.
I do have remote access to the Pi, and have now altered the vwPvoData view slightly to temporarily overcome the problem.
I think you are correct about it being an Inverter problem.  I have setup 3 other friends with the same inverter, and all are working great with your program.
At my age, I rely on notes to help me remember what I have done, and why.  Attached are my notes about SBFspot problems, and my solutions.  Thanks again for making SBFspot available - it is great.
Sid.
[Raspberry Pi - SBFspot Tweaks for Bad Data from Inverter.txt](https://github.com/SBFspot/SBFspot/files/8122456/Raspberry.Pi.-.SBFspot.Tweaks.for.Bad.Data.from.Inverter.txt)

closed 2022-02-23 by SjlCairns