SBFspot/SBFspot #524: installation poblem

wil76, 2022-02-22:
sorry, I'm not a linux specialist and running into a setup problem for sqlite:

 [smarthome@SmartHomeNG2 /home]$ curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config|sudo bash
SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 3 Model B Plus Rev 1.3
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=smarthome
userhome=/home/smarthome
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.2)
Searching for existing configuration(s)...
Reading defaults from /var/tmp/sbfspot.3/sbfspot.defaults... Done
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.2/sbfspot-sqlite-arm-linux-buster.tar.gz
Copying files...
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  rpi-eeprom-images
Verwenden Sie »apt autoremove«, um es zu entfernen.
Vorgeschlagene Pakete:
  sqlite3-doc
Die folgenden NEUEN Pakete werden installiert:
  sqlite3
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 28 nicht aktualisiert.
Es müssen 839 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 2.278 kB Plattenplatz zusätzlich benutzt.
Fehl:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf sqlite3 armhf 3.27.2-3
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Fehlschlag beim Holen von http://raspbian.raspberrypi.org/raspbian/pool/main/s/sqlite3/sqlite3_3.27.2-3_armhf.deb 404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Einige Archive konnten nicht heruntergeladen werden; vielleicht »apt-get update« ausführen oder mit »--fix-missing« probieren?
sqlite3 installation failed!

sbf-, 2022-02-22:
Try sudo apt --fix-missing

wil76, 2022-02-22:
worked but now I don't get a connection.
is this still during setup, or is it allready installed ? 

Running SBFspot...
SBFspot V3.9.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Tue Feb 22 12:30:23 2022: INFO: Starting...
sunrise: 06:50
sunset : 17:25
Connecting to 00:80:25:1D:2A:0E (1/10)
Connecting to 00:80:25:1D:2A:0E (2/10)
Connecting to 00:80:25:1D:2A:0E (3/10)
Connecting to 00:80:25:1D:2A:0E (4/10)
Connecting to 00:80:25:1D:2A:0E (5/10)
Connecting to 00:80:25:1D:2A:0E (6/10)
Connecting to 00:80:25:1D:2A:0E (7/10)
Connecting to 00:80:25:1D:2A:0E (8/10)
Connecting to 00:80:25:1D:2A:0E (9/10)
Connecting to 00:80:25:1D:2A:0E (10/10)
Tue Feb 22 12:30:32 2022: CRITICAL: bthConnect() returned -1
Tue Feb 22 12:30:32 2022: INFO: Done.
Installation aborted

sbf-, 2022-02-22:
This is during install.
SBFspot is installed without PVOutput upload (if requested)
Try moving the rpi closer to the inverter. What rpi is it?
You can rerun SBFspot with /usr/local/bin/sbfspot.3/SBFspot -v5

wil76, 2022-02-22:
I'm using a: Raspberry Pi 3 Model B+ v1.3, 1GB (Sony, UK)
I already places it close to the inverter in my garage.
in house I have a repeater between the garage and my office.
as I remember it has already been working several years ago without the repeater.
with -v5 it lists the config papameters:

 [smarthome@SmartHomeNG2 ~]$ /usr/local/bin/sbfspot.3/SBFspot -v5 -finq -nocsv
SBFspot V3.9.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:1D:2A:0E
Password=<undisclosed>
MIS_Enabled=1
Plantname=MyPlant
OutputPath=/home/smarthome/smadata/%Y
OutputPathEvents=/home/smarthome/smadata/%Y/Events
Latitude=48.21
Longitude=16.37
Timezone=Europe/Vienna
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/smarthome/smadata/SBFspot.db
### End of Config ###
Tue Feb 22 14:15:25 2022: INFO: Starting...
sunrise: 06:50
sunset : 17:25
Connecting to 00:80:25:1D:2A:0E (1/10)
Connecting to 00:80:25:1D:2A:0E (2/10)
Connecting to 00:80:25:1D:2A:0E (3/10)
Connecting to 00:80:25:1D:2A:0E (4/10)
Connecting to 00:80:25:1D:2A:0E (5/10)
Connecting to 00:80:25:1D:2A:0E (6/10)
Connecting to 00:80:25:1D:2A:0E (7/10)
Connecting to 00:80:25:1D:2A:0E (8/10)
Connecting to 00:80:25:1D:2A:0E (9/10)
Connecting to 00:80:25:1D:2A:0E (10/10)
Tue Feb 22 14:15:34 2022: CRITICAL: bthConnect() returned -1
Tue Feb 22 14:15:34 2022: INFO: Done.
 [smarthome@SmartHomeNG2 ~]$

sbf-, 2022-02-22:
It looks like your BT is used by another process.
The time between `Tue Feb 22 14:15:25 2022: INFO: Starting...` and `Tue Feb 22 14:15:34 2022: CRITICAL: bthConnect() returned -1` is too short (only 9 seconds)
This should be approx 50 seconds (10 x 5 seconds timeout)
So I don't think it's an SBFspot issue.

wil76, 2022-02-22:
sorry, you were right, many thanks for your assistance. 
now it's completed. 
now I will complete my setup and try to realize my monitoring via an MQTT integration. any special hint in this area ?

sbf-, 2022-02-24:
> now I will complete my setup and try to realize my monitoring via an MQTT integration. any special hint in this area ?

There is some info in the [wiki](https://github.com/SBFspot/SBFspot/wiki/MQTT)

closed 2022-02-24 by sbf-