SBFspot/SBFspot #527: Error: [event_data]mysql_stmt_execute() returned : Data too long for column 'OldValue' at row 1

deagel, 2022-03-02:
Hi,
i found this log output, and think its the reason to miss data for pvupload. can you help me to fix it?
[debug.log](https://github.com/SBFspot/SBFspot/files/8167923/debug.log)

silliewimons, 2022-03-02:
Dear deagel,   your system has three inverters,     i suggest that you install as a first step version 3.9.3
there are some improvements  that  might have an impact with multi-inverter installations 

can you modify the settings   and  apply/modify  debug options   -d5  -v5     (instead of -v -d4)

thank you   sillie

deagel, 2022-03-02:
hi  sillie,
did not help.
[debug-02.log](https://github.com/SBFspot/SBFspot/files/8168698/debug-02.log)

thanks

deagel

sbf-, 2022-03-02:
This error is not related to pvoutput. Take a look at the wiki pages

sbf-, 2022-03-02:
To get rid of the error, increase the size of OldVal/NewVal in the Eventdata table to 200

deagel, 2022-03-02:
Hi, 200 worked for me thanks for your fast support.

silliewimons, 2022-03-09:
dear  deagel,   in your output log   i could read that in your system

the AC-output from  your  mono-phase inverters was assigned to  a  dedicated phase 
sbfspot  assigns ACoutput  standard  to  phase1

can you   provide  more information on the way that you did  realise this,   
see part: discussions  item #526   this point is adressed,

please  reply  by adding the information to  discussion - 526  
kind regards  silliew

commit 2287f3a8 (2022-05-26): Fix #527 Data too long for column 'OldValue'
closed 2022-05-26 by sbf-