SBFspot/SBFspot #529: ISSUE installing after firmware upgrade of Raspberry - Debian 11 - Bullseye 64bit

geurtsd, 2022-03-11:
Hi,

After running a cleanup-install and upgrading the firmware of the Raspberry, the install scritp no longer works.
- upgrade with command: 
```
sudo rpi-update
```

It seems that with teh new firmware the output of the command 
```
lsb_release -si
```
no longer gives back "Raspbian", but it defaults to the standard "Debian".

I have tested the install script by modifying this line in your install script, and it installs, yet is not able to run SBFspot afterwards.
it gives error:

> pi@SolarLogger:/ $ sudo bash script
>SBFspot Configuration Tool V1.2.1
>Running on Raspberry Pi 3 Model B Rev 1.2
>Checking whiptail... Installed.
>architecture=arm64
>Debian GNU/Linux 11 (bullseye)
>username=pi
>userhome=/home/pi
>Minimum SBFspot version supported by this tool is 3.5.0
>Request to install latest version (3.9.3)
>Searching for existing configuration(s)...
>Reading settings from /usr/local/bin/sbfspot.3/SBFspot.cfg
>Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
>Performing some checks before installation... Done
>Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.3/sbfspot-sqlite-arm-linux-bullseye.tar.gz
>Copying files...
>SBFspot.db exists
>Checking database...
>Schema Version=1
>Saving existing SBFspot.cfg... Done
>Configuring SBFspot... Done
>Running SBFspot...
>script: line 1322: /usr/local/bin/sbfspot.3/SBFspot: No such file or directory
>
>Installation aborted

However the listed install directory is present and there is the binary file "SBFspot".... (it shows up with "ls")
>pi@SolarLogger:/usr/local/bin/sbfspot.3 $ ls
>date_time_zonespec.csv  monthdata  SBFspot.cfg                  SBFspot.cfg.20220311_141416  TagListEN-US.txt  TagListFR-FR.txt  TagListNL-NL.txt
>daydata                 SBFspot    SBFspot.cfg.20220311_134058  TagListDE-DE.txt             TagListES-ES.txt  TagListIT-IT.txt
>pi@SolarLogger:/usr/local/bin/sbfspot.3 $

Manually running the file gives error:
>pi@SolarLogger:/usr/local/bin/sbfspot.3 $ SBFspot -v
> -bash: SBFspot: command not found

any ideas? compatability issue?

geurtsd, 2022-03-11:
Could it be a 32 - 64 bit issue?
>pi@SolarLogger:/usr/local/bin/sbfspot.3 $ file SBFspot
>SBFspot: ELF ****32-bit** LSB executable**, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=21867e6f7f0f61390e4450e10890f284a175b878, for GNU/Linux 3.2.0, stripped

sbf-, 2022-03-11:
> Could it be a 32 - 64 bit issue?

Sure it is... pre-compiled binaries are 32-bit
You'll have to follow the wiki to compile it for now

geurtsd, 2022-03-11:
yea, uncovered it myself also via 
```
objdump -x SBFspot
```
several 32-bit NEEDED libraries.....

so this confirms my feeling that it are 2 problems in 1 ticket:
it answers the 32/64 bit issue, but the script item remains....

I'll do a reinstall of 32Bit Raspberry and will confirm after this .....

sbf-, 2022-03-11:
> Manually running the file gives error:
> 
> > pi@SolarLogger:/usr/local/bin/sbfspot.3 $ SBFspot -v
> > -bash: SBFspot: command not found

you should run it like this: `./SBFspot -v`

geurtsd, 2022-03-11:
> > Manually running the file gives error:
> > > pi@SolarLogger:/usr/local/bin/sbfspot.3 $ SBFspot -v
> > > -bash: SBFspot: command not found
> 
> you should run it like this: `./SBFspot -v`

I know, tried it also, same result...... (to lazy to adjust / change the command for post ;-))

geurtsd, 2022-03-12:
UPDATE:

Did full reinstall of RPi on bullseye 32bit.

his did the trick and resolved everything..... also the install script error. Some strange and confuzing decision that they (RPi) update the system's release only for 32bit and not for 64bit on their Pi OS. (maybe because they switched over from Raspbian to PI OS in their branding?)

Anyhow, Ticket can be closed. Maybe a point of improvement (yet not urgent): include a block to check for the architecture of the installed OS with one of following commands:
```
uname -m
arch
lscpu
```
to generate an error that 64bit is not supported with the script.
It does not solve the issue (= PICNIK problem) but gives a clearer hint why.... 

thanks for the fast support and sorry for the trouble... ;-)

PS: for those who do not know what a _PICNIK_ is: 
_Problem In Chair, Not In Keyboard_ (so USER problem ;-))

sbf-, 2022-03-12:
Thanks for the feedback.
I'm already using `dpkg --print-architecture` which returns `armhf` or `arm64` but I'm not using it ATM