SBFspot/SBFspot #532: SBFspotUpload service stops when remote SQL server is temporarily down

pagaille, 2022-04-09:
Hi,

I use SBFspotUploadDaemon with a remote MariaDB database. The service stops and must be manually restarted when it can't connect to the DB. It should stay alive and try again later.

Txs

sbf-, 2022-04-09:
Are you sure? See #415

pagaille, 2022-04-10:
I'm sure of the behaviour I reported. See by yourself : 

```
Apr 10 10:16:40 smalogger.lan systemd[1]: Started SBFspot Upload Daemon.
Apr 10 10:16:40 smalogger.lan SBFspotUploadDaemon[10360]: SBFspotUploadDaemon Version 3.0.1
Apr 10 10:16:40 smalogger.lan SBFspotUploadDaemon[10360]: Loglevel=2
Apr 10 10:16:40 smalogger.lan SBFspotUploadDaemon[10360]: LogDir=/var/log/sbfspot.3/
Apr 10 10:16:40 smalogger.lan SBFspotUploadDaemon[10360]: [2022-04-10 10:16:40.741] Error: Can't open MySQL db [SBFspot] : Can't connect to MySQL
Apr 10 10:16:40 smalogger.lan SBFspotUploadDaemon[10360]: Unable to open database. Check configuration.
Apr 10 10:16:40 smalogger.lan systemd[1]: SBFspotUpload.service: Main process exited, code=exited, status=1/FAILURE
Apr 10 10:16:40 smalogger.lan systemd[1]: SBFspotUpload.service: Failed with result 'exit-code'.
```

Why do you stop the service ? The OP in https://github.com/SBFspot/SBFspot/issues/415 complains about the missing logs in case of connection problem, but didn't ask for the service to stop.

Or is there something I don't get ?

pagaille, 2022-04-11:
Again yesterday : 

```
|[20:00:33.478] INFO: Uploading datapoint: 20220410,20:00,289148,12,,,,235.40 => OK (200)                       │
│[20:06:36.124] INFO: Uploading datapoint: 20220410,20:05,289148,0,,,,235.40 => OK (200)                        │
│[20:10:31.076] INFO: Uploading datapoint: 20220410,20:10,289149,12,,,,235.40 => OK (200)                       │
│[20:16:32.820] INFO: Uploading datapoint: 20220410,20:15,289149,0,,,,236.20 => OK (200)                        │
│[20:31:03.937] ERROR: Failed to open database: Unknown MySQL server host 'mattlabs.lan' (-3)                   │
│[20:31:03.938] INFO: Stopping Daemon...                                                                        │
│[20:31:14.231] INFO: SBFspotUploadDaemon Version 3.0.1 
```

pagaille, 2022-04-11:
Possible solution : 

https://github.com/SBFspot/SBFspot/blob/d340dc51161f9bedfffcac3eb20437fc551a75d6/SBFspotUploadDaemon/SBFspotUpload.service#L10

Replace by 
```

Restart=always
RestartSec=60
```

What do you think ?

sbf-, 2022-04-12:
> What do you think ?

Sounds reasonable to me, but I think you're in the best position to test.
The system file was written [based on this](https://unix.stackexchange.com/questions/451855/run-user-uid-not-mounted-when-daemon-starts) and TBH, I never understood the `Restart=on-success` thing

closed 2024-02-10 by sbf-

pagaille, 2024-02-10:
👍🙏

commit 23fa336e (2024-02-10): Fix #532 SBFspotUpload service stops when remote SQL server is temporarily down