SBFspot/SBFspot #534: Segmentation fault V3.9.4

Koen568, 2022-04-22:
Hallo,
Installed bullseye 32bit on new Raspberry pi.
/usr/local/bin/sbfspot.3/SBFspot -v5 -finq -nocsv
Gives me :

SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
LocalBTAddress=00:80:25:21:A2:21
Password=<undisclosed>
Plantname=Rust37
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=51.15
Longitude=4.33
Timezone=Europe/Brussels
Locale=en-US
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=0
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/pi/smadata/SBFspot.db
End of Config

Fri Apr 22 16:13:37 2022: INFO: Starting...
sunrise: 06:33
sunset : 20:48
Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 871262889
Logon OK
Segmentation fault

How can I find what is going wrong?

sbf-, 2022-04-22:
I guess your inverter is communication via bluetooth...
If so, `LocalBTAddress=00:80:25:21:A2:21` should be `BTAddress=00:80:25:21:A2:21`

Koen568, 2022-04-23:
That seem to be it, I thought I had to use LocalBTAddress as I have 2 (single faze) inverters running. 
Is there an easy way to put the readings of my second inverter on L2 as that would reflect the reality?
many thanks for your support.
Have a nice day

sbf-, 2022-04-25:
> Is there an easy way to put the readings of my second inverter on L2 as that would reflect the reality?

Not for CSV/MQTT
With SQL you can create a view (or modify vwSpotyData) to do this

closed 2022-04-26 by sbf-