SBFspot/SBFspot #535: Bluetooth connection is not working

Johannes106, 2022-04-22:
Hello,

thx for this great tool.
I use a raspberry pi and use this tool to get out the data of my inverters. I don't export the data to any location but i use the generated csv-file (by NOSQL) to import the values into a RRD.
After changing the config (by curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash) I can't establish to my inverters (by bluetooth). I have reduced the distance to my inverters and so on but without success. Is there a problem with bluetooth connections caused by current updates?
`
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash
SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.4)
Searching for existing configuration(s)...
Reading defaults from /var/tmp/sbfspot.3/sbfspot.defaults... Done
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.4/sbfspot-nosql-arm-linux-buster.tar.gz
Copying files...
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Apr 22 16:03:33 2022: INFO: Starting...
sunrise: 05:54
sunset : 20:15
Connecting to 00:80:25:09:A8:FA (1/10)
Initialising...
SUSyID: 125 - SessionID: 841398368
SMA netID=01
Fri Apr 22 16:03:34 2022: CRITICAL: Failed to initialise communication with inverter.
Fri Apr 22 16:03:34 2022: INFO: Done.
Installation aborted
`

sbf-, 2022-04-22:
Does your inverter have a very old firmware version (below 1.71)? 
What is the output of `/usr/local/bin/sbfspot.3/SBFspot -v5 -d5 -finq`

Johannes106, 2022-04-22:
`/usr/local/bin/sbfspot.3/SBFspot -v5 -d5 -finq`
```
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v5 -d5 -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
BTAddress=00:80:25:09:A8:FA
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=HouseFA
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=52.52
Longitude=13.4
Timezone=Europe/Berlin
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
End of Config

Fri Apr 22 17:42:42 2022: INFO: Starting...
sunrise: 05:54
sunset : 20:15
Connecting to 00:80:25:09:A8:FA (1/10)
Initialising...
SUSyID: 125 - SessionID: 876463173
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 9 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
<<<=================================>>>
MAX_pcktBuf is now 27 bytes
SMA netID=01
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00 
00000010: FA A8 09 25 80 00 02 00 00 04 
00000020: 70 00 01 00 00 00 00 01 00 00 
00000030: 00 
31 Bytes sent
getPacket(5)
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 0A 00 FA A8 
00000020: 09 25 80 00 01 5F D2 A5 EB 27 
00000030: B8 
cmd=10
Received 18 bytes
Received 16 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 05 00 FA A8 
00000020: 09 25 80 00 01 01 5F D2 A5 EB 
00000030: 27 B8 02 01 
cmd=5
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 05 00 FA A8 
00000020: 09 25 80 00 01 01 5F D2 A5 EB 
00000030: 27 B8 02 01 
<<<=================================>>>
MAX_pcktBuf is now 34 bytes
Local BT address: B8:27:EB:A5:D2:5F
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3F 00 41 5F D2 A5 EB 27 B8 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 09 A0 FF FF FF FF FF 
00000030: FF 00 00 7D 5D 00 45 C4 3D 34 
00000040: 00 00 00 00 00 00 02 80 00 02 
00000050: 00 00 00 00 00 00 00 00 00 00 
00000060: 19 EE 7E 
63 Bytes sent
getPacket(1)
Received 18 bytes
Received 5 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 17 00 69 FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 01 00 7E FF 
00000020: 03 60 65 
cmd=1
PacketLength=23
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 
<<<=================================>>>
Invalid chk 0xFF00 - Found 0x6003
Fri Apr 22 17:42:43 2022: CRITICAL: Failed to initialise communication with inverter.
Fri Apr 22 17:42:43 2022: INFO: Done.
```
if I use hcitool:
`hcitool scan`
```
        00:80:25:09:93:6B	SMA0001d SN: 2100063324 SN2100063324
	00:80:25:09:9B:41	SMA001d SN: 2100060901 SN2100060901
	00:80:25:09:A8:FA	SMA0001d SN: 2100068131 SN2100068131
```
and also ping works: 
`sudo l2ping 00:80:25:09:A8:FA`
```
Ping: 00:80:25:09:A8:FA from B8:27:EB:A5:D2:5F (data size 44) ...
18 bytes from 00:80:25:09:A8:FA id 0 time 10.23ms
18 bytes from 00:80:25:09:A8:FA id 1 time 17.03ms
18 bytes from 00:80:25:09:A8:FA id 2 time 23.48ms
18 bytes from 00:80:25:09:A8:FA id 3 time 12.26ms
```

sbf-, 2022-04-22:
Try with `MIS_Enabled=1` (Multi Inverter Support)

Johannes106, 2022-04-22:
yes, it worked from 20.09.2021 until 18.04.2022 and created one csv with data of one inverter (even i've enabled MIS). So I thought to use 3 different config files (one for each inverter). But since I executed the command (...sbfspot-config | sudo bash) I can't get a connection to my inverters.
Now i get a new error message:
```
SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.4)
Searching for existing configuration(s)...
Creating new config...
Scanning for Bluetooth devices, please wait...
Scanning for Bluetooth devices, please wait...
Scanning for Bluetooth devices, please wait...
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.4/sbfspot-nosql-arm-linux-buster.tar.gz
Copying files...
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Apr 22 18:36:53 2022: INFO: Starting...
sunrise: 05:54
sunset : 20:15
Connecting to 00:80:25:09:93:6B (1/10)
Initialising...
SUSyID: 125 - SessionID: 848081293
Fri Apr 22 18:36:54 2022: CRITICAL: Incompatible FW version detected.
Fri Apr 22 18:36:54 2022: INFO: Done.
Installation aborted
```

Johannes106, 2022-04-22:
How can I test the bluetooth connection by sbfspot?

silliewimons, 2022-04-23:
@all - I did control that v394 works correct on my system (windows10) 
                   only point that I'll need to investigate:
                   --due to a permanent active BTsender (SunRichxxx) i'm confronted with a high number of errros/warnings
                              -wrong sender; -packet-mismatch; -overwrtiting-records

@johannes106     from your information  your inverters have different reactions

-may i suggest to concentrate first on inverter: 00:80:25:09:A8:FA
  the easy approach is to apply a cfg  file for this inverter only  using same parameters as in earlier "operational" system,

?if that does not work - then eilas you should control whether the older (2021) environment is still working

-note the invalid FirmWare version is another issue - to handle seperate

kind regards  sillie

Johannes106, 2022-04-25:
@sbf- @silliewimons  thanks for your quick reply.
There must be a problem with my inverters. After a long research I think SBFsport works correctly. Sorry for this wrong issue.
Now I have 3 different config files and only 1 config works.
**Success**:
```
# SBFspot.cfg - Created Sun 24 Apr 18:41:12 CEST 2022 with sbfspot-config V1.2.1 and edited by johannes
BTAddress=00:80:25:09:9B:41
MIS_Enabled=1
BTConnectRetries=10
Password=0000
Plantname=garage41
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Timezone=Europe/Berlin
Latitude=52.52
Longitude=13.40
Locale=en-US
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
DecimalPoint=comma
CalculateMissingSpotValues=0
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Delimiter=semicolon
CSV_Spot_TimeSource=computer
CSV_Spot_WebboxHeader=0
# End of Config
```
**Fail**:
```
# SBFspot.cfg - Created Sun 24 Apr 18:41:12 CEST 2022 with sbfspot-config V1.2.1 and edited by johannes
BTAddress=00:80:25:09:93:6B
MIS_Enabled=1
BTConnectRetries=10
Password=0000
Plantname=garage6B
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Timezone=Europe/Berlin
Latitude=52.52
Longitude=13.40
Locale=en-US
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
DecimalPoint=comma
CalculateMissingSpotValues=0
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Delimiter=semicolon
CSV_Spot_TimeSource=computer
CSV_Spot_WebboxHeader=0
# End of Config
```
**Fail**:
```
SBFspot.cfg - Created Sun 24 Apr 18:43:55 CEST 2022 with sbfspot-config V1.2.1 and edited by johannes
BTAddress=00:80:25:09:A8:FA
MIS_Enabled=1
BTConnectRetries=10
Password=0000
Plantname=garageFA
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Timezone=Europe/Berlin
Latitude=52.52
Longitude=13.40
Locale=en-US
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
DecimalPoint=comma
CalculateMissingSpotValues=0
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Delimiter=semicolon
CSV_Spot_TimeSource=computer
CSV_Spot_WebboxHeader=0
# End of Config
```
**Errormessage:**
```
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -cfgSBFspot.cfg.20220424_FA
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg.20220424_FA'
Mon Apr 25 19:10:20 2022: INFO: Starting...
sunrise: 05:47
sunset : 20:20
Connecting to 00:80:25:09:A8:FA (1/10)
Initialising...
SUSyID: 125 - SessionID: 970435203
Mon Apr 25 19:10:21 2022: CRITICAL: Incompatible FW version detected.
Mon Apr 25 19:10:21 2022: INFO: Done.

```

silliewimons, 2022-04-25:
@johannes106   please inform us about the following,

"i refer to your text" yes, it worked from 20.09.2021 until 18.04.2022 and created one csv with data of one inverter (even i've enabled MIS). So I thought to use 3 different config files . . . . . "
?-a- do you have somewhere a copy of 'log-report' after executing  sbfspot  before 17april
can you confirm the BTadress / SerialNumber for that inverter  (Depending on the options in config the Ser.Nbr is part of header lines of your csv file)          
?-b- which type/model of inverters are you using
?-c- do you know/remember the date that each of these three  systems were installed/purchased,
?-d- before your choice to apply sbfspot, have you been using another tool to access these inverters,
        if so, which tool was this   and is that option still operational

rational:  the communication options for SMA inverters have a few times modfied, so we need to understand 
what kind of environment that you are working in.
 
kr  sillie

silliewimons, 2022-04-26:
@johannes   in addition to the above   please  provide the following  (by using full-debug-options we obtain added info) 

-J- the output log-file from sbfspot   after  execution   for each of your inverters   using the following rules
  in each config    put  MIS enabled  to  0   
  options for execution   -d5 -v5 -finq     
  capture the output and give that them individual names - example of list below   (you add files via this system)
  either bij using the redirect command to write to file example   1> logG6Bou.txt)  

- - - to make it easy to identify the outputs  either use  "id" or  "plant-name" as part of the name of each seperate file 
-  id         BT                             info                                                                        plant-name
-G6B     00:80:25:09:93:6B    SMA0001d SN: 2100063324 SN2100063324           Garage6B
-G41     00:80:25:09:9B:41    SMA001d SN: 2100060901 SN2100060901             Garage41    
-HFA     00:80:25:09:A8:FA    SMA0001d SN: 2100068131 SN2100068131           HouseFA    

kind regards   sillie

johannes, 2022-04-26:
Sorry I can't provide anything. (And please double check mentions, thanks 🤸)

Johannes106, 2022-04-27:
I'm pleased to hear about your interest of my problem. :-) 
I will try to deliver the informations you are asking for:
- **environment**: 
  - PC: `cat /sys/firmware/devicetree/base/model:` ``` Raspberry Pi 3 Model B Rev 1.2```
  - INVERTER: ```SN: 2100060901;SB 5000TL-20;2100060901```
- **installation date**: I think it was 11 years ago. but if you need the exact date i can search for it.
- **software**: I was looking  for a simple method to log the produced energy of my inverters automatically. The software of sma is only for windows but my plan is to use my raspberry pi for this. And so I found SBFSPOT. 
-  **Logs**: see below

kind regards johannes

Johannes106, 2022-04-27:
`less /var/log/sbfspot.3/smaGarage_20210910.log`:
```
SBFspot V3.8.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)

Compiled for Linux (LE) 32 bitCommandline Args: -v -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Sep 10 06:00:01 2021: INFO: Starting...
sunrise: 06:34
sunset : 19:32
Nothing to do... it's dark. Use -finq to force inquiry.
SBFspot V3.8.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Sep 10 06:05:01 2021: INFO: Starting...
sunrise: 06:34
sunset : 19:32
Nothing to do... it's dark. Use -finq to force inquiry.
SBFspot V3.8.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Sep 10 06:10:01 2021: INFO: Starting...
sunrise: 06:34
sunset : 19:32
Nothing to do... it's dark. Use -finq to force inquiry.
SBFspot V3.8.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Sep 10 06:15:01 2021: INFO: Starting...
sunrise: 06:34
sunset : 19:32
Connecting to 00:80:25:09:9B:41 (1/10)
Connecting to 00:80:25:09:9B:41 (2/10)
Connecting to 00:80:25:09:9B:41 (3/10)
Connecting to 00:80:25:09:9B:41 (4/10)
Connecting to 00:80:25:09:9B:41 (5/10)
Connecting to 00:80:25:09:9B:41 (6/10)
Connecting to 00:80:25:09:9B:41 (7/10)
Connecting to 00:80:25:09:9B:41 (8/10)
Connecting to 00:80:25:09:9B:41 (9/10)
Connecting to 00:80:25:09:9B:41 (10/10)
Fri Sep 10 06:21:02 2021: CRITICAL: bthConnect() returned -1
Fri Sep 10 06:21:02 2021: INFO: Done.
SBFspot V3.8.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2021, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Sep 10 06:25:01 2021: INFO: Starting...
sunrise: 06:34
sunset : 19:32
Connecting to 00:80:25:09:9B:41 (1/10)
Connecting to 00:80:25:09:9B:41 (2/10)
Connecting to 00:80:25:09:9B:41 (3/10)
Connecting to 00:80:25:09:9B:41 (4/10)
Connecting to 00:80:25:09:9B:41 (5/10)
Connecting to 00:80:25:09:9B:41 (6/10)
Connecting to 00:80:25:09:9B:41 (7/10)
Connecting to 00:80:25:09:9B:41 (8/10)
Connecting to 00:80:25:09:9B:41 (9/10)
Connecting to 00:80:25:09:9B:41 (10/10)
Fri Sep 10 06:26:01 2021: CRITICAL: bthConnect() returned -1
Fri Sep 10 06:26:01 2021: INFO: Done.
```

Johannes106, 2022-04-27:
`CSV-File`:
```
sep=;
Version CSV1|Tool SBFspot3.8.3 (Linux)|Linebreaks LF|Delimiter semicolon|Decimalpoint comma|Precision 3

;;;;Watt;Watt;Amp;Amp;Volt;Volt;Watt;Watt;Watt;Amp;Amp;Amp;Volt;Volt;Volt;Watt;Watt;%;kWh;kWh;Hz;Hours;Hours;%;Status;Status;degC
dd/MM/yyyy HH:mm:ss;DeviceName;DeviceType;Serial;Pdc1;Pdc2;Idc1;Idc2;Udc1;Udc2;Pac1;Pac2;Pac3;Iac1;Iac2;Iac3;Uac1;Uac2;Uac3;PdcTot;PacTot;Efficiency;EToday;ETotal;Frequency;OperatingTime;FeedInTime;BT_Signal;Condition;GridRelay;Temperature
22/10/2021 12:05:05;SN: 2100060901;SB 5000TL-20;2100060901;2265,000;2218,000;7,994;8,254;283,430;268,780;4374,000;0,000;0,000;18,152;0,000;0,000;240,950;0,000;0,000;4483,000;4374,000;97,569;6,998;80852,874;49,950;51689,589;49947,873;65,490;Ok;Closed;53,290
22/10/2021 12:10:14;SN: 2100060901;SB 5000TL-20;2100060901;2240,000;2174,000;8,024;8,354;279,220;260,330;4302,000;0,000;0,000;17,909;0,000;0,000;240,250;0,000;0,000;4414,000;4302,000;97,463;7,373;80853,248;50,000;51689,675;49947,959;66,667;Ok;Closed;56,220
22/10/2021 12:15:14;SN: 2100060901;SB 5000TL-20;2100060901;2289,000;2234,000;8,409;8,400;272,290;266,020;4406,000;0,000;0,000;18,278;0,000;0,000;241,090;0,000;0,000;4523,000;4406,000;97,413;7,737;80853,611;50,000;51689,758;49948,042;65,490;Ok;Closed;57,390

```

silliewimons, 2022-04-28:
OK,   from that info we do know     for G41 = Garage41 

-G41 00:80:25:09:9B:41 SMA001d SN: 2100060901 SN2100060901 Garage41   Type: SB 5000TL-20
 did work with sbfspot - v3.8.2        as such we expect that it should work also with v3.9.4
 if you give this is a try - make certain to apply full-debug options -d5 -v5   even if all works fine we like to see that result

- if that is OK  then you can create/try  with a seperate setup the results for the two other inverters (one step at a time) 

About your question - can sbfspot inform about the BTquality - yes - but only once and after you were able to connect/logon
- BT_signal (date 22okt) was at 66% - that is good,  - see the table one of the last columns
- on the display of your inverter there is also an indicator about BT
Another indication available from sbfspot  is part of the 'log-file'  
- Connecting to 00:80:25:09:9B:41 (4/10)      4 is retry counter - a normal BT-system needs less then 3retries
   the above example corresponds with the situation that your inverter is not yet "awake" eg before sunrise 
If you want to supervise the BT for a longer period, you will need a piece of SW, _sbfspot does read the status once_
- using a windows computer i can recommend  BTview from NIRsoft, that tool does listen to BTsenders that are active

On the label on your inverter - there might be the manufacturing date,
- if sbfspot is opartional then you can obtain the start of operation for your inverter via  sbfspot
- SMAinverters keep a long-term history for the day-by-day production - up to 30years   
    :-)    if you execute sbfspot  once   with the option -am144  you will get that information for the last 12years 

looking forward for the subsequent results     kr  sillie

Johannes106, 2022-04-28:
today i tried to make a new config with the wizard of sbfspot. i had chosen all 3 BTaddresses of my inverters and enabled MIS. I wondered about the message: "Installation complete". then I looked into the related config file (SBFspot.cfg.20220428_175715) but there were the values of the (last) config before. So I'm thinking that SBFspot don't use the current values of the gui but the values of the (last) config before. Even if I enable MIS there is just one BTadress in the config-file.

I was looking on the (paper)labels of the inverters but couldn't find the date of installation or manufacturing. But i saw there are different types of inverters: SB 5000TL-20 ( 00:80:25:09:9B:41), SB 4000TL-20 ( 00:80:25:09:93:6B), AB 4000TL-20 (00:80:25:09:A8:FA).
So I think that SB 4000TL-20 do not work with sbfspot.

Johannes106, 2022-04-28:
`41_am144.log`: 
```
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -cfgSBFspot.cfg.20220424_41 -am144
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg.20220424_41'
Thu Apr 28 18:23:54 2022: INFO: Starting...
sunrise: 05:41
sunset : 20:26
Connecting to 00:80:25:09:9B:41 (1/10)
Initialising...
SUSyID: 125 - SessionID: 853983179
SMA netID=01
SUSyID: 78 - SN: 2100060901
BT Signal=67.1%
Logon OK
Failed to get current plant time (-1)
SetPlantTime returned an error: -1
SUSyID: 78 - SN: 2100060901
Device Name:      SN: 2100060901
Device Class:     Solar Inverters
Device Type:      SB 5000TL-20
Software Version: 03.20.01.R
Serial number:    2100060901
Packet status: 21
SUSyID: 78 - SN: 2100060901
Device Status:      Ok
SUSyID: 78 - SN: 2100060901
Device Temperature: 47.3°C
SUSyID: 78 - SN: 2100060901
GridRelay Status:      Closed
SUSyID: 78 - SN: 2100060901
Energy Production:
	EToday: 39.514kWh
	ETotal: 83452.887kWh
	Operation Time: 53514.41h
	Feed-In Time  : 51689.59h
SUSyID: 78 - SN: 2100060901
DC Spot Data:
	MPPT 1 Pdc:   0.596kW - Udc: 284.12V - Idc:  2.100A
	MPPT 2 Pdc:   0.595kW - Udc: 279.52V - Idc:  2.131A
	Calculated Total Pdc:   1.191kW
AC Spot Data:
	Phase 1 Pac :   1.203kW - Uac: 232.63V - Iac:  5.173A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   1.203kW - Calculated Pac:   1.203kW
	Efficiency  :  101.01%
SUSyID: 78 - SN: 2100060901
Grid Freq. : 49.98Hz
SUSyID: 78 - SN: 2100060901
Current Inverter Time: 10/05/1974 20:22:01
Inverter Wake-Up Time: 10/05/1974 08:05:41
Inverter Sleep Time  : 10/05/1974 20:22:05
********************
* ArchiveDayData() *
********************
startTime: 28/04/2022 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime: 01/04/2022 13:00:00
ExportMonthDataToCSV: There is no data to export!
**********************
* ArchiveMonthData() *
**********************
startTime: 01/03/2022 11:00:00
ExportMonthDataToCSV: There is no data to export!
**********************
[... 688 lines cut ...]
ExportMonthDataToCSV: There is no data to export!
**********************
* ArchiveMonthData() *
**********************
startTime: 01/08/2010 12:00:00
ExportMonthDataToCSV: There is no data to export!
**********************
* ArchiveMonthData() *
**********************
startTime: 01/07/2010 12:00:00
ExportMonthDataToCSV: There is no data to export!
**********************
* ArchiveMonthData() *
**********************
startTime: 01/06/2010 12:00:00
ExportMonthDataToCSV: There is no data to export!
**********************
* ArchiveMonthData() *
**********************
startTime: 01/05/2010 12:00:00
ExportMonthDataToCSV: There is no data to export!
Reading events: 2022-Apr-01
ExportEventsToCSV()
Thu Apr 28 18:24:17 2022: INFO: Done.
```

Johannes106, 2022-04-28:
`41_d5v5.log`:
```
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -cfgSBFspot.cfg.20220424_41 -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg.20220424_41'

Configuration settings:
BTAddress=00:80:25:09:9B:41
MIS_Enabled=1
BTConnectRetries=10
Password=<undisclosed>
Plantname=garage41
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=52.52
Longitude=13.4
Timezone=Europe/Berlin
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=1
CSV_Spot_WebboxHeader=0
End of Config

Thu Apr 28 17:50:41 2022: INFO: Starting...
sunrise: 05:41
sunset : 20:26
Connecting to 00:80:25:09:9B:41 (1/10)
Initialising...
SUSyID: 125 - SessionID: 894962828
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 17 00 69 00 00 00 00 00 00 
00000010: 01 00 00 00 00 00 01 02 76 65 
00000020: 72 0D 0A 
23 Bytes sent
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 41 9B 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 01 00 00 00 00 01 00 00 
00000030: 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 41 9B 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 01 00 00 00 00 01 00 00 
00000030: 00 
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=01
--------: 00 01 02 03 04 05 06 07 08 09
[... 981 lines cut ...]
00000040: 5D 00 00 15 00 00 00 18 80 01 
00000050: 02 10 70 80 40 46 62 80 CD 6D 
00000060: 62 B4 5B 7E 
cmd=1
PacketLength=64
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 09 D0 7D 00 8C 
00000010: 0C 58 35 00 E0 4E 00 E5 62 2C 
00000020: 7D 00 00 15 00 00 00 18 80 01 
00000030: 02 10 70 80 40 46 62 80 CD 6D 
00000040: 62 B4 5B 7E 
<<<=================================>>>
ExportEventsToCSV()
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3B 00 45 04 71 DA 7D 1A 00 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 08 A0 FF FF FF FF FF 
00000030: FF 00 03 7D 5D 00 8C 0C 58 35 
00000040: 00 03 00 00 00 00 19 80 0E 01 
00000050: FD FF FF FF FF FF F6 0A 7E 
59 Bytes sent
Thu Apr 28 17:50:54 2022: INFO: Done.
```

Johannes106, 2022-04-28:
`6B_d5v5.log`:
```
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -cfgSBFspot.cfg.20220424_6B -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg.20220424_6B'

Configuration settings:
BTAddress=00:80:25:09:93:6B
MIS_Enabled=1
BTConnectRetries=10
Password=<undisclosed>
Plantname=garage6B
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=52.52
Longitude=13.4
Timezone=Europe/Berlin
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=1
CSV_Spot_WebboxHeader=0
End of Config

Thu Apr 28 17:50:10 2022: INFO: Starting...
sunrise: 05:41
sunset : 20:26
Connecting to 00:80:25:09:93:6B (1/10)
Initialising...
SUSyID: 125 - SessionID: 910505454
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 17 00 69 00 00 00 00 00 00 
00000010: 01 00 00 00 00 00 01 02 76 65 
00000020: 72 0D 0A 
23 Bytes sent
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 9 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 6B 93 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 6B 93 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
<<<=================================>>>
MAX_pcktBuf is now 27 bytes
Thu Apr 28 17:50:12 2022: CRITICAL: Incompatible FW version detected.
Thu Apr 28 17:50:12 2022: INFO: Done.
```

Johannes106, 2022-04-28:
`FA_d5v5.log`:
```
SBFspot V3.9.4
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bitCommandline Args: -v -cfgSBFspot.cfg.20220424_FA -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg.20220424_FA'

Configuration settings:
BTAddress=00:80:25:09:A8:FA
MIS_Enabled=1
BTConnectRetries=10
Password=<undisclosed>
Plantname=garageFA
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=52.52
Longitude=13.4
Timezone=Europe/Berlin
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=1
CSV_Spot_WebboxHeader=0
End of Config

Thu Apr 28 17:52:59 2022: INFO: Starting...
sunrise: 05:41
sunset : 20:26
Connecting to 00:80:25:09:A8:FA (1/10)
Initialising...
SUSyID: 125 - SessionID: 911677815
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 17 00 69 00 00 00 00 00 00 
00000010: 01 00 00 00 00 00 01 02 76 65 
00000020: 72 0D 0A 
23 Bytes sent
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 9 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1B 00 65 FA A8 09 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 03 
00000020: 70 00 01 01 00 00 00 
<<<=================================>>>
MAX_pcktBuf is now 27 bytes
Thu Apr 28 17:53:00 2022: CRITICAL: Incompatible FW version detected.
Thu Apr 28 17:53:00 2022: INFO: Done.
```

silliewimons, 2022-04-29:
dear joe/johannes106   thank you for getting all information,         after reading your info, we find    in summary

id      BT                          info                                        plant-nam   type                 status/comment  
-G6B 00:80:25:09:93:6B  SMA0001d SN: 2100063324  Garage6B   SB 4000TL-20   uses older-protocol/firmware
-G41 00:80:25:09:9B:41  SMA001d SN: 2100060901    Garage41   SB 5000TL-20   date-is-not-actual      1974mei
-HFA 00:80:25:09:A8:FA  SMA0001d SN: 2100068131  HouseFA    AB 4000TL-20   uses older-protocol/firmware

-G41-status
-communication from sbfspot with your inverter G41 works correct - BT is OK - retrieval of spot data from your inverter is OK
**main-problem** - "system-date" on your inverter is not set correctly - your inverter assumes that it is actually  1974mayxx,
   - records in spot-table are notified using the time-stamp from your raspberry (a choice in the config)
consequent:
   - sbfspot uses '2022apr' as reference to search for  day-table  and  month-table    
   - sbfspot does not modify the date-time:  delta-correction is 'limited to =~1hour

-G41- solution:   correct  the  date/time  by using one of the tools that SMA has available
 - from my information - there is no function that allows to set Date/Time via direct manipulation eg."knocking-on-cabinet"
 - apply:  "windows system - BT-stick"   - SMA's "SunnyExplorer"
   BE AWARE:  SMA has improved the security rules - SMA wants that on all installations a 'strong-password' is used
        as a result - latest versions of SunExplorer REQUIRE to modify password from default '0000' to a 'strong password'
        if you agree to do this - remember that you must insert that new password in the sbfspot config file
- - - - - - 

-G6B  &  HFA   from our observation / reading of your log-information: 
- we assume that your inverters SB4000TL-20 an AB4000 only support the older SMA-protocol (Data1)
  Data1 is an open protocol 

-XX-solution:_**  SMA has "SunnyDataControl" software,  on internet you can find this software  and there are user-groups
  we do not have experience whether SunnyExplorer supports the two types of protocol

info: SMA started ~2008 to develop the newer Data2+ protocol  (Data2+ is proprietery SMA)  
- the development of the project sbfspot is fully oriented towards the SMA Data2+ protocol, 
         which is not downward compatible with Data1

kr sillie

Johannes106, 2022-04-29:
dear sillie thank you for your help. 
Now i am understanding the problem. 
I will search for a solution to get the data of the old SB4000TL with the open protocol. I can't find any link to download "SunnyDataControl".

kr johannes

sbf-, 2022-04-30:
> I can't find any link to download "SunnyDataControl".

You can find it [here](https://www.sma.de/en/service/downloads.html) in the Archive/Software section

You can also find more recent firmware versions in Archive/Sunny Boy to make your inverters SunnyExplorer/SBFspot compatible (You didn't tell us which FW versions are installed on the different inverters)

By setting SMA netID > 1 for ALL your inverters they will show up as one plant in SunnyExplorer/SBFspot (provided they have FW version > 1.71)
Changing netID requires the inverters to open, so be careful.
Anyway, you will need SunnyExplorer to set the date/time because SBFspot is unable to do it

closed 2022-05-26 by sbf-