SBFspot/SBFspot #537: add installer user option in cfg file

peterb99, 2022-05-05:
The option to use installer as user can only be set on the command line . When doing this the password field in the cfg file is ignored. 
Please add option in cfg file to specify installer as user login.

silliewimons, 2022-05-05:
dear peter,     the usage of installer options are an exception    its main usage is to extract history of "installer"events
    
a login as installer does not add additional information: not for RealTime = spot-table   neither for daily or monthly files
for that reason   the one who needs a login at "installer" level can do this via the command line options 
option1:   -installer        instructs to Login as installer     option2: -password:xxxx      Installer password

kr silly

silliewimons, 2022-05-06:
Ahahah - i understand the problem                               perhaps other members may have a suggestion / experience   

 - for your information - older versions of SunnyExplorer  did not instruct to modify the password,
   but that does not eleminate your problem   "unknown" user password
 - whenever you should get blocked with the 'installer' password - there is a procedure with SMA to unlock-passwords
    but from my last information there is a fee to be paid
    with that procedure - SMA provide  the equivalent  of  "PUK-code" (compare with GSM-PIN-PUK)
kr  silly

peterb99, 2022-05-06:
I have a installer password, but changes of user password does not seem to be accepted by one inverter. I had a hell of a time to get all the SMA components to use the same password. Used the latest explorer. Got there in the end except for one. One thing to be aware of is when you change passwords to also change the system password in sunny portal , option is not easy to find there. In sunny portal, go to configuration, device overview, select home manager, properties, select edit. This will bring you to the password field to edit, PV System Password: . Set this to the installer password.

closed 2022-05-26 by sbf-