SBFspot/SBFspot #538: add sbfspot man pages

peterb99, 2022-05-08:
Please add man pages for the sbfspot executables.

SBFspot, 2022-05-09:
Try SBFspot -? for help

peterb99, 2022-05-09:
How about a bit of explanation where/how the tag files fit in with the data read from sma devices.

Im getting garbage data for SMA Sunny Island 6.0H in the csv file in de device type and description fields . Looks like I need to read the actual code to get an understanding of the relationship between the SBFspot data and the way data is added to the csv files.
Line from <plantname>Battery-20220507.csv
07/05/2022 20:50:02; ^B<93>^A^R;à^C<93>^A^Q;1261010721;300,000;0,000;0,000;1,600;0,000;0,000;248,100;0,000;0,000;320,000;0,000;0,000;50,000;0,000;0,000;Ok;55,000;28,300;55,500;6,100;29,000;0,000

sbf-, 2022-05-09:
> Line from Battery-20220507.csv
> 
> ```
> 07/05/2022 20:50:02; ^B<93>^A^R;à^C<93>^A^Q;1261010721;300,000;0,000;0,000;1,600;0,000;0,000;248,100;0,000;0,000;320,000;0,000;0,000;50,000;0,000;0,000;Ok;55,000;28,300;55,500;6,100;29,000;0,000
> ```

Hmm, that's a bug (See #539)

peterb99, 2022-05-10:
background 1: See attached files
background 2: yes it can do 3 phase, see modes specs in 13_32009R_V10/modbuslist_en.htm
[MODBUS-HTML_SI44M-80H-13_32009R_V10.zip](https://github.com/SBFspot/SBFspot/files/8660124/MODBUS-HTML_SI44M-80H-13_32009R_V10.zip)
[Edgehead-Battery-20220510.csv](https://github.com/SBFspot/SBFspot/files/8660133/Edgehead-Battery-20220510.csv)
[MyPlant_20220510.log](https://github.com/SBFspot/SBFspot/files/8660141/MyPlant_20220510.log)
Note: I use 3 config files, one for Ip inverter sb3600, one for ip sunny island and one for BT. I cannot reach the SN1700 on BT network directly hence seeing all the BT devices. SBFspot seem to have issues with the loadswitches on BT, seeing them as inverters.
l

peterb99, 2022-05-10:
Debug output from /usr/local/bin/sbfspot.3/SBFspot -v5 -d4 -cfg/usr/local/bin/sbfspot.3/SBFspot-Island.cfg > /tmp/island-debug.log
[island-debug.log](https://github.com/SBFspot/SBFspot/files/8660255/island-debug.log)

peterb99, 2022-05-10:
Would it be an idea to add a parameter in the cfg file to specify 1 or 3 phase as plant property, so extra phase parameters from inverters can be discarded if only one phase is used at the plant.

peterb99, 2022-05-10:
screenshots of island data via sma explorer:
[island data screenshots.pdf](https://github.com/SBFspot/SBFspot/files/8660635/island.data.screenshots.pdf)

silliewimons, 2022-05-11:
hello peterb99,  looking to your list of remarks/questions    one does refer to several 'independent' cases

allow me to suggest to handle each case seperate - eg. to open a new issue for the points that need further analysis
-?- you have a lot of equipment in use: -SB3600; -SunIsland; -SB1700; -BTrepeater; -PCwithSunExplorer; -RPi?-withSBFspot;
     you also refer to loadswitches (brand-info?);  if this is about load control, then there is a control-unit involved 

-case1-SB3600(IPconnected):  from your log-file and information - one does not see wrong/strange behaviour
            one can assume this inverter is operating as expected

-case2-SunIsland(IPconncected): from the log-file (for reporting problems, please apply the options -d5 -v5)
            one can assume that sbfspot does not yet include all "interrogations" for 'data'
            as you will be aware - one needed 'reverse-engineering' to understand the codes of SMA-equipment,
   to extend the 'sbfspot'knowledge, we do need your assistance,  
            that does mean, eg that one uses SunnyExplorer and tools to capture 'data-stream' to/from inverter
            for instance - sbfspot does not find an archive for month-data ?? 

-case3-SB1700(BT)  and BTrepeater(s): 
    please inform about the communication-setup from your RPi  up to  inverter
    your log indicates that, in the BT-communication-channel there are (at least) 3retransmissions done    
              with retransmissions in communication, the effect of "time-outs / repeats" can lead to a status where 
              --start-point detects time-out and resends the request while the earlier request is still in process
              --a common approach is that "units" go silent for a short while     
    can you inform/confirm:                 (??you did not select the full-options -d5 -v5 - so this is a guess)
start-point: RPi    
hop1:   ?? - ??     SN-0172008928   -> to SN-0172009412 
hop2:   ?? - ??     SN-0172009412   -> to SN-0172003165  
hop3:   ?? - ??     SN-0172003165   -> to SN-0170006109       
hop4: BTrepeater - SMA     SN-0170006109  -> to  SN-2002060299 
end-point: SB1700 SN-2002060299      BTadress??

kr silly

peterb99, 2022-05-11:
case2: opened new case with requested data
case 1: No bug , general comment/suggestion to add a parameter in config file to select 1 phase or 3 phase setup
case 3: No real bugs there, issue I have is that hcitool ( raspberry) only sees one of the load controllers , so the BT address of that one is used in the cfg file. The load controllers also seem to be acting as BT hub repeaters, with the SB1700 at the end , the last BT node is a real BT repeater.
So I need to specify MIS_Enabled=1 to get to the SB1700, but really only want data from the SB1700. Would it be possible for SFBspot to check for type==inverter and ignore all other BT nodes?

silliewimons, 2022-05-11:
dear peterb99          case2-noted
case1: see sbfspot - discussions - item526
  -- basis-limitation - this item should be adressed by "SMA" at config-level,
case3: SB1700 and BT - can you inform about the details for the "loadcontrollers",  which equipment supervises these units
  -- in the SMAdocumentation for BT the repeat aspect does not include "loadcontrollers"   "repeat" is a general BTfeature
  -- if you have a WindowsPC (with BT) there is the alternative to use BTview from NIRsoft,
      that tool does show a list of BTequipment - perhaps it shows the BTadress of the repeater / inverter
  -- if intermedatie hops with the loadcontrollers are not used  -  is option to modify netID on SMAinverter/repeater valid??
      OR you can try/test to put MIS=0  and  use the adress of the BTrepeater
   -- be aware about limits of the BTrange as specified by SMA-itself - see BT-in-practice 
kr silly

silliewimons, 2022-05-13:
hej,  peterb99 - about case3  BTcommunication:    - SB1700&BTrepeat  <->   -HM&load-controllers
        this item will need more   study&analysis         let me explain
  --- usage of BT-HP (class1=HighPower)  does not fit well with BT-LE (class2.5> LowEnergy)
       HP -> long-distance   (limited nbr of senders per 2500m²)     LE -> short&extreme-short distance (lots of senders on 16m²)  
       applications HP: -cell-phone to car; -. . .           -applic-LE: -eardops; -PCmouse; -PCkeyboard; -. . . .     
  --- one need more info to understand whether BTnetworks   for   HP <-> LE  should be kept/configured seperate                

HP-from2008 till 201x    SMA's usage of BTcommunication:  -master: SunnyExplorer&BT-SunnyBeam   slave: inverters
        usage of BTclass1 sender/receivers - BTclass1 is for distance upto 100meter 
LE-recently:    SMA uses BT related to IoTapplication (Internet-of-Things):   -master: HomeManager    slave: loadcontrol
        IoT uses BT5.x  incl.network-mesh & low-energy BTclass2.5 (distance <10m.)  class3 (distance <3m.)  class4 (distance <1m.)
        !!! a distance of ~2meter covers a circle of 16m² 

kr   silly

sbf-, 2024-01-19:
I'm a bit lost here... in case there's still an issue I propose to start a new issue with recent debug info (-v5 -d5)

closed 2024-01-19 by sbf-