SBFspot/SBFspot #539: Garbage in battery CSV file

sbf-, 2022-05-09:
I'm getting garbage data for SMA Sunny Island 6.0H in the csv file in the device type and description fields.
Line from Battery-20220507.csv
```
07/05/2022 20:50:02; ^B<93>^A^R;à^C<93>^A^Q;1261010721;300,000;0,000;0,000;1,600;0,000;0,000;248,100;0,000;0,000;320,000;0,000;0,000;50,000;0,000;0,000;Ok;55,000;28,300;55,500;6,100;29,000;0,000
```

sbf-, 2022-05-26:
> but it is an ugly workaround

Say no more ;-)

kayD, 2022-05-31:
Hi.
I tried to update to the fix version 3.9.5, but it seems not to be available:
'''
~ $ curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash -s 3.9.5
SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 3 Model B Plus Rev 1.3
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 11 (bullseye)
username=akarui
userhome=/home/akarui
Minimum SBFspot version supported by this tool is 3.5.0
Request to install version 3.9.5
Version 3.9.5 is not available for installation.
Select one of 3.8.3 3.8.4 3.9.2 3.9.3 3.9.4
'''

commit d7220136 (2022-05-26): Fix #539 Garbage in battery CSV file