SBFspot/SBFspot #540: error values and missing data for sunny island battery inverter

peterb99, 2022-05-11:
I've attached a wireshark trace of data communication between sma explorer and sunny island. Not sure if it captures all data as I only see UDP traffic . Can you have a look at it if it makes sense or needs further tweaking of wireshark capture. 
SMA explorer runs in virtual box on windows 7 on my mac, wireshark directly on the mac filtering data to and from the sma island inverters IP address
[islandtrace.pcapng.zip](https://github.com/SBFspot/SBFspot/files/8669395/islandtrace.pcapng.zip)
.

peterb99, 2022-05-11:
debug -d5 output,
[island-debug5.log](https://github.com/SBFspot/SBFspot/files/8669444/island-debug5.log)

sbf-, 2022-05-13:
Trace looks OK, but ATM I don't have much time.
I'll have a look coming week

sbf-, 2023-03-31:
Some time ago...
Is this still an issue?

I see some errors in the debug log, battery data looks OK, but I'm not sure about the inverter time.

closed 2024-01-19 by sbf-