SBFspot/SBFspot #545: compile errors redhat 8 SBFspot V3.9.5

DJE007, 2022-06-06:
compile errors redhat 8 SBFspot V3.9.5

make mysql && make install_mysql
test -d mysql/obj/ || mkdir -p mysql/obj/
test -d mysql/bin/ || mkdir -p mysql/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/boost_ext.o
g++ main.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/main.o
g++ misc.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/misc.o
g++ sunrise_sunset.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/sunrise_sunset.o
g++ SBFNet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/SBFNet.o
g++ CSVexport.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/CSVexport.o
g++ Ethernet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/Ethernet.o
g++ EventData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/EventData.o
EventData.cpp: In member function ‘std::__cxx11::string EventData::EventDescription() const’:
EventData.cpp:116:30: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 3 has type ‘size_t’ {aka ‘long unsigned int’} [-Wformat=]
                 sprintf(idx, "|x%x|", i);
                              ^~~~~~~  ~
EventData.cpp:123:34: warning: format ‘%X’ expects argument of type ‘unsigned int’, but argument 3 has type ‘size_t’ {aka ‘long unsigned int’} [-Wformat=]
                     sprintf(idx, "|x%X|", i);
                                  ^~~~~~~  ~
g++ Inverter.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/Inverter.o
g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/ArchData.o
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/SBFspot.o
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/TagDefs.o
g++ Bluetooth.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/Bluetooth.o
g++ mqtt.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/mqtt.o
g++ db_MySQL.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/db_MySQL.o
g++ db_MySQL_Export.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/db_MySQL_Export.o
db_MySQL_Export.cpp: In member function ‘int db_SQL_Export::exportEventData(InverterData**, TagDefs&)’:
db_MySQL_Export.cpp:292:5: error: ‘my_bool’ was not declared in this scope
     my_bool is_null = true;
     ^~~~~~~
db_MySQL_Export.cpp:292:5: note: suggested alternative: ‘bool’
     my_bool is_null = true;
     ^~~~~~~
     bool
db_MySQL_Export.cpp:391:42: error: ‘is_null’ was not declared in this scope
                     values[9].is_null = &is_null;
                                          ^~~~~~~
db_MySQL_Export.cpp:391:42: note: suggested alternative: ‘isnanl’
                     values[9].is_null = &is_null;
                                          ^~~~~~~
                                          isnanl
db_MySQL_Export.cpp:399:43: error: ‘is_null’ was not declared in this scope
                     values[10].is_null = &is_null;
                                           ^~~~~~~
db_MySQL_Export.cpp:399:43: note: suggested alternative: ‘isnanl’
                     values[10].is_null = &is_null;
                                           ^~~~~~~
                                           isnanl
make: *** [makefile:116: mysql/obj/db_MySQL_Export.o] Error 1

DJE007, 2022-06-06:
Yes, x86_64

DJE007, 2022-06-06:
cat /etc/redhat-release
Red Hat Enterprise Linux release 8.6 (Ootpa)

SBFspot, 2022-06-06:
I compile/test with MariaDB instead of MySQL
[This issue only affects MySQL; MariaDB will continue to support my_bool.](https://github.com/Motion-Project/motion/issues/995)
I’ll fix it asap

closed 2022-06-06 by sbf-

DJE007, 2022-06-07:
More errors from master

make mysql && make install_mysql
test -d mysql/obj/ || mkdir -p mysql/obj/
test -d mysql/bin/ || mkdir -p mysql/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/boost_ext.o
g++ main.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/main.o
g++ misc.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/misc.o
g++ sunrise_sunset.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/sunrise_sunset.o
g++ SBFNet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/SBFNet.o
g++ CSVexport.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/CSVexport.o
g++ Ethernet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/Ethernet.o
g++ EventData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/EventData.o
g++ Inverter.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/Inverter.o
g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/ArchData.o
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/SBFspot.o
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/TagDefs.o
g++ Bluetooth.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/Bluetooth.o
g++ mqtt.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/mqtt.o
g++ db_MySQL.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/db_MySQL.o
g++ db_MySQL_Export.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -std=c++11 -DUSE_MYSQL -o mysql/obj/db_MySQL_Export.o
db_MySQL_Export.cpp: In member function ‘int db_SQL_Export::exportEventData(InverterData**, TagDefs&)’:
db_MySQL_Export.cpp:394:42: error: cannot convert ‘char*’ to ‘bool*’ in assignment
                     values[9].is_null = &is_null;
                                          ^~~~~~~
db_MySQL_Export.cpp:402:43: error: cannot convert ‘char*’ to ‘bool*’ in assignment
                     values[10].is_null = &is_null;
                                           ^~~~~~~
make: *** [makefile:116: mysql/obj/db_MySQL_Export.o] Error 1

commit 2f7e26c9 (2022-06-06): Fix #545