SBFspot/SBFspot #549: After upgrade to v395, the CSV file is no more written

kayD, 2022-06-07:
I was using v394 with "no" issues (except #539, why I upgraded), on Raspi.
After upgrade to v395, the CSV file is no more written.

Some investigation shows that CSV_Export=0 is assumed while it is set to 1:
```
akarui@akarui:~/sbf $ grep CSV_Exp battery-day.cfg 
# CSV_Export (default 1 = Enabled)
CSV_Export=1
akarui@akarui:~/sbf $ /usr/local/bin/sbfspot.3/SBFspot -v5 -ad1 -am0 -ae1 -finq -cfg/home/akarui/sbf/battery-day.cfg | grep CSV_Exp
CSV_Export=0
```
I confirmed that the config file is indeed read, by setting CSV_Export=2:
```
akarui@akarui:~/sbf $ nano battery-day.cfg 
akarui@akarui:~/sbf $ /usr/local/bin/sbfspot.3/SBFspot -v5 -ad1 -am0 -ae1 -finq -cfg/home/akarui/sbf/battery-day.cfg | grep CSV_Exp
Invalid value for 'CSV_Export' (0-1)
```
So *something* is setting this to 0...

kayD, 2022-06-07:
Having read that the default of CSV_Export would be 1 anyway, I tried to outsmart sbf by not setting the value, but alas. it is still 0.
Conclusion: maybe not a problem of the config parser, but reset to 0 later.

kayD, 2022-06-07:
I found the bug: Line 1888 in SBFspot/SBFspot.cpp.
See https://github.com/SBFspot/SBFspot/commit/05acde8f665b33638a6866170bc639cf93bb25be#r75555650

silliewimons, 2022-06-08:
at kayD  
THANK YOU  -  you even included the 'workaround' that allows to further explore - additional functions  in v395
kr sillie

cambronf, 2022-08-15:
The meaning of -nocsv is still inverted:

Reason fix #549 is wrongly implemented:

in SBFspot.cpp at line 1887: true should be false
    //Overrule CSV_Export from config with Commandline setting -nocsv
    if (cfg->nocsv)
        cfg->CSV_Export = **false**;

commit 6e3b5b96 (2022-06-08): Fix #549
commit 95ac40d4 (2022-08-27): Fix #549 (was not correctly fixed before)
closed 2022-08-27 by sbf-